echavet/MitsubishiCN105ESPHome #3: Compilation problem: byte

MarkoPaasila, 2024-01-16:
```
In file included from src/esphome.h:26,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/climateControls.cpp:1:
src/esphome/components/cn105/heatpumpFunctions.h:17:5: error: 'byte' does not name a type
   17 |     byte raw[MAX_FUNCTION_CODE_COUNT];
      |     ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:21:17: error: 'byte' has not been declared
   21 |     int getCode(byte b);
      |                 ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:22:18: error: 'byte' has not been declared
   22 |     int getValue(byte b);
      |                  ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:30:19: error: 'byte' has not been declared
   30 |     void setData1(byte* data);
      |                   ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:31:19: error: 'byte' has not been declared
   31 |     void setData2(byte* data);
      |                   ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:32:19: error: 'byte' has not been declared
   32 |     void getData1(byte* data) const;
      |                   ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:33:19: error: 'byte' has not been declared
   33 |     void getData2(byte* data) const;
      |                   ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:37:9: error: 'int heatpumpFunctions::getValue(int)' cannot be overloaded with 'int heatpumpFunctions::getValue(int)'
   37 |     int getValue(int code);
      |         ^~~~~~~~
src/esphome/components/cn105/heatpumpFunctions.h:22:9: note: previous declaration 'int heatpumpFunctions::getValue(int)'
   22 |     int getValue(byte b);
      |         ^~~~~~~~
In file included from src/esphome.h:26,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/cn105.cpp:4:
src/esphome/components/cn105/heatpumpFunctions.h:17:5: error: 'byte' does not name a type
   17 |     byte raw[MAX_FUNCTION_CODE_COUNT];
      |     ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:21:17: error: 'byte' has not been declared
   21 |     int getCode(byte b);
      |                 ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:22:18: error: 'byte' has not been declared
   22 |     int getValue(byte b);
      |                  ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:30:19: error: 'byte' has not been declared
   30 |     void setData1(byte* data);
      |                   ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:31:19: error: 'byte' has not been declared
   31 |     void setData2(byte* data);
      |                   ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:32:19: error: 'byte' has not been declared
   32 |     void getData1(byte* data) const;
      |                   ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:33:19: error: 'byte' has not been declared
   33 |     void getData2(byte* data) const;
      |                   ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:37:9: error: 'int heatpumpFunctions::getValue(int)' cannot be overloaded with 'int heatpumpFunctions::getValue(int)'
   37 |     int getValue(int code);
      |         ^~~~~~~~
src/esphome/components/cn105/heatpumpFunctions.h:22:9: note: previous declaration 'int heatpumpFunctions::getValue(int)'
   22 |     int getValue(byte b);
      |         ^~~~~~~~
*** [.pioenvs/msz-fd50/src/esphome/components/cn105/climateControls.cpp.o] Error 1
*** [.pioenvs/msz-fd50/src/esphome/components/cn105/cn105.cpp.o] Error 1
In file included from src/esphome.h:26,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/componentEntries.cpp:1:
src/esphome/components/cn105/heatpumpFunctions.h:17:5: error: 'byte' does not name a type
   17 |     byte raw[MAX_FUNCTION_CODE_COUNT];
      |     ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:21:17: error: 'byte' has not been declared
   21 |     int getCode(byte b);
      |                 ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:22:18: error: 'byte' has not been declared
   22 |     int getValue(byte b);
      |                  ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:30:19: error: 'byte' has not been declared
   30 |     void setData1(byte* data);
      |                   ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:31:19: error: 'byte' has not been declared
   31 |     void setData2(byte* data);
      |                   ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:32:19: error: 'byte' has not been declared
   32 |     void getData1(byte* data) const;
      |                   ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:33:19: error: 'byte' has not been declared
   33 |     void getData2(byte* data) const;
      |                   ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:37:9: error: 'int heatpumpFunctions::getValue(int)' cannot be overloaded with 'int heatpumpFunctions::getValue(int)'
   37 |     int getValue(int code);
      |         ^~~~~~~~
src/esphome/components/cn105/heatpumpFunctions.h:22:9: note: previous declaration 'int heatpumpFunctions::getValue(int)'
   22 |     int getValue(byte b);
      |         ^~~~~~~~
In file included from src/esphome.h:26,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/extraComponents.cpp:1:
src/esphome/components/cn105/heatpumpFunctions.h:17:5: error: 'byte' does not name a type
   17 |     byte raw[MAX_FUNCTION_CODE_COUNT];
      |     ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:21:17: error: 'byte' has not been declared
   21 |     int getCode(byte b);
      |                 ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:22:18: error: 'byte' has not been declared
   22 |     int getValue(byte b);
      |                  ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:30:19: error: 'byte' has not been declared
   30 |     void setData1(byte* data);
      |                   ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:31:19: error: 'byte' has not been declared
   31 |     void setData2(byte* data);
      |                   ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:32:19: error: 'byte' has not been declared
   32 |     void getData1(byte* data) const;
      |                   ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:33:19: error: 'byte' has not been declared
   33 |     void getData2(byte* data) const;
      |                   ^~~~
src/esphome/components/cn105/heatpumpFunctions.h:37:9: error: 'int heatpumpFunctions::getValue(int)' cannot be overloaded with 'int heatpumpFunctions::getValue(int)'
   37 |     int getValue(int code);
      |         ^~~~~~~~
src/esphome/components/cn105/heatpumpFunctions.h:22:9: note: previous declaration 'int heatpumpFunctions::getValue(int)'
   22 |     int getValue(byte b);
      |         ^~~~~~~~
*** [.pioenvs/msz-fd50/src/esphome/components/cn105/componentEntries.cpp.o] Error 1
*** [.pioenvs/msz-fd50/src/esphome/components/cn105/extraComponents.cpp.o] Error 1
```

echavet, 2024-01-16:
I will try to reproduce asap.

MarkoPaasila, 2024-01-17:
I'm using a ESP01, and it seems that I can't specify platform type, and only arduino is available. Here's my config:

```
esphome:
  name: msz-fd50

esp8266:
  board: esp01_1m

api:
  encryption:
    key: "lhauih34rl8ihlai8hflwoi84haloihwlioy3u="
  on_client_disconnected:
    - logger.log: "API client disconnected!"
    - climate.control:
        id: hp
        mode: HEAT_COOL
        target_temperature: 20
        fan_mode: AUTO

logger:
  baud_rate: 0

ota:
  password: "hiag4i7gfaiuw4gfaigwleiuo4"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  use_address: 192.168.10.22
  power_save_mode: LIGHT
  output_power: 12dB

  ap:
    ssid: "Msz-Fd50 Fallback Hotspot"

external_components:
  - source: github://echavet/MitsubishiCN105ESPHome

time:
  - platform: sntp
    id: sntp_time
    on_time:
      - seconds: 01
        minutes: 01
        hours: 22
        then:
          - climate.control:
              id: hp
              mode: HEAT_COOL
              target_temperature: 20
              fan_mode: AUTO

sensor:
  - platform: uptime
    name: MSZ-FD50 Uptime

  - platform: wifi_signal
    name: MSZ-FD50 WiFi Signal
    update_interval: 60s

climate:
  - platform: cn105
    id: hp
    name: "MSZ-FD50"
    hardware_uart: UART0
    update_interval: 500ms
      
    
button:
  - platform: restart
    name: "MSZ-FD50 Esphome Restart"
```

MarkoPaasila, 2024-01-17:
Forgot to mention that I'm using ESPHome 2023.12.7. I also did "Clean Build Files".

echavet, 2024-01-18:
ok, I found something. The original project from geoffdavis/esphome-mitsubishiheatpump uses the swiCago lib which includes arduino.h.
My test yaml are including some other libs that make the byte available.
Yours do not. 
But that that's my fault as the code must not depend on the yaml.
So I pulled changes in the Globals.h to include the required lib. And I've changed the "byte" occurences by uint8_t.
But I could not test on my heatpump. 
But I can close the issue as it is solved by includes.
Something tells me you have other problems to come in you're conf because of the use of time... good luck for that.

closed 2024-01-18 by echavet

MarkoPaasila, 2024-01-18:
Now it compiles properly for ESP8266. Excellent!

I have some other problems indeed, but I think I will first acquire some regulators as you recommend.