echavet/MitsubishiCN105ESPHome #5: Please make RX/TX configurable for ESP32 one way or the other :-)

Bascht74, 2024-01-20:
Hi, 

great work so far!
I would rather use your component for my Mitsubishi components than others. 

I am using a M5Stack ATOM Lite:
[image]

with a 50 cm groove cable:
[image]

For connecting I removed the white connector on one end of the groove cable and inserted the clamp of the cables to a CN105-Connector, same as in your picture:
[image]

So I have just the ATOM Lite with the cable that you can directly connect to the Mitsubishi device, nothging more (keep it simple and easy)

BUT: the ATOM Lite uses GPIO26 and GPIO32
[image]

I am right now using the fork from wrouesnel, because he allowed to change the RX/TX pin here:
https://github.com/wrouesnel/esphome-mitsubishiheatpump/commit/127d2a0ef9b4e755026a8e1fdf2e510200f3a986

Would it be possible to add that possibility as well in your fork?

Or... could you maybe revert the UART part of the custom integration back to the original ESPHome UART-Implementation?
According to https://esphome.io/components/uart.html the problems with the parity have been fixed for the ESP8266: 

> From ESPHome 2021.8 the ESP8266SoftwareSerial UART write_byte function had the parity bit fixed to be correct for the data being sent. This could cause unexpected issues if you are using the Software UART and have devices that explicity check the parity. Most likely you will need to flip the parity flag in YAML.
> parity (Optional): The parity used on the UART bus. Options: NONE, EVEN, ODD. Defaults to NONE.

... that way the component would follow more the ESPHome standards and everybody could use the ESPHome to config the pins...

Cheers from good old Germany, 
Sebastian

echavet, 2024-01-22:
Hi,
Thanks for your mail.
I've just committed changes for your ask.

could you please test, because I don't have any device to check it's ok.

don't forget to add the refresh options to force update to head

external_components:
- source: github://echavet/MitsubishiCN105ESPHome
refresh: 0s

Bascht74, 2024-01-22:
Will check…

Bascht74, 2024-01-22:
Compile failed:

```
Compiling .pioenvs/klima-og-tim/src/esphome/components/api/api_connection.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/api/api_frame_helper.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/api/api_pb2.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/api/api_pb2_service.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/api/api_server.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/api/list_entities.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/api/proto.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/binary_sensor/automation.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/binary_sensor/filter.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/climate/climate.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/climate/climate_mode.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/climate/climate_traits.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/cn105/climateControls.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/cn105/cn105.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/cn105/componentEntries.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/cn105/extraComponents.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/cn105/heatpumpFunctions.cpp.o
Compiling .pioenvs/klima-og-tim/src/esphome/components/cn105/hp_readings.cpp.o
src/esphome/components/cn105/cn105.cpp: In member function 'void CN105Climate::setupUART()':
src/esphome/components/cn105/cn105.cpp:126:37: error: 'class HardwareSerial' has no member named 'pins'; did you mean 'find'?
             this->get_hw_serial_()->pins(this->tx_pin_, this->rx_pin_);
                                     ^~~~
                                     find
*** [.pioenvs/klima-og-tim/src/esphome/components/cn105/cn105.cpp.o] Error 1
========================= [FAILED] Took 31.55 seconds =========================
```

echavet, 2024-01-23:
Ok thanks
Sorry for that...

echavet, 2024-01-23:
the HardwareSerial class is not the same for ESP32 and ESP8266:

so I changed this:

```
#ifdef ESP8266
            this->get_hw_serial_()->begin(this->baud_, SERIAL_8E1);
            this->get_hw_serial_()->pins(this->tx_pin_, this->rx_pin_);
#elif defined(ESP32)                        
            this->get_hw_serial_()->begin(this->baud_, SERIAL_8E1, this->tx_pin_, this->rx_pin_);
#endif
```

you can try again

Bascht74, 2024-01-23:
It works, but I had to change / RX/TX. Could you change to: 

this->get_hw_serial_()->begin(this->baud_, SERIAL_8E1, this->rx_pin_, this->tx_pin_);
See: https://microcontrollerslab.com/esp32-uart-communication-pins-example/ (first parameter is rx ) 

:-)

Except that, it works with basic configuration. 
I will check more this evening and will migrate all devices…

echavet, 2024-01-23:
done
thanks for your help

Bascht74, 2024-01-23:
Thx, working now!

I see that you updated the readme! Thx!

Just one question: 

You wrote:

> Note: The update_interval is set here to 4s for debugging purposes. However, it is recommended to use a interval longer or equal to 1s because the underlying process divides this interval into three separate requests.

What do you suggest for the update_interval?
You said 4s for debugging (so that not that many messages will come in the log), and for normal operation and "snappy", quick update in Home Assistant of the local remotes just 1s?

echavet, 2024-01-24:
1 second seems like a good interval if your logs aren't set to be too verbose. The method buildAndSendRequestsInfoPackets defines 2 sub-intervals within update_interval_ by taking the longer of 300 ms or (update_interval_ / 4). This is likely to evolve based on user feedback. Maybe on this, GeoffDavis's choice is better...
Regardless, 3 types of information requests are sent between 2 intervals.
So setting it to 1 second actually equates to 250-300 ms in reality.

Bascht74, 2024-01-24:
Thx, one second is fine with me...

closed 2024-01-24 by Bascht74

tchlyah, 2025-07-10:
Thanks @Bascht74 for testing and validating this. Can you please share the yaml file you used with this configuration? 

Also, would this cable work without modification?

https://a.aliexpress.com/_EHBf4kw

Thanks !