echavet/MitsubishiCN105ESPHome #7: How to use the vane feature

Bascht74, 2024-01-24:
Hi, 

I was using settings for the horizontal and vertical vane with the old component

```
#    horizontal_vane_select:
#      name: Horizontale Luftstromrichtung
#    vertical_vane_select:
#      name: Verticale Luftstromrichtung

```

You wrote that this is available with your component, too: 

> Additional components: vane orientation (fully supporting the Swicago map), compressor frequency for energy monitoring, and i-see sensor.

But how?

I don't find any hint/documentation/example about it!
Maybe I am blind, could you send me to the right place?

Sebastian

Bascht74, 2024-01-24:
I fould information about differencing both verical / horizontal vanes and naming them in the yaml here:
https://github.com/wrouesnel/esphome-mitsubishiheatpump/commit/db95f70e16f6fab3bdb6a01aae4a27b54f5755f0

echavet, 2024-01-24:
Hello,
For the moment there is an extreComponent feature that builds a select named "Van orientation" associated with the climate component. It is filled with options from the VAN_MAP from the SwiCago lib.
It's a good idea to offer a way to customize the name. 
But for now it can be done (override name) in Home Assistant.
I will provide this option asap.

Bascht74, 2024-01-24:
Thx for your answer.

I have this option: 
[image]

But... is it for the vertical or horizonatal vane?
I have both:
[image]

in the SwiCago lib:
```
heatpumpSettings settings = {
    "ON",  /* ON/OFF */
    "FAN", /* HEAT/COOL/FAN/DRY/AUTO */
    26,    /* Between 16 and 31 */
    "4",   /* Fan speed: 1-4, AUTO, or QUIET */
    "3",   /* Air direction (vertical): 1-5, SWING, or AUTO */
    "|"    /* Air direction (horizontal): <<, <, |, >, >>, <>, or SWING */
};

bool operator==(const heatpumpSettings& lhs, const heatpumpSettings& rhs) {
  return lhs.power == rhs.power && 
         lhs.mode == rhs.mode && 
         lhs.temperature == rhs.temperature && 
         lhs.fan == rhs.fan &&
         lhs.vane == rhs.vane &&
         lhs.wideVane == rhs.wideVane &&
         lhs.iSee == rhs.iSee; 
}
```
So the "wideVane" (= horizontal) is missing, right now only the vertical is available within your component. 
The question is, if you could provide the horizontal as well as a select entity...

echavet, 2024-01-25:
It's for the vertical.
Horizontal is not supported yet because I have no module to test it.
But I will implement it asap

Bascht74, 2024-01-25:
Thx, I will test it then :-)
I have two different models that support it...

echavet, 2024-01-29:
The [new branch ](https://github.com/echavet/MitsubishiCN105ESPHome/tree/extra_components) is a work in progress to implement [wrouesnel](https://github.com/wrouesnel) extra select options for vertical and horizontal vane. 
I'm currently testing the component stability before merging. it looks good for the moment.

echavet, 2024-01-29:
```yaml
climate:
  - platform: cn105  
    name: ${friendly_name}
    compressor_frequency_sensor:
      name: Compressor frequency (clim Sejour)    
    vertical_vane_select:
      name: Orientation de la Vane Verticale
    horizontal_vane_select:
      name: Orientation de la Vane Horizontale
    remote_temperature_timeout: 30min
    update_interval: 10s   
```

Bascht74, 2024-01-29:
Great, that you enabled the possibility to change the names!

I will also try to disable the compressor_frequency_sensor with

 ```
   compressor_frequency_sensor:
      name: Compressor frequency (clim Sejour)    
      internal: true
```

I will test the new possibilities tomorrow!