echavet/MitsubishiCN105ESPHome #8: Processing timeouts for the external temperature feature

Bascht74, 2024-01-24:
Hi, sorry to disturb you again.

One last thing: 
As you now, I am using the external temp. sesors (I thing like you do). 
In the old integration there were some options about a fallback to the internal sensors:

#    remote_temperature_operating_timeout_minutes: 5 # The number of minutes before a set_remote_temperature request becomes stale, while the heatpump is heating or cooling. Unless a new set_remote_temperature request was made within the time duration, the heatpump will revert back to it's internal temperature sensor.
#    remote_temperature_idle_timeout_minutes: 20 # The number of minutes before a set_remote_temperature request becomes stale while the heatpump is idle. Unless a new set_remote_temperature request is made within the time duration, the heatpump will revert back to it's internal temperature sensor.
    # remote_temperature_ping_timeout_minutes: 5 # The number of minutes before a set_remote_temperature request becomes stale, if a ping request wasn't received from your ESPHome controller. This will result in the heatpump reverting to it's internal temperature sensor if the heatpump loses it's WiFi connection.

I thought this is very usefull if HA become unavailable or the battery for the bluetooth temp sensor dies. 
How is the fallback implemented within your component? Is there a strategy or anything by default?
I don't want to heat up the room becaus the external temp doesn't get updated...

Sebastian

echavet, 2024-01-24:
Yes I agree. I saw this is a great idea. I will implement this soon.
Thanks for you message

echavet, 2024-01-29:
done with only one option: remote_temperature_timeout
because I didn't see the point of having 3 timeout??
          

```
climate:
  - platform: cn105  
    name: ${friendly_name}
    id: "esp32_clim"
    compressor_frequency_sensor:
      name: Compressor frequency (clim Sejour)    
    vertical_vane_select:
      name: Orientation de la Vane Verticale
    horizontal_vane_select:
      name: Orientation de la Vane Horizontale
    remote_temperature_timeout: 30min
    update_interval: 10s   
```

closed 2024-01-29 by echavet

Bascht74, 2024-01-29:
Yeah! You are probably right.

Will that work with your way ("sending the temps via HA-Service-Calls")?
I think there as "remote_temperature_ping_timeout_minutes" for that, but I didn't get that either...

Bascht74, 2024-01-29:
Will test that!

echavet, 2024-02-05:
Yes it will work as soon as there are no call to set remote temperature for a longer time than the defined timeout period.

Bascht74, 2024-02-05:
Yes, it works