echavet/MitsubishiCN105ESPHome #9: HVAC_action wrong?

Bascht74, 2024-01-25:
Home Assistant shows the climate with: 
hvac_action: idle
if it is working fine and heating the room.

But the log of ESPHOME shows: 
```
[22:26:16][I][EVT_SETS:056]: [wantedSettings]-> [power: ON, target °C: 26.000000, mode: HEAT, fan: 3, vane: AUTO, hasChanged ? -> YES, hasBeenSent ? -> YES]
[22:26:16][I][SETTINGS:080]: [current]-> [power: ON, target °C: 26.000000, mode: HEAT, fan: 3, vane: AUTO]
[...]
[22:40:31][I][STATUS:100]: [received]-> [room C°: 22.0, operating: YES, compressor freq:  0 Hz]
[22:40:31][I][STATUS:100]: [current]-> [room C°: 22.0, operating: YES, compressor freq:  0 Hz]
[22:40:32][I][STATUS:100]: [received]-> [room C°: 22.0, operating: YES, compressor freq:  0 Hz]
[22:40:32][I][STATUS:100]: [current]-> [room C°: 22.0, operating: YES, compressor freq:  0 Hz]
```

So `Power: ON`, `operating: YES` --> `hvac_action: idle`?

possible actions are: 
```
HVACAction.OFF | Device is turned off.
HVACAction.PREHEATING | Device is preheating.
HVACAction.HEATING | Device is heating.
HVACAction.COOLING | Device is cooling.
HVACAction.DRYING | Device is drying.
HVACAction.FAN | Device has fan on.
HVACAction.IDLE | Device is idle.
```
ESPHome here: https://github.com/esphome/esphome/blob/dev/esphome/components/climate/climate_mode.h

At the beginning I thing the HVAC_Action should be 
"FAN" (Power: ON, operating: No) and later on 
"HEATING" (Power: ON, operating: YES, HVAC_mode = HEAT, probably same with cool, dry...)

btw. the compressor feq seems not to be supported with my models... (one base outside and multiple devices)...

Bascht74, 2024-01-25:
I think I fould out, that you are using the frequency for setting the device to idle I the fequency is "0". 
But my devices don't send any frequency. I think this is, becaue there is only on device outside and multiple devices inside.

So 
HVAC_MODE= HEAT Power: ON, operating: YES should still be HVAC_Action "HEATING" in my case. 
HVAC_MODE= HEAT Power: ON, operating: NO should still be HVAC_Action "FAN" in my case. 
HVAC_MODE= HEAT Power: OFF, operating: NO should still be HVAC_Action "Off" in my case. 

Sometime at the beginning the device shows -1 hz one time and then only 0 hz.

If I could set something like:
`frequency_available: false` in yaml 
... there could be different logic for that case (and the fequency sensor won't be generated).

echavet, 2024-01-26:
Yes !
Thanks for all your feedback.

echavet, 2024-01-26:
should be ok now

closed 2024-01-26 by echavet

fwump38, 2025-04-25:
Is there a way to get these changes somehow? My unit doesn't send compressor frequency and thus my state is always IDLE

Do I need to specify a version for this external component or?