echavet/MitsubishiCN105ESPHome #11: Add example for external temp sensors

Bascht74, 2024-01-26:
Maybe you could add an example for external temp sensors. 

If you like, you could use my code: 

```
sensor:
  - platform: pvvx_mithermometer
    mac_address: "A4:C1:38:XX:XX:XX"
    temperature:
      name: Thermometer
      id: temperature
      device_class: temperature
      state_class: measurement
      filters:
        throttle_average: 90s
      on_value:
        then:
          - lambda: 'id(use_your_name).set_remote_temperature(x);'
```

echavet, 2024-01-27:
Sure. I have to.
But today I can't work on it. I promised to help a friend move in.

Would you like to be a contributor on this project?

Bascht74, 2024-01-29:
> Would you like to be a contributor on this project?

As I am not a programmer but just a guy who understands logic I am a bit worried that I break stuff...
If there is something I can help you with, please let me know.

echavet, 2024-02-05:
Well with GitHub if you break something we just have to rollback!
I'll have a look at the contributing options of GitHub. I'm new to GitHub...