echavet/MitsubishiCN105ESPHome #12: Min/Max Temp?

disruptivepatternmaterial, 2024-01-27:
I looked through the code and didnt see a way to enforce a min / max code. I know I can do this on the HA side, but would like to also make sure it is enforced no matter what.

I also have vert and horizontal with iSee units, so happy to test anything you do with these more uncommon / advanced options.

echavet, 2024-01-27:
Hi,

Thanks for your message.

I'm currently working on a new version which will have separate select for
horizontal and vertical vanes. For now there is a only a separated vertical
vane select which support full range of orientations. But some of them are
not compatible with the climate component which can result in a specific
msg in the ha UI.

I will then work on timeouts to fall back to internal temperature when the
external sensor is not available.

I did not test the iSee sensor but I think it should work. You should find
an extra sensor in the device entities from the ha esp integration.
You will have to use automations and set triggers on iSee sensor changes...
If you have any idea or will of specific feature you're welcome.
I'm not sure I understand what you mean by :

"I looked through the code and didnt see a way to enforce a min / max code.
I know I can do this on the HA side, but would like to also make sure it is
enforced no matter what".

Could you explain me please.

Be aware that there is now, at the head of main, a version which requires a
different way to configure UART compared to what we were used to do with
Geoff Davis version. The read me shows examples.

jasestu, 2024-01-27:
Are you referring to setting a min and max temperature so that it starts cooling when above the max and starts heating when below the min while displaying that new red and blue double temperature climate card in Home Assistant?

Bascht74, 2024-01-28:
my configuration for min/max:

```
climate:
  - platform: cn105
    id: hp
    name: "${friendly_name}"
    supports:
      mode: ["HEAT_COOL", "DRY", "COOL", "HEAT", "FAN_ONLY"]
      fan_mode: ["AUTO", "DIFFUSE", "LOW", "MEDIUM", "MIDDLE", "HIGH"]
      swing_mode: ["OFF", "VERTICAL"]
    visual:
      min_temperature: 18
      max_temperature: 27
      temperature_step:
        target_temperature: 1
        current_temperature: 0.1
    update_interval: 1s
```

closed 2024-01-29 by echavet