echavet/MitsubishiCN105ESPHome #15: Getting a lot of log messages since the last update

Bascht74, 2024-01-29:
```
[00:09:02][I][app:102]: ESPHome version 2023.12.9 compiled on Jan 30 2024, 00:05:23
[00:09:03][W][lookup:172]: Attention valeur 0 non trouvée, on retourne la valeur au rang 0
[00:09:04][W][lookup:172]: Attention valeur 0 non trouvée, on retourne la valeur au rang 0
[00:09:05][W][lookup:172]: Attention valeur 0 non trouvée, on retourne la valeur au rang 0
```

I will check, if that device doesn't support the horizontal vane...
If that is true, maybe the value "0" could do a log output "not supported"... :-)

Bascht74, 2024-01-29:
I removed the horizontal vane:

```
...
    update_interval: 1s  
#    compressor_frequency_sensor:
#      name: Compressor frequency (clim Sejour) 
#      internal: true   
    vertical_vane_select:
      name: Verticale Luftstromrichtung
#    horizontal_vane_select:
#      name: Horizontal Vane
    remote_temperature_timeout: 30min  
```

But still getting the log output. 
Is there a way to find out where it comes from?

echavet, 2024-01-30:
Thanks. I'll have a look today.
But you are right it comes from one of these maps values binding functions.
But I don't know which one yet. Hope I'll be able to reproduce.
Are you sure it is the temperature one ?
However I'll first add some logs.

Bascht74, 2024-01-31:
Yes, that is the case:

```
[23:49:31][W][lookup:172]: wideVane reading caution: value 0 not found, returning value at index 0
[23:49:32][W][lookup:172]: wideVane reading caution: value 0 not found, returning value at index 0
```

Bascht74, 2024-01-31:
It isn’t possible to chance the setting, reverting back to 0x00…

Bascht74, 2024-01-31:
I commented out the horizontal vane select for that device:

```
climate:
  - platform: cn105
    id: hp
    name: "${friendly_name}"
    supports:
      mode: ["HEAT_COOL", "DRY", "COOL", "HEAT", "FAN_ONLY"]
      fan_mode: ["AUTO", "DIFFUSE", "LOW", "MEDIUM", "MIDDLE", "HIGH"]
      swing_mode: ["OFF", "VERTICAL", "HORIZONTAL", "BOTH"]
    visual:
      min_temperature: 18
      max_temperature: 27
      temperature_step:
        target_temperature: 1
        current_temperature: 0.1
    update_interval: 1s  
#    compressor_frequency_sensor:
#      name: Compressor frequency (clim Sejour) 
#      internal: true   
    vertical_vane_select:
      name: Verticale Luftstromrichtung
#    horizontal_vane_select:
#      name: Horizontal Vane
    remote_temperature_timeout: 30min  
```

No entity is showing up in HA, but the component still does the lookup.

Would it be a good idea to discard lookups for sensors not created or do you want to leave it that way?
If you want to leave it that way, maybe it might the good to change the log level for that error to verbose?

echavet, 2024-02-01:
thanks for feedback, I finally changed the lookupByteMapValue signature to add a defaultValue.

`    const char* lookupByteMapValue(const char* valuesMap[], const uint8_t byteMap[], int len, uint8_t byteValue, const char* debugInfo = "", const char* defaultValue = nullptr);`
if provided, the defaultValue is returned with no warning:

```
const char* CN105Climate::lookupByteMapValue(const char* valuesMap[], const uint8_t byteMap[], int len, uint8_t byteValue, const char* debugInfo, const char* defaultValue) {
    for (int i = 0; i < len; i++) {
        if (byteMap[i] == byteValue) {
            return valuesMap[i];
        }
    }

    if (defaultValue != nullptr) {
        return defaultValue;
    } else {
        ESP_LOGW("lookup", "%s caution: value %d not found, returning value at index 0", debugInfo, byteValue);
        return valuesMap[0];
    }

}
```
and in the decoder part it now looks like this:
`receivedSettings.wideVane = lookupByteMapValue(WIDEVANE_MAP, WIDEVANE, 7, data[10] & 0x0F, "wideVane reading", WIDEVANE_MAP[2]);`
which is the "|" option by default.

I'm not sure this is the best solution. What do you think ?

> It isn’t possible to chance the setting, reverting back to 0x00…
sorry what do you mean ?

Bascht74, 2024-02-01:
The other solution could be not to call the `lookupByteMapValue` if there is no name tag in the yaml and no select entity is created.

So if you comment out: 
#    horizontal_vane_select:
#      name: Horizontal Vane

then no select entity for that will be created and - accordingly - lookupByteMapValue shouldn't be called (cause decoder doesn't need to decode this) -> no log 

------
It isn’t possible to chance the setting, reverting back to 0x00…
sorry what do you mean ?

That I cannot change the value, if I change the select entity it will change back after one second. In my opinion the device doesn't support it if the device doesn't accept new values be changing the select entity and sill send "0" back in the next message.

echavet, 2024-02-01:
Yes that's what's happening.

echavet, 2024-02-02:
"yes I began with the option you described.
I think you're right. It is the best way to manage this." 
well, actually not so simple :

Configuring the climate component not to support the control of the horizontal vane doesn't mean the heatpump itself does not support the feature. It can be just because you don't want to manage it from homeassistant or you don't want to have the external select component for it. But being able to read its state can still make sense. 

Could it make more sense if it was based on the presence of `VERTICAL` or `BOTH` swing modes in the hp supporting traits?

Bascht74, 2024-02-03:
I don't know. Sounds like a good idea if you can use that swing mode data from the HP to generate the vane select entities.

But if you want to read the vanes states, but not want to populate the sensor to Home Assistant you could easily add `internal: true` under the name...

for me, both seems to be fine. 
With the actual version it works for me :-)

closed 2024-02-03 by echavet