echavet/MitsubishiCN105ESPHome #17: Unable to build

enomam, 2024-01-31:
Hi,

I'm trying out to build this, but am getting a compilation error when trying to install.
I'm using an ESP-01, and am running ESPHome 2023.10.13.

```
In file included from src/esphome.h:13,
                 from src/main.cpp:3:
src/esphome/components/cn105/cn105.h:23:51: error: 'esphome::sensor' has not been declared
   23 |     void set_compressor_frequency_sensor(esphome::sensor::Sensor* compressor_frequency_sensor);
      |                                                   ^~~~~~
src/esphome/components/cn105/cn105.h:23:65: error: expected ',' or '...' before '*' token
   23 |     void set_compressor_frequency_sensor(esphome::sensor::Sensor* compressor_frequency_sensor);
      |                                                                 ^
src/esphome/components/cn105/cn105.h:34:5: error: 'sensor' does not name a type
   34 |     sensor::Sensor* compressor_frequency_sensor_ =
      |     ^~~~~~
```

My yaml looks a bit like this:
```
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome
 
uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3

climate:
  - platform: cn105  # remplis avec la plateforme de ton choix
    name: ${friendly_name}
    id: "study_ac"    
    update_interval: 4s         # shouldn't be less than 1 second
    # horizontal_vane_select:
    #   name: ${name} Horizontal Vane
    #   id: "clim_sejour_horizontal_vane"
    #vertical_vane_select:
    #  name: ${friendly_name} Vertical Vane
    #  id: "study_ac_vertical_vane"
    #compressor_frequency_sensor:
    #  name: ${friendly_name} Compressor Frequency
    #  id: "study_ac_compressor_frequency"
```

echavet, 2024-01-31:
Sorry for that.
Let me try to reproduce...

echavet, 2024-01-31:
Well I can't manage to reproduce. 

- try to add add the refresh = 0 attribute and do a `esphome clean` before compile
```yaml
external_components:
 - source: github://echavet/MitsubishiCN105ESPHome
   refresh: 0s
```
- try to upgrade you esphome version ?

Previous version didn't have the compressor_frequency_sensor option. This is a new feature. Perhaps your building workspace needs to be refreshed? 
If you need more support please send me more of your yaml configuration.

Bascht74, 2024-01-31:
I can build, your suggestions should do the trick...

enomam, 2024-01-31:
Still no luck.
Updated to 2023.12.9.
Using the webui, and clicked "clean build files".

```
Compiling .pioenvs/esphome-web-15a184/libfb1/ESP8266WiFi/ESP8266WiFiGratuitous.cpp.o
In file included from src/esphome.h:13,
                 from src/main.cpp:3:
src/esphome/components/cn105/cn105.h:23:51: error: 'esphome::sensor' has not been declared
   23 |     void set_compressor_frequency_sensor(esphome::sensor::Sensor* compressor_frequency_sensor);
      |                                                   ^~~~~~
src/esphome/components/cn105/cn105.h:23:65: error: expected ',' or '...' before '*' token
   23 |     void set_compressor_frequency_sensor(esphome::sensor::Sensor* compressor_frequency_sensor);
      |                                                                 ^
src/esphome/components/cn105/cn105.h:34:5: error: 'sensor' does not name a type
   34 |     sensor::Sensor* compressor_frequency_sensor_ =
      |     ^~~~~~
src/main.cpp: In function 'void setup()':
src/main.cpp:396:45: error: invalid conversion from 'esphome::CompressorFrequencySensor*' to 'int' [-fpermissive]
  396 |   study_ac->set_compressor_frequency_sensor(study_ac_compressor_frequency);
      |                                             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                             |
      |                                             esphome::CompressorFrequencySensor*
In file included from src/esphome.h:13,
Compiling .pioenvs/esphome-web-15a184/libfb1/ESP8266WiFi/ESP8266WiFiMulti.cpp.o
                 from src/main.cpp:3:
src/esphome/components/cn105/cn105.h:23:59: note:   initializing argument 1 of 'void CN105Climate::set_compressor_frequency_sensor(int)'
   23 |     void set_compressor_frequency_sensor(esphome::sensor::Sensor* compressor_frequency_sensor);
      |                                          ~~~~~~~~~~~~~~~~~^~~~~~
Compiling .pioenvs/esphome-web-15a184/libfb1/ESP8266WiFi/ESP8266WiFiSTA-WPS.cpp.o
Compiling .pioenvs/esphome-web-15a184/libfb1/ESP8266WiFi/ESP8266WiFiSTA.cpp.o
Compiling .pioenvs/esphome-web-15a184/libfb1/ESP8266WiFi/ESP8266WiFiScan.cpp.o
Compiling .pioenvs/esphome-web-15a184/libfb1/ESP8266WiFi/WiFiClient.cpp.o
Compiling .pioenvs/esphome-web-15a184/libfb1/ESP8266WiFi/WiFiClientSecureBearSSL.cpp.o
*** [.pioenvs/esphome-web-15a184/src/main.cpp.o] Error 1
```

This is my complete yaml. I kept the comments in, so you can see that I came from the geaoffdavis version initially.
```
substitutions:
  name: esphome-web-15a184
  friendly_name: Study AC - ESP01

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

#external_components:
#  - source: github://geoffdavis/esphome-mitsubishiheatpump

external_components:
  - source: github://echavet/MitsubishiCN105ESPHome
    refresh: 0s

esp8266:
  board: esp01_1m

logger:
  # ESP8266 only - disable serial port logging, as the HeatPump component
  # needs the sole hardware UART on the ESP8266
  baud_rate: 0
  logs:
    component: ERROR

mqtt:
  broker: 10.0.10.110

  #on_json_message:
  #  topic: zigbee2mqtt/0x00158d0002e6ea83
  #  then:
  #    - lambda: |-
  #        if (x.containsKey("temperature")){
  #          id(study_ac).set_remote_temperature(x["temperature"]);
  #        }

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-15A184"
    password: "****"

captive_portal:

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3

climate:
  - platform: cn105  # remplis avec la plateforme de ton choix
    name: ${friendly_name}
    id: "study_ac"    
    update_interval: 4s         # shouldn't be less than 1 second
    horizontal_vane_select:
       name: ${name} Horizontal Vane
       id: "clim_sejour_horizontal_vane"
    vertical_vane_select:
      name: ${friendly_name} Vertical Vane
      id: "study_ac_vertical_vane"
    compressor_frequency_sensor:
      name: ${friendly_name} Compressor Frequency
      id: "study_ac_compressor_frequency"

#climate:
#  - platform: mitsubishi_heatpump
#    name: ${friendly_name}
#    hardware_uart: UART0
#    baud_rate: 2400
#    id: study_ac
#   #update_interval: 500ms
#    supports:
#      mode: [HEAT_COOL, COOL, HEAT, DRY, FAN_ONLY]
#      fan_mode: [AUTO, DIFFUSE, LOW, MEDIUM, HIGH]
#      swing_mode: ["OFF", VERTICAL]
#    visual:
#      min_temperature: 16
#      max_temperature: 31
#      temperature_step: 0.5

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: ${friendly_name} Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: ${friendly_name} WiFi Signal
    update_interval: 60s
    
```

echavet, 2024-02-01:
ok thanks a lot for this feedback
I've been able to reproduce the compile error. 
It's weird but there is no more error when you add an api section.
So to solve the issue, for the moment, just add something like this in your yaml file:

```
api:
  encryption:
    key: !secret encryption_key
```

I'm gonna try to find why and correct this.
Thanks again

Bascht74, 2024-02-01:
better use: 

```
api:
  encryption:
    key: !secret encryption_key
    reboot_timeout: 0s # disable reboot, default is 15 min
```

echavet, 2024-02-01:
ok I found that an include was missing.
You shouldn't have anymore compile errors with head version

enomam, 2024-02-01:
Great!
I'll try it out tonight after work!

enomam, 2024-02-02:
Yup!
That did the trick!

Working well, but two issues (let me know if I should raise a new topic).

1. Compressor frequency is always 0Hz. I've tried this on an MSZ-AP20VG, and MSZ-AP35VG
2. The blue LED on the ESP-01 blinks 3 times every few seconds - not 100% how to turn this off. Seems to be at the same rate as the console logs.

echavet, 2024-02-02:
- For the compressor frequency: do you have multiple modules installed on a single external compressor ? Are you sure your MSZ-APXXVG support the feature?
- For the blinking led I know I have to work on that. You can open an issue for that.

closed 2024-02-02 by echavet

enomam, 2024-02-02:
There are multiple modules attached to the compressor.
And I'm not sure if the  MSZ-APXXVG support reporting on the frequency. Where should I look?

And thanks - I'll raise an issue for the blinking LED - thanks for your help on this one.

Bascht74, 2024-02-03:
For me the frequency is not reported as well in my multi device setup.