echavet/MitsubishiCN105ESPHome #18: add possibilty to name the iSee entity (or disabling it by not naming it)

Bascht74, 2024-01-31:
Right now for all my devices an iSee sensor is created: 
[image]

Othen than the vanes for frequency sensors it is not possible to name the sensor. 
Not naming should do that the iSee sensor is not getting created in HA...

This is the last little thing I found so far. 
I will now change all other 5 devices (three different types) to the new version.

echavet, 2024-02-01:
I don't think it causes troubles in home assistant because sensors are entities attached to a single device and with a unique ID. Well I'm not absolutely sure of that!
However. I committed the update but couldn't really test it because I don't have any isee sensor on my heatpumps.

disruptivepatternmaterial, 2024-02-02:
related to this...is it possible to turn the iSee to a switch? I want to reenable it after it seems to get disabled sometimes.

echavet, 2024-02-02:
I think this is a readonly sensor. 
Unless I did not understand the sensor utility, to me it makes no sense to turn it into a button as it is used to detect persons in the room. 
I might be wrong, my heatpumps do not have such sensors.

echavet, 2024-02-02:
or I see what you mean now.
You want to enable/disable the use of the sensor by your modules?
Well I don't actually know what packet to send to the heatpump for that. I think SwiCago did not implement this in his lib.

disruptivepatternmaterial, 2024-02-02:
I figured that might be the case 👍🏽 I think iSee if uncommon (or has been) in North America. I have seen them in Europe and Asia for a long time. It was hard to get one here in America, but I have 1. At some point I am going to retrofit to ceiling cassettes with iSee, but for now only have 1. If I can provide logs to help (if you are interested) I'll send along, but otherwise it is really minor. I can always turn it back on with the remote.

echavet, 2024-02-03:
OK thanks for your feedback. 
Do you think there is an issue with the iSee sensor ?
Because I'd gladly watch your logs but I'm not really sure I understand what the problem is:
Do you mean the MitsubishiCN105ESPHome component does switch the iSee feature **OFF** on your unit, and that you have to set it back **ON** with the remote ?

disruptivepatternmaterial, 2024-02-05:
yeah something is getting lost here :) I think the problem is I tinker late and night or early in the morning HAHAHA I'll try to be more detailed: but I will send along logs when I can. So what happens is I want iSee most of the time because this unit is in my office, which is not used most of the day/night. But when I manually change the temp using your code it turns off iSee, so when I come back into the office in the morning I have to turn it back on with the remote. Ideally I would love to automate this. So I can make an automation that turns iSee back on say at 8am every workday.

echavet, 2024-02-05:
did you try other implementations (from other reposirories)? Do they disable the unit iSee sensor too ?

disruptivepatternmaterial, 2024-02-05:
I have tried them all. To my understanding none of them even are aware of iSee, but the kumo adapter (when it decided to randomly work) would. I have an idea here: I think the bit you are reporting on is if the unit is currently using iSee. Not if it is enabled. Why do I think this? Because it would seem when I am in the office it is 1, when I am not it is 0, but I will keep testing this - Ill not it as I go in and out of the room through the day.

closed 2024-02-06 by echavet