echavet/MitsubishiCN105ESPHome #20: Blue LED blinks regularly

enomam, 2024-02-03:
I've got this running on an ESP-01, attached to an  MSZ-AP20VG.
When in operation, the blue LED blinks 3 times, followed by a pause, and then repeats. It appears to be in time with the 4s update rate.

This is raised following up from the discussion here: https://github.com/echavet/MitsubishiCN105ESPHome/issues/17#issuecomment-1923489314

Let me know if you need more details, or can provide additional debug.

echavet, 2024-02-03:
I need to search for the reason. It could be because of logging instructions.

echavet, 2024-02-05:
I've tried to disable logging. Actually I've  limited them to only report changing activity. Because I want to keep an eye on things... and it looks like there is no blink. Bu it is on my esp32. Could you try to reduce the logs too and tell me more?

```yaml
# Enable logging
logger:    
  level: DEBUG
  logs:
    EVT_SETS : INFO
    WIFI : INFO
    MQTT : INFO
    WRITE_SETTINGS : INFO
    SETTINGS : INFO
    STATUS : INFO
    CN105Climate: WARN
    CN105: INFO
    climate: WARN
    sensor: WARN
    chkSum : INFO
    WRITE : WARN
    READ : WARN
    Header: INFO
    Decoder : INFO
    CONTROL_WANTED_SETTINGS: INFO  
```

enomam, 2024-02-06:
Still getting blinking on the ESP_01 with the following logger details:

```
esp8266:
  board: esp01_1m

logger:
  baud_rate: 0
  #hardware_uart: UART1
  level: DEBUG
  logs:
    EVT_SETS : INFO
    WIFI : INFO
    MQTT : INFO
    WRITE_SETTINGS : INFO
    SETTINGS : INFO
    STATUS : INFO
    CN105Climate: WARN
    CN105: INFO
    climate: WARN
    sensor: WARN
    chkSum : INFO
    WRITE : WARN
    READ : WARN
    Header: INFO
    Decoder : INFO
    CONTROL_WANTED_SETTINGS: INFO  
```

enomam, 2024-02-08:
In your README, you mention this:
>I recommended to use a interval longer or equal to 1s because the underlying process divides this interval into three separate requests.

Three blinks, three requests? Not 100% sure what triggers the blue LED on the ESP-01

echavet, 2024-02-08:
The blue led is not well documented.Difficult to have an idea but I
remember having read somewhere that if was related to the log. I can't
remember exactly but it was something like il you log something to warning
or error level the blue led blinks... the lack of documentation forces to
look into the esphome sources to get information. And I haven't done it yet.

disruptivepatternmaterial, 2024-02-21:
You can always put some nail polish or tape on it :) 

maybe this?

```
esphome:
name: ${name}
platform: ESP8266
board: nodemcuv2
status_led:
pin:
number: <<<<<YOUR PIN>>>> (I think it is 
inverted: true
```

But more details are here https://www.instructables.com/How-to-use-the-ESP8266-01-pins/

and @echavet is right, since most of them put the LED on the TX PIN, turning off logging all together on serial use should turn it off. But if you are using that for the connection to the split...you are out of luck. I'd just put tape over it.

closed 2024-02-22 by echavet