echavet/MitsubishiCN105ESPHome #21: Delay in Applying Target Temperature Setting - requires calling twice

jasestu, 2024-02-03:
With https://github.com/geoffdavis/esphome-mitsubishiheatpump I was able to use a badge button on a dashboard to turn on and set temperature:

    badges:
      - entity: climate.kitchen_heat_pump_2
        type: custom:climate-mode-entity-row
        modes:
          - hvac_mode: cool
            temperature: 23
            icon: mdi:snowflake
            color: '#B3E5FC'
          - hvac_mode: 'off'
          - hvac_mode: heat
            temperature: 21
            icon: mdi:fire
            color: '#FFC107'
            
However, I'm finding with this component I need to hit the button twice - the first time turns it on, but doesn't set the temperature, I need to hit the button a second time to have that happen.

Relevant Logs:

    Initial cool mode command with incorrect target temperature retention:

[12:37:01][I][control:160]: changing mode to COOL
[12:37:01][I][EVT_SETS:059]: [control (wantedSettings)]-> [power: ON, target °C: 27.000000, mode: COOL]

Delayed acceptance of target temperature change to 23°C:

[12:37:13][I][control:050]: Setting heatpump setpoint: 23.0
[12:37:13][I][EVT_SETS:059]: [wantedSettings ]-> [power: ON, target °C: 23.000000, mode: COOL]

Final successful application of the target temperature:

    [12:37:22][D][climate:399]: Mode: COOL
    [12:37:22][D][climate:425]: Target Temperature: 23.00°C

Expected Behavior:
The heat pump should set the target temperature to 23°C immediately upon setting the mode to cool without requiring a second command.

Actual Behavior:
The target temperature initially remains at an incorrect setting (27°C) until the command is sent again, after which it correctly updates to 23°C.

Steps to Reproduce:

    Set the heat pump to cool mode with a target temperature of 23°C through the Home Assistant interface.
    Observe that the target temperature does not apply until the command is issued a second time.

Environment:
    ESPHome version: 2023.12.9

echavet, 2024-02-03:
Thanks for this nice feedback.
I'll have a look.

echavet, 2024-02-03:
Thank you for the `lovelace-climate-mode-entity-row` tip, I didn't know this one.
So I installed it and **test your scenario** on both of my units, one is esp32 the other is esp8266. And it works fine. I mean, no problem with setting both the power ON and the temperature. 
But I've had the idea to test the "_mqtt pathway_". Because I have a MQTT section in my climates yaml configuration. **Do you have one too?**
What I found is that I can reproduce the exact same behavior when lovelace-climate-mode-entity-row  points to the MQTT entity. 
Of course it doesn't mean this is not an issue but a MQTT control one...
If it is your case please test with a reference of you unit through the esphome integration.
If not, I will have to dig more...

jasestu, 2024-02-03:
I'll try and do some more structured testing later today and see what I can find.
Not using MQTT, but I do have an external temperature sensor for this one (although I have others without external temperature sensors and I get the same behaviour).

```
substitutions:
  name: "kitchen-heatpump"
  friendly_name: Kitchen Heat Pump

esphome:
  name: ${name}

esp8266:
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

captive_portal:

external_components:
  - source: github://echavet/MitsubishiCN105ESPHome

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3

climate:
  - platform: cn105
    name: ${friendly_name}
    id: kithp
    update_interval: 4s
    remote_temperature_timeout: 30min

logger:
  baud_rate: 0

web_server:
  port: 80

time:
  - platform: homeassistant
    id: homeassistant_time

text_sensor:
  - platform: version
    name: ${name} ESPHome Version
  - platform: wifi_info
    ip_address:
      name: ${name} IP
    ssid:
      name: ${name} SSID
    bssid:
      name: ${name} BSSID

sensor:
  - platform: uptime
    name: ${name} Uptime
  - platform: wifi_signal
    name: ${name} WiFi Signal
    update_interval: 60s
  # External Temperature Sensor
  - platform: homeassistant
    name: "'AG One' Temperature Sensor From Home Assistant"
    entity_id: sensor.ag_one_temperature
    on_value:
      then:
        - lambda: 'id(kithp).set_remote_temperature(x);'

```

echavet, 2024-02-03:
Thanks for all that. 
I know what's happening. It is a real bug. Requests are coming very closed to each others because the [lovelace-climate-mode-entity-row](https://github.com/piitaya/lovelace-climate-mode-entity-row/blob/master/climate-mode-entity-row.js) component in the setMode() method line 113, is calling `this._hass.callService("climate", ...)` as many times as there are settings assigned to a button. So the my climate control() method is called as many times in a short period.
It would have been simpler if a service was available to send all the settings in one call.
However my component shouldn't loose any of these settings. But it does!!
So I'm gonna work on this and come back to you.
Sorry for the issue!

jasestu, 2024-02-03:
> Thank you for the `lovelace-climate-mode-entity-row` tip, I didn't know this one.

I haven't been able to codify a complete set and forget model for what I want the heat pumps to do, so little buttons are nice to have:

[image]

> Sorry for the issue!

Don't be! It's awesome to have someone actively developing for this. I was excited to tap into the compressor data, but my units don't seem to send anything.

Indoor <> Outdoor
MSZ-AP35VGD <> MUZ-AP35VG
MSZ-AP60VGD <> MUZ-AP60VG
MSZ-AP25VGD (x4) <> MXZ-4F80VGD
MSZ-AP71VGD <> MUZ-AP71VG

echavet, 2024-02-04:
Yes it's a nice interface.
It would be nice to have the cool or the heat but not both depending on the season.

> I was excited to tap into the compressor data, but my units don't seem to send anything.

It seems to have a lot of units not supporting the compressor data. 
I think swicago libraries has been developed with a reverse engineering of a regular WiFi module. 
There might be slight differences between different unit ref.  
Having said that, I don't see data in your logs because the unit is always OFF. But I see in your configuration that you did not define the option for the compressor sensor to be advertise. I don't want to give you false hopes as it looks like you don't have the up to date version that featured this change, but for the last version, you now have to define a` compressor_frequency_sensor` section in the climate component.

```yaml
climate:
  - platform: cn105  
    name: ${friendly_name}
    id: "esp32_clim"
    compressor_frequency_sensor:
      name: Compressor frequency (clim Sejour)    
    vertical_vane_select:
      name: Orientation de la Vane Verticale
    horizontal_vane_select:
      name: Orientation de la Vane Horizontale
    isee_sensor:
      name: ISEE Sensor
    remote_temperature_timeout: 30min
    update_interval: 10s         # shouldn't be less
```
if you want to be sure to compile the last head sources each time you build the firmware, you also have to add the
`refresh: 0s` option in the` external_components:` section.

echavet, 2024-02-05:
I've just committed a correction to address this issue.
A new parameter can be customized is `debounce_delay`. It prevent multiple wantedSettings from being sent if control method is called too many times in a short delay.
By default the delay is set to 400ms.

Configuration example:
```yaml
uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: GPIO17
  rx_pin: GPIO16

climate:
  - platform: cn105  
    name: ${friendly_name}
    id: "esp32_clim"
    compressor_frequency_sensor:
      name: Compressor frequency (clim Sejour)    
    vertical_vane_select:
      name: Orientation de la Vane Verticale
    horizontal_vane_select:
      name: Orientation de la Vane Horizontale
    isee_sensor:
      name: ISEE Sensor
    remote_temperature_timeout: 5min
    update_interval: 1s         # shouldn't be less than 1 second
    debounce_delay : 500ms       # delay to prevent bouncing 
```

jasestu, 2024-02-06:
Brilliant - I recompiled without modifying my config and the default debounce delay seems to have done the trick.
I'll refresh the rest of my heat pumps and see if it's working as expected over the next couple of days.

Regarding the compressor reporting - whether or not I define the sensor in the configuration, I still get 0hz reported, so I'm assuming my models aren't compatible.

[17:21:26][I][STATUS:106]: [received       ]-> [room C°: 25.5, operating: NO , compressor freq:  0 Hz]
[17:21:26][I][STATUS:106]: [current        ]-> [room C°: 25.0, operating: NO , compressor freq:  0 Hz]
[17:21:26][D][sensor:093]: 'Compressor frequency (Office)': Sending state 0.00000 Hz with 0 decimals of accuracy
[17:22:02][I][STATUS:106]: [received       ]-> [room C°: 25.5, operating: YES, compressor freq:  0 Hz]
[17:22:02][I][STATUS:106]: [current        ]-> [room C°: 25.5, operating: NO , compressor freq:  0 Hz]
[17:22:02][D][sensor:093]: 'Compressor frequency (Office)': Sending state 0.00000 Hz with 0 decimals of accuracy
[17:22:41][I][STATUS:106]: [received       ]-> [room C°: 25.0, operating: YES, compressor freq:  0 Hz]
[17:22:41][I][STATUS:106]: [current        ]-> [room C°: 25.5, operating: YES, compressor freq:  0 Hz]
[17:22:41][D][sensor:093]: 'Compressor frequency (Office)': Sending state 0.00000 Hz with 0 decimals of accuracy

echavet, 2024-02-06:
Yes SwiCago lib gets the data from the packet whether the feature is supported or not.
Defining the sensor or not only advertise a compressor frequency Homeassistant sensor or not.
But you will still see the info in the logs.
At the end we are supposed to set the logs off, for production purposes.