echavet/MitsubishiCN105ESPHome #22: Remote temp timeout not firing

phidauex, 2024-02-03:
Hi - great job on this component, loving all the improvements!

However, I'm not getting the remote temp timeout to work, running a Wemos D1 mini (ESP8266), on a MSZ-GL15NA, last built today with no errors. All other functions working as expected, including updating remote temp through an automation which calls the `set_remote_temperature` service.

**ESPHome config (relevant sections):**

```
esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  
esp8266:
  board: d1_mini

external_components:
  - source: github://echavet/MitsubishiCN105ESPHome

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3

# Enable logging
logger:
  hardware_uart: UART1
  level: DEBUG
  logs:
    EVT_SETS : DEBUG
    WIFI : INFO
    MQTT : INFO
    WRITE_SETTINGS : DEBUG
    SETTINGS : DEBUG
    STATUS : INFO
    CN105Climate: WARN
    CN105: DEBUG
    climate: WARN
    sensor: WARN
    chkSum : INFO
    WRITE : WARN
    READ : WARN
    Header: INFO
    Decoder : DEBUG
    CONTROL_WANTED_SETTINGS: DEBUG

api:
  encryption:
    key: !secret api_key
  services:
    - service: set_remote_temperature
      variables:
        temperature: float
      then:
# Select between the C version and the F version
#        - lambda: 'id(hp).set_remote_temperature(temperature);'
        - lambda: 'id(hp).set_remote_temperature((temperature - 32.0) * (5.0 / 9.0));'
    - service: use_internal_temperature
      then:
        - lambda: 'id(hp).set_remote_temperature(0);'

climate:
  - platform: cn105
    id: hp
    name: "${friendly_name}"
    icon: mdi:heat-pump
    visual:
      min_temperature: 15
      max_temperature: 31
      temperature_step: 1.0
    vertical_vane_select:
      name: ${name} Vertical Vane
    remote_temperature_timeout: 10min
    update_interval: 5s
```

Log excerpt showing that the remote temperature has been set. At this point I disabled the update automation so no more updates would be sent.

```
[12:55:02][D][CN105:322]: creating Info packet
[12:55:02][D][CN105:081]: writing packet...
[12:55:02][D][Decoder:203]: [0x06 is status]
[12:55:02][D][CN105:372]: sending remote temperature packet...
[12:55:02][D][CN105:081]: writing packet...
[12:55:02][I][CN105:275]: Last heatpump data update successful!
[12:55:02][I][CN105:287]: And it was a setExternalTemperature() ACK!
[12:55:02][D][EVT_SETS:378]: External C° update success
```

**Expected Result:** At time 13:05:02, log entry showing `Remote temperature timeout occured, fall back to internal temperature!` and reset to internal temp sensor. But while the normal settings updates continue to flow, there is no more mention of the remote temp, and the unit does not fall back to internal temperature. Ran for another 20 minutes to check, but logs remain normal, and remote temp remains at the last remote set value.

Is there an obvious configuration issue? Another log entry I should check for in troubleshooting?

echavet, 2024-02-05:
Hello,
thanks for feedback. I could not test it yet but there was an issue in the climate.py preventing the timeout from being set correctly. 
I think it's good now. 
Sorry I didn't test seriously this feature.

phidauex, 2024-02-05:
No apologies needed, you are already helping tremendously!

That correction seems to have fixed the behavior. Here is the timeout occurring, 10 minutes after I disable the update automation:
```
[09:48:56][W][EVT_SETS:083]: Remote temperature timeout occured, fall back to internal temperature!
[09:48:56][D][CN105:373]: sending remote temperature packet...
[09:48:56][D][CN105:081]: writing packet...
[09:48:56][I][CN105:275]: Last heatpump data update successful!
[09:48:56][I][CN105:287]: And it was a setExternalTemperature() ACK!
[09:48:56][D][EVT_SETS:378]: External C° update success
```

The unit correctly reset from 25C (temp sensor in my pocket) to 19.5C internal measurement.

Then, a few minutes later after enabling the automation:
```
[09:53:02][D][CN105:373]: sending remote temperature packet...
[09:53:02][D][CN105:081]: writing packet...
[09:53:02][I][CN105:275]: Last heatpump data update successful!
[09:53:02][I][CN105:287]: And it was a setExternalTemperature() ACK!
[09:53:02][D][EVT_SETS:378]: External C° update success
```

The unit updates again, now to the 21C measurement from the remote sensor. In both cases the change was reflected correctly in the HA climate card.

Thanks! This is a great feature, because one of my fears running this remote update method for the last 9 months or so is that my HA instance would die, and the heating system would either get locked in the on position, or worse, locked off, because it never got another remote update. My automation protects against dead sensors (it disables remote updates if the sensor is unavailable or unknown), but I didn't have any protection from a UPS failure, server failure, etc.

To help out with the project I'll spend some energy on the documentation and submit a pull request - I'm not the best programmer, but I'm pretty handy with documents and examples.

closed 2024-02-05 by phidauex