echavet/MitsubishiCN105ESPHome #24: checkPendingWantedSettings- SHOULDN'T HAPPEN

MarkoPaasila, 2024-02-12:
This software is already really good. I just updated to latest version and like it very much!

It seems to work properly, but I get this warning: ``` checkPendingWantedSettings- SHOULDN'T HAPPEN :  this->wantedSettings.hasChanged is true but this->currentSettings == this->wantedSettings,```

Do you have an idea of why this is happening anyway?

By the way, I like the log messages. They are nice and explanatory :-)

echavet, 2024-02-16:
Thanks for your message.
Yes I am currently working on improving the functioning, particularly in terms of concurrent access between the control() method called by home assistant, the main loop() and the recurring calls at the update interval.
As long as you don't get flooded with recurring logs messages and the overall operation remains coherent, there's no need to worry.

echavet, 2024-02-21:
with the later update, head of the main branch, this cannot happen anymore. 
But this is currently in pre-release. Needs more tests.

closed 2024-02-22 by echavet