echavet/MitsubishiCN105ESPHome #52: Warning in logs, "current stage for delivery caution: value 6 not found, returning value at index 0"

enomam, 2024-02-27:
Hey all,

I recently ran an update (ran this today), and am getting a lot of these warnings on my MSZAP20VG:

**[21:15:48][W][lookup:191]: current stage for delivery caution: value 6 not found, returning value at index 0**

I've attached a log snippet.
Relevent yaml also attached

```
[21:15:42][I][CYCLE:038]: 1: Cycle start
[21:15:42][D][CYCLE:314]: 2a: Sending settings request (0x02)
[21:15:43][D][CYCLE:265]: 2b: Receiving settings response
[21:15:43][I][CN105:485]: widevane setting (extra select component) changed
[21:15:43][D][select:015]: 'esphome-web-f53416 Horizontal Vane': Sending state | (index 2)
[21:15:43][D][CYCLE:268]: 3a: Sending room °C request (0x03)
[21:15:43][D][CYCLE:274]: 3b: Receiving room °C response
[21:15:43][D][CYCLE:277]: 4a: Sending status request (0x06)
[21:15:43][D][CYCLE:295]: 4b: Receiving status response
[21:15:43][D][CYCLE:297]: 5a: Sending power request (0x09)
[21:15:43][D][CYCLE:303]: 5b: Receiving Power/Standby response
[21:15:43][W][lookup:191]: current stage for delivery caution: value 6 not found, returning value at index 0
[21:15:43][D][text_sensor:064]: 'Stage Sensor': Sending state 'STAGE 0'
[21:15:43][D][text_sensor:064]: 'Sub Mode Sensor': Sending state 'NORMAL'
[21:15:43][D][text_sensor:064]: 'Auto Sub Mode Sensor': Sending state 'AUTO_OFF'
[21:15:43][I][CYCLE:051]: 6: Cycle ended in 0.9 seconds (with timeout?:  NO)
[21:15:47][D][UPDT_ITVL:045]: triggering infopacket because of update interval tick
[21:15:47][I][CYCLE:038]: 1: Cycle start
[21:15:47][D][CYCLE:314]: 2a: Sending settings request (0x02)
[21:15:47][D][CYCLE:265]: 2b: Receiving settings response
[21:15:47][I][CN105:485]: widevane setting (extra select component) changed
[21:15:47][D][select:015]: 'esphome-web-f53416 Horizontal Vane': Sending state | (index 2)
[21:15:47][D][CYCLE:268]: 3a: Sending room °C request (0x03)
[21:15:48][D][CYCLE:274]: 3b: Receiving room °C response
[21:15:48][D][CYCLE:277]: 4a: Sending status request (0x06)
[21:15:48][D][CYCLE:295]: 4b: Receiving status response
[21:15:48][D][CYCLE:297]: 5a: Sending power request (0x09)
[21:15:48][D][CYCLE:303]: 5b: Receiving Power/Standby response
[21:15:48][W][lookup:191]: current stage for delivery caution: value 6 not found, returning value at index 0
[21:15:48][D][text_sensor:064]: 'Stage Sensor': Sending state 'STAGE 0'
[21:15:48][D][text_sensor:064]: 'Sub Mode Sensor': Sending state 'NORMAL'
[21:15:48][D][text_sensor:064]: 'Auto Sub Mode Sensor': Sending state 'AUTO_OFF'
[21:15:48][I][CYCLE:051]: 6: Cycle ended in 0.9 seconds (with timeout?:  NO)
[21:15:52][D][UPDT_ITVL:045]: triggering infopacket because of update interval tick
[21:15:52][I][CYCLE:038]: 1: Cycle start
[21:15:52][D][CYCLE:314]: 2a: Sending settings request (0x02)
[21:15:52][D][CYCLE:265]: 2b: Receiving settings response
[21:15:52][I][CN105:485]: widevane setting (extra select component) changed
[21:15:52][D][select:015]: 'esphome-web-f53416 Horizontal Vane': Sending state | (index 2)
[21:15:52][D][CYCLE:268]: 3a: Sending room °C request (0x03)
[21:15:53][D][CYCLE:274]: 3b: Receiving room °C response
[21:15:53][D][CYCLE:277]: 4a: Sending status request (0x06)
[21:15:53][D][CYCLE:295]: 4b: Receiving status response
[21:15:53][D][CYCLE:297]: 5a: Sending power request (0x09)
[21:15:53][D][CYCLE:303]: 5b: Receiving Power/Standby response
[21:15:53][W][lookup:191]: current stage for delivery caution: value 6 not found, returning value at index 0
[21:15:53][D][text_sensor:064]: 'Stage Sensor': Sending state 'STAGE 0'
[21:15:53][D][text_sensor:064]: 'Sub Mode Sensor': Sending state 'NORMAL'
[21:15:53][D][text_sensor:064]: 'Auto Sub Mode Sensor': Sending state 'AUTO_OFF'
[21:15:53][I][CYCLE:051]: 6: Cycle ended in 0.9 seconds (with timeout?:  NO)
```

YAML
```
climate:
  - platform: cn105  # remplis avec la plateforme de ton choix
    name: ${friendly_name}
    id: master_ac
    update_interval: 4s         # shouldn't be less than 1 second
    horizontal_vane_select:
       name: ${name} Horizontal Vane
       id: clim_sejour_horizontal_vane
    vertical_vane_select:
      name: ${friendly_name} Vertical Vane
      id: study_ac_vertical_vane
    compressor_frequency_sensor:
      name: ${friendly_name} Compressor Frequency
      id: study_ac_compressor_frequency
```

echavet, 2024-02-27:
Hi and thanks for your message.
This is an experimental sensor (stage) whose operation has nothing to do with that of the air conditioning. So, no worries about that, and thanks again for the feedback. We will see what it's about.

enomam, 2024-02-28:
For context - this warning only comes up when I've got the indoor unit in "quiet" or "diffuse" mode. It goes away when i switch to Auto

echavet, 2024-02-28:
good info, I'll have to see that with @disruptivepatternmaterial 
he's the one who deals with this stage sensor.
It looks like the STAGE_MAP is missing a value:
```
static const uint8_t STAGE[6] = { 0x00, 0x01, 0x02, 0x03, 0x04, 0x05 };
static const char* STAGE_MAP[6] = { "STAGE 0", "STAGE 1", "STAGE 2", "STAGE 3", "STAGE 4", "STAGE 5" };
```

phidauex, 2024-02-28:
I just ran a quick test on one of my indoor units (MSZ-GL12NA), using both HA climate fan modes and the remote control. Here is what I get. I added a speed ranking which is my impression of the actual speed of the fan relative to other settings. Recommended Name is just brainstorming some more human friendly values.

Thinking about the stage sensor in general, I think "Fan Stage" would be a good ongoing sensor name, which would match well with the Fan Mode entity.

| Stage Code  | Stage Name | Actual Speed Ranking | HA Climate Name | Remote Control Setting | Note | Recommended Name |
| ------------- | ------------- | ------------- | ------------- | ------------- | ------------ | ------------ |
| 0x0 | STAGE 0  | 0 | none  | none  | Not directly accessible, used by Auto when idle | 0 - Idle |
| 0x1 | STAGE 1  | 1 | none  | none  | Not directly accessible, used by Auto when heating/cooling | 1 - Very Low |
| 0x2 | STAGE 2  | 2 | Low | One bar | | 2 - Low |
| 0x3 | STAGE 3  | 3 | Medium  | Two bars  | | 3 - Medium |
| 0x4 | STAGE 4  | 4 | Middle  | Three bars | This one is "out of order" in the HA climate options | 4 - ??? |
| 0x5 | STAGE 5  | 5 | High  | Four bars  | | 5 - High |
| 0x6 | STAGE 6  | 0 | Quiet | Quiet (ears symbol)  | Need to add to the stage map | 6 - Quiet |

echavet, 2024-02-29:
[It could be Idle, Low, Gentle, Medium, Moderate, High and quiet](https://github.com/echavet/MitsubishiCN105ESPHome/issues/19#issuecomment-1970610167)

closed 2024-02-29 by echavet

echavet, 2024-02-29:
"I think 'Diffuse' is a better choice here, as it aligns with the internal logic of the module in managing the setpoints specified by the user. Based on @disruptivepatternmaterial's explanation of these modes, they act as internal selections made by the module to best fulfill the user's setpoints, taking into account factors such as external temperature, the state of the compressor, and the demands of other units. 'Quiet' might be a user's preference, but it doesn't necessarily correspond to the unit's operation. 'Diffuse' seems more fitting as it better represents the unit's nuanced response to these setpoints.
```
static const uint8_t STAGE[7] = { 0x00, 0x01, 0x02, 0x03, 0x04, 0x05, 0x06 };
static const char* STAGE_MAP[7] = { "IDLE", "LOW", "GENTLE", "MEDIUM", "MODERATE", "HIGH", "DIFFUSE" };
```

phidauex, 2024-02-29:
To make things even more confusing, here is what Mitsubishi calls the fan speeds for the GL09 to GL18 units, as well as the GL24 units, which are different. The smaller units use: Quiet, Low, Medium, High, Super High. Largest unit uses: Low, Medium, High, Super High, Powerful.

I have no idea how these map to the STAGE codes - IE, is 0x2 "Low" on some units, and "Medium" on others? Or does 0x6 give Powerful on some units and Quiet on others? May need more Kumo cloud dumps...

[image]

disruptivepatternmaterial, 2024-03-01:
I am still away from home - will be home early next week - what I should do it list the exact model numbers I have and then that helps in the understanding of the diversity of the equipment.

I also have a Saleae Logic and my plan is to put the kumo adapter back on one of my units, change the settings, and then dump the JSON from the adapter while looking at the bytes coming across. Then I hope to do a couple things:

1 - get some idea on how to change some of the important settings you can change in the kumo
2 - make sure my understanding of the extra bytes I have shared info on is really the real story.

#2 might help with some of your comments @phidauex to make sure we are all on the right track.

PR #54 (2024-02-29): Warning in logs, "current stage for delivery caution: value 6 not found, returning value at index 0"
Fixes #52
Covers #19
commit 483f58cc (2024-02-29): Warning in logs, "current stage for delivery caution: value 6 not found, returning value at index 0"
Fixes #52
Covers #19