echavet/MitsubishiCN105ESPHome #53: Occasional failure to connect to heatpump on reboot

phidauex, 2024-02-28:
Sometimes, after either rebooting or updating the installation on the D1 Mini, the device will fail to connect to the heatpump, and will generate a large number of log entries (below). The state persists until I reset the D1 device (typically through the `restart` button I add to my configurations). On restart, the unit will usually work again. In one case I had to restart twice.

```
16:05:11 | [E] | [CN105:317] | sync impossible: heatpump not connected
16:05:11 | [E] | [CN105:317] | sync impossible: heatpump not connected
16:05:11 | [E] | [CN105:317] | sync impossible: heatpump not connected
16:05:11 | [E] | [CN105:317] | sync impossible: heatpump not connected
16:05:11 | [E] | [CN105:317] | sync impossible: heatpump not connected
```

Unsure what is causing it yet, but it might be an opportunity for an error handling routine where if there is a failure to connect, the unit will wait before retrying.

echavet, 2024-02-29:
Thanks for your message. Will have a look...

In the previous version (with exclusive esp8266 framework) UART connection
was easier to manage. With the esphome abstraction which allows to use both
esp-idf and esp8266 Arduino framework I'm not sure I will find a way.
Perhaps by forcing soft reboot....

echavet, 2024-02-29:
I'we had a `sendFirstConnectionPacket()` call to try again to connect to the heatpump when the error occurs.

PR #55 (2024-02-29): Occasional failure to connect to heatpump on reboot
Hope it Fixes #53 !
commit 876a8989 (2024-02-29): Occasional failure to connect to heatpump on reboot
Hope it Fixes #53 !
closed 2024-02-29 by echavet