echavet/MitsubishiCN105ESPHome #56: Logger spam if unable to connect to the heatpump

enomam, 2024-02-29:
Hey again,

I'm currently on the latest again.
Usually after an OTA update, the esp_01 doesn't connect to the heatpump - so I need to force a reboot on the esp_01.
I get a lot of log spam during this window:

```
INFO Upload took 5.21 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from esphome-web-15a184/debug
INFO Connected to MQTT broker!
[10:49:41][I][app:134]: Rebooting safely...
[10:50:01][I][mqtt:274]: MQTT Connected!
[10:50:01][I][CN105:023]: tx_pin: -1 rx_pin: -1
[10:50:01][I][CN105:024]: remote_temp_timeout is set to -1
[10:50:01][I][CN105:025]: debounce_delay is set to 500
[10:50:01][I][CN105:092]: setupUART() with baudrate 2400
[10:50:01][I][app:062]: setup() finished successfully!
[10:50:01][E][CN105:317]: sync impossible: heatpump not connected
[10:50:01][I][CN105:318]: Trying to connect again
[10:50:01][E][CN105:317]: sync impossible: heatpump not connected
[10:50:01][I][CN105:318]: Trying to connect again
[10:50:01][E][CN105:317]: sync impossible: heatpump not connected
[10:50:01][I][CN105:318]: Trying to connect again
[10:50:01][E][CN105:317]: sync impossible: heatpump not connected
[10:50:01][I][CN105:318]: Trying to connect again
[10:50:01][E][CN105:317]: sync impossible: heatpump not connected
[10:50:01][I][CN105:318]: Trying to connect again
[10:50:01][E][CN105:317]: sync impossible: heatpump not connected
...
...
...
[10:50:07][I][CN105:318]: Trying to connect again
[10:50:07][E][CN105:317]: sync impossible: heatpump not connected
[10:50:07][I][CN105:318]: Trying to connect again
[10:50:07][E][CN105:317]: sync impossible: heatpump not connected
[10:50:07][I][CN105:318]: Trying to connect again
[10:50:07][I][restart:016]: Restarting device...
[10:50:08][I][app:134]: Rebooting safely...
[10:50:16][I][mqtt:274]: MQTT Connected!
[10:50:16][I][CN105:023]: tx_pin: -1 rx_pin: -1
[10:50:16][I][CN105:024]: remote_temp_timeout is set to -1
[10:50:16][I][CN105:025]: debounce_delay is set to 500
[10:50:16][I][CN105:092]: setupUART() with baudrate 2400
[10:50:16][I][app:062]: setup() finished successfully!
[10:50:16][I][app:102]: ESPHome version 2023.12.9 compiled on Mar  1 2024, 10:49:30
[10:50:16][I][CN105:355]: --> Heatpump did reply: connection success! <--
[10:50:20][I][CYCLE:038]: 1: Cycle start
[10:50:20][I][CN105:528]: power or mode changed
[10:50:20][I][CN105:499]: fan setting changed
[10:50:20][I][CN105:445]: vane or widevane setting changed
[10:50:20][I][CN105:479]: vane setting (extra select component) changed
[10:50:20][I][CN105:485]: widevane setting (extra select component) changed
[10:50:20][I][STATUS:121]: [received       ]-> [room C°: 23.0, operating: NO , compressor freq: -1 Hz]
[10:50:20][I][STATUS:121]: [current        ]-> [room C°: 0.0, operating: NO , compressor freq: -1 Hz]
[10:50:21][I][STATUS:121]: [received       ]-> [room C°: 23.0, operating: YES, compressor freq:  0 Hz]
[10:50:21][I][STATUS:121]: [current        ]-> [room C°: 23.0, operating: NO , compressor freq: -1 Hz]
[10:50:21][I][CYCLE:051]: 6: Cycle ended in 0.9 seconds (with timeout?:  NO)
```

enomam, 2024-03-06:
In terms of debug though - any logging I can enable to provide some info?
And yes - really using an ESP_01. Wasn't keen on the hassle of the various d1 modules which may, or may not require cutting traces to work. The swicago gitter chat showed people had fewer hassles with the esp01

disruptivepatternmaterial, 2024-03-06:
> In terms of debug though - any logging I can enable to provide some info? And yes - really using an ESP_01. Wasn't keen on the hassle of the various d1 modules which may, or may not require cutting traces to work. The swicago gitter chat showed people had fewer hassles with the esp01

The m5stamp and the sled it comes with, plus a 5 min cable are rock solid. Gives me a RGB LED to help indicate situations. Plus I have a version that I am testing which allows me to add in env sensors and such. Also using it to to be a BT Proxy at the same time. Since I can pipe off 5v and 12v, plus access to AC in the cabinet...the situation is endless! These stamps are 9$ total. I realize this is more than an ESP_01, but I trust their QC way more than a random mess from aliexpress or something.

echavet, 2024-03-06:
I'm afraid network logs are a little useless during the setup process because they are not initialized soon enough 
In branch [echavet/issue56](https://github.com/echavet/MitsubishiCN105ESPHome/tree/echavet/issue56) I'm trying some things to fix this.
Could you test with this ?
```
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome@echavet/issue56
    refresh: 0s
```

enomam, 2024-03-08:
Sorry for the delay - I'm away for the weekend, but can test early next week.

enomam, 2024-03-11:
Tried on two heatpumps, and seemed to work!

INFO Uploading /config/.esphome/build/esphome-web-a77526/.pioenvs/esphome-web-a77526/firmware.bin (471424 bytes)
INFO Compressed to 324590 bytes
Uploading: [============================================================] 100% Done...

```
INFO Upload took 5.37 seconds, waiting for result...
INFO OTA successful
INFO Successfully uploaded program.
INFO Starting log output from esphome-web-a77526/debug
INFO Connected to MQTT broker!
[09:18:14][I][app:134]: Rebooting safely...
[09:18:34][I][mqtt:274]: MQTT Connected!
[09:18:34][I][CN105:023]: tx_pin: -1 rx_pin: -1
[09:18:34][I][CN105:024]: remote_temp_timeout is set to -1
[09:18:34][I][CN105:025]: debounce_delay is set to 500
[09:18:34][I][CN105:091]: setupUART() with baudrate 2400
[09:18:44][E][CN105:030]: --> Heatpump did not reply: NOT CONNECTED <--
[09:18:44][I][CN105:031]: Trying to connect again...
[09:18:44][I][CN105:355]: --> Heatpump did reply: connection success! <--
```

echavet, 2024-03-13:
I reopen because introduced an issue:
- loosing sync with heatpump due to bad management of connection loss

echavet, 2024-03-14:
Thank you for your feedback. I introduced an issue in the version you have tested: heatpump and component ends up loosing sync. 
So I've had to change again many things. As a consequence, your scenario needs to be tested again. But this time with the head main version.
Sorry again for that.

```yaml
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome
    refresh: 0s
```

enomam, 2024-03-15:
Looking good!
Just bulk updates 7 units, and all working well :)

commit 42a87184 (2024-03-12): Logger spam if unable to connect to the heatpump
Fixes #56
PR #60 (2024-03-12): Logger spam if unable to connect to the heatpump
Fixes #56
closed 2024-03-15 by echavet