echavet/MitsubishiCN105ESPHome #57: Compressor Frequency Not Displaying in HA

alan665, 2024-03-07:
Hey,

Love your work building on the geoffdavis project. Ive just hooked everything up using a D1 mini 3.0 and noticed I cant get the compressor frequency in HA.

[https://i.imgur.com/A3BIlxg.png](url)

Yaml:

```
substitutions:
  name: "aircon-alan-bedroom"
  friendly_name: "Aircon Alan Bedroom"

esphome:
  name: ${name}

esp8266:
  board: d1_mini

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

logger:
  hardware_uart: UART1
  level: INFO
  logs:
    EVT_SETS : INFO
    WIFI : INFO
    MQTT : INFO
    WRITE_SETTINGS : INFO
    SETTINGS : INFO
    STATUS : INFO
    CN105Climate: WARN
    CN105: INFO
    climate: WARN
    sensor: WARN
    chkSum : INFO
    WRITE : WARN
    READ : WARN
    Header: INFO
    Decoder : INFO
    CONTROL_WANTED_SETTINGS: INFO

api:

ota:

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3

external_components:
  - source: github://echavet/MitsubishiCN105ESPHome

climate:
  - platform: cn105
    id: hp
    name: "${friendly_name}"
    icon: mdi:heat-pump
    visual:
      min_temperature: 16
      max_temperature: 31
      temperature_step:
        target_temperature: 1
        current_temperature: 0.1
    compressor_frequency_sensor:
      name: ${name} Compressor Frequency
    vertical_vane_select:
      name: ${name} Vertical Vane
    horizontal_vane_select:
      name: ${name} Horizontal Vane
    isee_sensor:
      name: ${name} ISEE Sensor
    remote_temperature_timeout: 30min
    debounce_delay : 500ms
    update_interval: 2s

closed 2024-03-07 by echavet

alan665, 2024-03-07:
That seems to have partially worked. However I cant see the frequency as shown in your photo. Is there anything im missing?

[image]

phidauex, 2024-03-07:
I don’t think you are missing anything, it just seems from experimentation that the Compressor Frequency register is not supported by many units. I have an MSX-SM42NAMHZ outdoor unit and MSZ-GLxxNA indoor units, and they don’t give a compressor frequency value either (though sometimes it will go from 0 to -1, but I haven’t determined yet if that means anything).

As we delve deeper into the supported registers I suspect that more will be found that are only supported by some units, and will be blank or 0 for others.