echavet/MitsubishiCN105ESPHome #65: I-see sensor not working

echavet, 2024-03-21:
https://github.com/echavet/MitsubishiCN105ESPHome/discussions/64#discussioncomment-8858689

disruptivepatternmaterial, 2024-03-31:
my units with the iSee are a real pain to get to to put the kumo back on them and see what it says, but I can turn it on and off with the ir remote and see what comes out in the debug...would that help?

echavet, 2024-03-31:
I don't think it would help.
You can try of course but I think nothing is gonna change in the status packet. SwiCago would have noticed that.
The only way could be the kumo but this is not so important! (Not to me because I don't have any isee sensor!! 😂)

disruptivepatternmaterial, 2024-04-02:
The one I am using for testing is 2 heads on 1 outside unit, so I thought that was more interesting to see all the ways they share load and stay aligned.

But the one in my office is a higher end model, it has iSee and vanes in both directions. So I am interested in repeating my tests on that unit see what's up. It will just take me a bit longer because it is hard to get to. But I have to because the ESP32 I put in there is different than the other ones in my house and I want to make my life easier and have all my split on the same hardware. I have the new ESP32 (m5 stamp) and just need to find time to climb up there...so might as well see what the kumo says. But there HAS to be a way to turn iSee on and off, since you can do it with the remote, no?!?!

echavet, 2024-04-02:
Yes I think so. You already have discovered some new features. There might be some more remaining in the dark !!

drawks, 2024-05-21:
> The one I am using for testing is 2 heads on 1 outside unit, so I thought that was more interesting to see all the ways they share load and stay aligned.
>
> But the one in my office is a higher end [...]

I've also got a 4 way ceiling cartridge with an i-see sensor, I'll be climbing up in the attic to fit an esp to it this weekend. I don't actually own any remote with i-see controls on it, so I'm very interested in getting additional i-sse controls via this project. If you capture any interesting protocol data streams I'd be glad to see them and see if I can hack in additional support. As well as helping with test anything which is already implemented.

disruptivepatternmaterial, 2024-05-21:
So far I am not convinced (meaning have not found HAHAHA) that this can be set via the cn105...I have my hacking kit on the unit with iSee and I can observe the bits to tell me that iSee is on, but there appears to be nothing to tell it to turn off. Kind of annoying...because when it is on it really works to help control the units automatically in parts of the house I dont use that often.

But I'm still on the hunt!

So the only thing I think you can do is if have a real kumo adapter and are capable, hit it locally and are the json it returns. This would help me, after that you need to spend some cash on a logic analyzer, which unless you are super nerdy and do this for a living seems like a bit too much money to spend on this, but I leave that to you! HA!

drawks, 2024-05-21:
Well, I don't have a kumocloud adaptor, I do have a couple of aidoo/airzone adaptors which unfortunately do not seem to interact with the i-see sensor at all. I actually do have some logic analyzer hardware, but without any blackbox hardware to generate the desired data streams I'm left without anything to sniff

drawks, 2024-05-21:
To be clear, has anyone with a kumocloud adaptor confirmed that the isee sensor can be set to direct/indirect on/off from kumocloud? I've just read MANY Mitsubishi manuals and product catalog and AFAICT the only products which have documented isee controls connect via cn90 not cn105.

I've not found any data captures from cn90 comms online, but I'd guess it uses something more akin to the IR protocol. A single packet struct that indicates the entire system state that is applied or that is requested depending on which tx or rx line that it appears.

disruptivepatternmaterial, 2024-07-22:
@drawks and @echavet I have spend hours and hours trying to track this down and what is next is to sniff the ir from my remote and do some tracing along the motherboard because I'm with you. I do not think the iSee is controllable from kumo, or if it is in there somewhere it is not exposed to Kumo. *sigh*