echavet/MitsubishiCN105ESPHome #70: UART unavailable on D1 Mini (ESP8226) after ~12hrs

phidauex, 2024-04-04:
On one of my D1 Minis (ESP8226), after approximately 12 hours of operation (not exactly, but more than a few hours, and less than a day), the unit throws a write error in the CN105 component due to the UART not being connected. This then spams the logs with the below warnings, and no more updates are made or received. This is running the most current branch. A reboot of the D1 solves the problem temporarily. I have not caught it in the act of failing, so there may be an illuminating log message that I'm missing.

My other 4 units do not seem to be having this trouble, so maybe I just have a bad D1 mini, but I wanted to share the log to see if it triggered any ideas.

**Log Excerpt** - the "could not write as asked" line repeats every few seconds, with the checkPendingWantedSettings spamming between.

```
[11:44:49][W][CN105:092]: could not write as asked, because UART is not connected
[11:44:49][W][CN105:097]: delaying packet writing because we need to reconnect first...
[11:44:49][I][EVT_SETS:013]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[11:44:49][I][EVT_SETS:013]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[11:44:49][I][EVT_SETS:013]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[11:44:49][I][EVT_SETS:013]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[11:44:49][I][EVT_SETS:013]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[11:44:49][I][EVT_SETS:013]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[11:44:49][I][EVT_SETS:013]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[11:44:49][I][EVT_SETS:013]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[11:44:49][I][EVT_SETS:013]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
```

**UART and CN105 component configuration**

```
uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3

climate:
  - platform: cn105
    id: hp
    name: "${friendly_name}"
    icon: mdi:heat-pump
    visual:
      min_temperature: 15
      max_temperature: 31
      temperature_step:
        target_temperature: 1.0
        current_temperature: 0.1
#    compressor_frequency_sensor:
#      name: ${name} Compressor Frequency
    vertical_vane_select:
      name: ${name} Vertical Vane
#    horizontal_vane_select:
#      name: ${name} Horizontal Vane
#    isee_sensor:
#      name: ${name} ISEE Sensor
    remote_temperature_timeout: 30min
    update_interval: 4s
    debounce_delay : 500ms
    stage_sensor:
      name: ${name} Stage Sensor
    sub_mode_sensor:
      name: ${name} Sub Mode Sensor
    auto_sub_mode_sensor:
      name: ${name} Auto Sub Mode Sensor
```

echavet, 2024-04-05:
hello @phidauex,
can you confirm please that you are absolutely sure to have the latest version installed? 
When did you (last) flash your devices ?

phidauex, 2024-04-06:
I had last updated on April 1st, and re-pulled and updated again on April 3rd just in case.

echavet, 2024-04-07:
Ok sure you have the latest version.
I'm going to investigate what can be happening...

phidauex, 2024-04-09:
Issue happening again after a few days off - none of my other units are doing it so hardware issue is a real possibility. I do have a spare D1 Mini so if you don't see anything obvious in the code then I'll flash to a new unit and see how it works. I've also considered switching to an ESP32 device so I can have it run a bluetooth proxy as well!

As an interesting side note, this gives a very clear view of the improvement from remote temperature sensing. When it stops updating the indoor unit switches back to internal temperature source, and my modulation gets terrible.

Here is an image of two days of power readings - the arrows point to where I rebooted the ESP, and then when it stopped communicating with the UART again. Nice clean power modulation when it is running, then massive short cycling when it isn't! Proves the value of this project as a way to improve heatpump performance considerably.

[image]

echavet, 2024-04-10:
Hello Sam,
I don't know if it's a hardware issue but the component should not spam the log and should try to connect again. Failing that, it should perform a software reboot.
Thank you for your feedback, all of that is very interesting and helps improving the code.
I'm still working on a method to improve diagnostics in such situations. Additionally, I'm exploring ways to adapt the behavior of the component.

greg-flexidao, 2024-04-10:
I have exactly the same setup, config and... issue.
I have 3 controllers working with different units and only one has this issue. 
I will try to setup another controller and see if it fixes the issue. In the meantime I'm running it with debug level logs but I don't see much clues there. Will share before the next restart of that unit in ~12h :)

echavet, 2024-04-11:
> I have exactly the same setup, config and... issue. I have 3 controllers working with different units and only one has this issue. I will try to setup another controller and see if it fixes the [...]

It would be great to get the full log before it starts to spam.
In the meantime, I plan to add some diagnostic sensors to get the benefit of homeassistant history and statistics.

echavet, 2024-04-11:
Hi guys,
So I'we made a few changes so as to get some more info on this issue.
First I do not set the this->isUARTConnected_ to false when hp connection timeout occurs.

Then I've built a new sensor called hp_uptime_connection_sensor. It should help up to know when connection dropped down.

You can test this by pointing to the issue branch like this:
```yaml
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome@echavet/issue70
    refresh: 0s
```
then you can activate the new sensor by adding **hp_uptime_connection_sensor** option:
```yaml
climate:
  - platform: cn105  
    name: ${friendly_name}
    id: "esp32_clim"
    ...
    hp_uptime_connection_sensor:
      name: ${name} HP Uptime Connection
      update_interval: 30s
    ...

  supports:
      ...
```

I've also temporally added in my yaml some template sensors to monitor other values. <details>
<summary>
You can do so to add them in the diagnostic section of your device in HA. Of course, take care to change the climate device id (here esp32_clim) according to the one set in the climate section.
</summary>

```yaml
sensor:
  - platform: template
    name: "dg_uart_connected"
    entity_category: DIAGNOSTIC
    lambda: |-
      return (bool) id(esp32_clim).isUARTConnected_;
    update_interval: 30s

  - platform: template
    name: "dg_hp_connected"
    entity_category: DIAGNOSTIC
    lambda: |-
      return (bool) id(esp32_clim).isUARTConnected_;
    update_interval: 30s

  - platform: template
    name: "dg_complete_cycles"
    entity_category: DIAGNOSTIC
    accuracy_decimals: 0
    lambda: |-
      return (unsigned long) id(esp32_clim).nbCompleteCycles_;
    update_interval: 60s

  - platform: template
    name: "dg_total_cycles"
    accuracy_decimals: 0
    entity_category: DIAGNOSTIC
    lambda: |-
      return (unsigned long) id(esp32_clim).nbCycles_;
    update_interval: 60s
  - platform: template
    name: "dg_nb_hp_connections"
    accuracy_decimals: 0
    entity_category: DIAGNOSTIC
    lambda: |-
      return (unsigned int) id(esp32_clim).nbHeatpumpConnections_;
    update_interval: 60s

  - platform: template
    name: "dg_complete_cycles_percent"
    unit_of_measurement: "%"
    accuracy_decimals: 1
    entity_category: DIAGNOSTIC
    lambda: |-      
      unsigned long nbCompleteCycles = id(esp32_clim).nbCompleteCycles_;
      unsigned long nbCycles = id(esp32_clim).nbCycles_;
      if (nbCycles == 0) {
        return 0.0;
      }
      return (float) nbCompleteCycles / nbCycles * 100.0;
    update_interval: 60s
```
<details/>

phidauex, 2024-04-13:
Great work! I'm slightly gratified to see that someone else is experiencing the problem as well (a replacement D1 would be cheaper, but at least I know I'm not crazy).

I've implemented the new sensors as well and I'll see if I can catch it in the act over the next day or two.

mfinnz, 2024-04-16:
I'm also having UART connection issues with D1 Mini, although it has never worked. I raise the query here, as hopefully I can provide some input if needed and also to help others with similar issues. I have tried 2 D1 Minis, and they never connect, so I don't think it's a hardware issue.

There is a chance that it's because I'm just connected directly to the HP, but I have a version 2 D1 Mini, so it has the 5V pin. It seems some directly connected that people RX works, but with others it doesn't, so would like to know if there is a consensus if resistors are really needed or not with the V2 DI Mini?

I have used the original esphome-mitsubishiheatpump library to send commands to the HP and that works well, but I'm getting NaN on the temperature and the mode never changes, so I suspect I just have a RX problem, hence this library is smart enough to know things aren't quite right.

My config:

`substitutions:
  name: heatpump
  friendly_name: Heatpump

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  
# For ESP8266 Devices
esp8266:
  board: d1_mini

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3

# For ESP32 Devices
#esp32:
#  board: esp32doit-devkit-v1
#  framework:
#    type: esp-idf   
#
#uart:
#  id: HP_UART
#  baud_rate: 2400
#  tx_pin: GPIO17
#  rx_pin: GPIO16

external_components:
  - source: github://echavet/MitsubishiCN105ESPHome
#    refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${friendly_name} ESP"
    password: !secret wifi_fallback_password

captive_portal:

# Enable logging
logger:
  hardware_uart: UART1
  level: INFO
  logs:
    EVT_SETS : INFO
    WIFI : INFO
    MQTT : INFO
    WRITE_SETTINGS : INFO
    SETTINGS : INFO
    STATUS : INFO
    CN105Climate: WARN
    CN105: INFO
    climate: WARN
    sensor: WARN
    chkSum : INFO
    WRITE : WARN
[... 145 lines cut ...]
[13:04:50][W][CN105:138]: Heatpump has not replied for 48 s
[13:04:50][I][CN105:139]: We think Heatpump is not connected anymore..
[13:04:50][I][CN105:093]: setupUART() with baudrate 2400
[13:04:54][W][CN105:138]: Heatpump has not replied for 52 s
[13:04:54][I][CN105:139]: We think Heatpump is not connected anymore..
[13:04:54][I][CN105:093]: setupUART() with baudrate 2400
[13:04:58][W][CN105:138]: Heatpump has not replied for 56 s
[13:04:58][I][CN105:139]: We think Heatpump is not connected anymore..
[13:04:58][I][CN105:093]: setupUART() with baudrate 2400
[13:05:02][W][CN105:138]: Heatpump has not replied for 60 s
[13:05:02][I][CN105:139]: We think Heatpump is not connected anymore..
[13:05:02][I][CN105:093]: setupUART() with baudrate 2400
[13:05:06][W][CN105:138]: Heatpump has not replied for 64 s
[13:05:06][I][CN105:139]: We think Heatpump is not connected anymore..
[13:05:06][I][CN105:093]: setupUART() with baudrate 2400
[13:05:10][W][CN105:138]: Heatpump has not replied for 68 s
[13:05:10][I][CN105:139]: We think Heatpump is not connected anymore..

> 

2400 baud worked for the esphome-mitsubishiheatpump library so I'm pretty sure that is OK.

Interesting in anything else I can check, but I will get a 1k ohm resistor to try across the RX/G pin to see if that helps. Currently the voltage across RX and G is 3.3V, same for TX/G. I'm not sure if that's a good thing or not.

Thanks. It's great that this library is being updated and maintained and I'd really like to run this as it seems to be a great evolution of the original

echavet, 2024-04-16:
> Great work! I'm slightly gratified to see that someone else is experiencing the problem as well (a replacement D1 would be cheaper, but at least I know I'm not crazy).
> 
> I've implemented the new sensors as well and I'll see if I can catch it in the act over the next day or two.

Sure you're not crazy!!

I'm not very surprised because I've already met these weird situations where the controller is losing its connection to the hp. And I'm not very proud of the way I managed to deal with that.

When I decided to use the ESPhome UART abstraction (so as to support IDF)  I noticed that the uart begin() method had become out of my scope. The UART is provided "as is" by the esphome interfaces with no way to connect or disconnect. This was disrupting the previous implementation mechanism and I did not handle that well.
I knew someday I'd have to work to optimize that... the day has come !

Hello, 
It sounds familiar to me!! 
I think it's a voltage trouble. 

This component absolutely wants to get an answer from the heatpump. So you might be able to send commands but not to read the answer. In this case, no way to setup.

I'm using a regulator on my D1 mini.
Some units do not need any, some others won't work without. 

https://www.amazon.fr/dp/B072FTMS89?ref=ppx_yo2ov_dt_b_product_details&th=1

Or you can get an esp32 like the esp32s3 which will work with tx setup on gpio00 and Rx on gpio04.

echavet, 2024-04-16:
@phidauex : Do you still have troubles on one of your unit with the 
github://echavet/MitsubishiCN105ESPHome@echavet/issue70 version ?

phidauex, 2024-04-16:
Unfortunately (fortunately?) I have not had the UART disconnect since running the update to the version with the diagnostic datapoints. Good luck? Or because there was a relevant fix somewhere in this code or the ESPHome version (2024.3.1)? I'm not sure. I'll keep watching the logs and hope for a disconnect that we can learn from.

Curious if @bytniak has seen any disconnects since updating.

bytniak, 2024-04-17:
Same here. 5 days, with no single "faulty" behavior spotted again. 

In my case, this time is also correlated with the change of weather and for last 5 days, the "faulty" AC is in the OFF mode. Before that the "faulty" AC was the one working almost 24/7 on heat mode, while the others were not. If it happens again that might bring more light to the direction of the issue. @phidauex could be that your misbehaving unit was the one the most utilized (any continuous non OFF mode)?

echavet, 2024-04-18:
well I guess it is thanks to the change on this->isUARTConnected_ that I don't set to false anymore in this version. 
I think I can merge the branch.

phidauex, 2024-04-18:
@bytniak True, my misbehaving unit is my biggest indoor unit covering the largest room space, so it runs a lot more than the others. Things were warm earlier this week, but the last 24 hours have been cold and the unit has been running quite a bit, and still hasn't disconnected. Hopefully that is a sign that the `isUARTConnected` change actually did fix the issue. I'll keep all the diagnostic sensors running on this unit in case something happens again.

baldarn, 2024-04-19:
jumping in because I have this error

```
In file included from src/esphome.h:31,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/hp_readings.cpp:1:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/uptime_sensor.h: No such file or directory
    2 | #include "esphome/components/uptime/uptime_sensor.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

this is my conf

```
substitutions:
  device_name: little-room-split

esphome:
  name: ${device_name}
  platform: ESP8266
  board: d1_mini

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  domain: !secret wifi_domain

  ap:
    ssid: ${device_name} Hotspot
    password: !secret hotspot_password

api:
ota:
web_server:

external_components:
  - source: github://echavet/MitsubishiCN105ESPHome

climate:
  - platform: cn105
    name: Little room split
    update_interval: 10s

logger:
  hardware_uart: UART1
  level: INFO
```

echavet, 2024-04-19:
sorry @baldarn and thank you for your message.
This is due to the last update.
I will add the correction as soon as possible.
For now, try to add this to your `sensor:` section
```yaml
  - platform: uptime
    name: ${name} Uptime
```
this will force espHome to include needed classe `UptimeSensor`

echavet, 2024-04-19:
Just made the update in the climate.py file
sorry again

closed 2024-04-19 by echavet

mfinnz, 2024-04-19:
Thanks, I have ordered a regulator and will try that when it arrives to hopefully fix the UART connection issue.

Something you may want to look at though is that if I do send a command, not that I expect it to work, that the log file just shows a continuous loop of these errors. Of course it's expected to error, but wonder if a code change can be made so that it doesn't continuously loop and does give up and stop when the UART connection has failed? Just to prevent log files filling up.

```
10:18:49	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:49	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:49	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:49	[W]	[CN105:092]	
could not write as asked, because UART is not connected
10:18:49	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:49	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:50	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:50	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:50	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:50	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:50	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:50	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:50	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:51	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:51	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:51	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:51	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:51	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:51	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:51	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
10:18:51	[I]	[EVT_SETS:013]	
checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
```

Just a thought to evolve the code a bit more.

Thanks

echavet, 2024-04-20:
> Thanks, I have ordered a regulator and will try that when it arrives to hopefully fix the UART connection issue.
>
> Something you may want to look at though is that if I do send a command, not that I [...]

You’re right. It’s necessary to find a way to reduce log spam. I can simulate this situation by physically disconnecting the reception pin (rx pin). This roughly mirrors what is happening in your case. 

Also, I could allow the component to send commands to the heat pump even though it can't receive any responses. Some projects have chosen this approach, resulting in a component that operates only in transmission mode, similar to an infrared remote control. This is why so many users report 'NAN' errors when they receive no valid data in return. However, I have opted for an implementation that requires bidirectional communication. This allows for a sort of protocol to unfold in the sequencing of exchanges. I'm not too fond of the idea of sending commands like messages in bottles. Thanks anyway for your feedback. I will try to correct this behavior.

mfinnz, 2024-04-20:
> Also, I could allow the component to send commands to the heat pump even though it can't receive any responses. Some projects have chosen this approach, resulting in a component that operates only [...]

I completely agree with your approach which is why I'm hoping to get this working when the regulators arrive. I'm planning to use this setup even when I'm not at the house as it will be nice to pre-warm or pre-cool the room before getting home and knowing that the commands have worked will be key. Appreciate your work on this

echavet, 2024-04-24:
> Thanks, I have ordered a regulator and will try that when it arrives to hopefully fix the UART connection issue.
>
> Something you may want to look at though is that if I do send a command, not that I [...]

Hey @mfinnz,
did you install your regulator ?
If not, could you test this please,

```yaml
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome@echavet/issue74
    refresh: 0s
```
and tell me if you are ok with the behavior ?

mfinnz, 2024-05-01:
I had added a 3.3V regulator and I still can't connect. I will shift my issue update over to [https://github.com/echavet/MitsubishiCN105ESPHome/issues/85](url)

paravoid, 2024-05-01:
> I'm also having UART connection issues with D1 Mini, although it has never worked. I raise the query here, as hopefully I can provide some input if needed and also to help others with similar [...]

For what it's worth, I had the exact same experience, and fought a lot to make it work. I had quite a few ESP8266s lying around, and lots of people were saying that it works for them, so I was doubting my work and double-checking my cabling etc.

In the end, I replaced the D1 Mini with an AtomU (ESP32) and it worked immediately. In a second unit, I used a Lolin C3 Mini (ESP32-C3), and while it had other entirely unrelated issues that I had to work around (in short: needed `platformio_options: board_build.flash_mode: dio`, and `wifi: output_power: 8.5dB`), it also worked fine. Both are in my A/C units (MSZ-FD25VA) for a few weeks now, and work without any issues. I haven't tested them but I'd bet the ESP32-S2/ESP32-S3 chips would be similar too. I didn't use a level shifter with either of them and they haven't died yet. (Whether the ESP32 is 5V tolerant has been a matter of intense speculation for years.)

Simple boards with a chip from the ESP32 should cost < $5, so for me that's money well spent. YMMV :)

commit e23e2edb (2024-04-19): UART unavailable on D1 Mini (ESP8226) after ~12hrs
Fixes #70
missing include correction
PR #72 (2024-04-19): UART unavailable on D1 Mini (ESP8226) after ~12hrs
Fixes #70
missing include correction