echavet/MitsubishiCN105ESPHome #74: External temperature updates don't always persist on the unit

enomam, 2024-04-20:
Hey 👋 

I've been using an external zigbee temperature sensor as the thermostat, listening to mqtt events as the trigger.
The automation part looks like this:

```
mqtt:
  broker: 10.0.10.110

  on_json_message:
    topic: zigbee2mqtt/0x00158d0002b952ad
    then:
      - lambda: |-
          if (x.containsKey("temperature")){
            id(ac).set_remote_temperature(x["temperature"]);
          }
````

Sometimes, however, the new temperature does not seem to be set on the unit (the current temp value). Forcing a temp refresh (by publishing a new mqtt event) doesn't always update the value either. Interestingly though - if I do a platform restart on the ESP device - it picks up the correct temp, even though it didn't receive an mqtt event (unless subscribing to a topic gives you the last value ...).

The units are MSZ-AP20VG

Not sure what's the best way to test/debug this one - especially as its intermittent ...

akifbayram, 2024-04-21:
Try adding a heartbeat to send the temperature value periodically to the unit.

```
    filters:
      - heartbeat: 10s
```

enomam, 2024-04-21:
Where would that go? I can't add a filter to an mqtt json message

echavet, 2024-04-21:
Hi,

I'm not sure this will solve the issue because it might be something else I missed in the code. 
Perhaps the heatpump is expecting a regular temperature update. I have not noticed such a behaviour but I could have missed that too!

However I'm also using an external ZigBee temperature sensor through ZigBee2mqtt.
It is then known as a standard sensor in Homeassistant and can be imported into esphome this way:

,,,yaml
  - platform: homeassistant
    id: ha_cdeg_sejour_et_cuisine
    entity_id: sensor.cdeg_sejour_et_cuisine
    internal: true
    on_value:
      then:
        - lambda: |-
            id(use_your_heatpump_id).set_remote_temperature(x);
'''
Like @akifbayram told you can then add a filter on this imported sensor.

Look at here for inspiration:
https://github.com/echavet/MitsubishiCN105ESPHome/wiki/Configure-a-timeout-to-fallback-into-using-the-internal-temperature-sensor

enomam, 2024-04-22:
Yeah - I'll try that.
The preference was to avoid relying on HA though - hence using mqtt.

I'll try it on one of my units though - see how it goes.
In the meantime though - is there any debug I could enable (there are lots of logging/debug options - so not sure which to use :) )

echavet, 2024-04-22:
Well, logs are not really structured by features (rather by functions or
location). But I'll have a look if I can tune them specifically so as to
exclude or not the component responsibility in your troubles. External
temperature is a key feature. We need to be sure it works well.

For MQTT I understand but I'm sure your Matt server is a ha module. And
ZigBee2mqtt too. So it might be a good idea to separate the concerns but
actually, if they all depend on HA there shouldn't be many trouble
tolerance différences.

I might be wrong but I think the platform:homeassistant section allows to
have a stronger integration than with MQTT. Proof is made in your case as
it looks like with MQTT you cannot use sensor filters.

enomam, 2024-04-24:
Hey again.

I did some additional debug. In the sensor yaml, I added the following:
```
- platform: homeassistant
    id: study_ha
    entity_id: sensor.0x00158d0002e6ea83_temperature
    internal: true
    filters:
      - heartbeat: 60s
    on_value:
      then:
        - lambda: |-
            ESP_LOGI("custom", "The value of sensor is: %f", x);
            id(study_ac).set_remote_temperature(x);
```

And I observed this in the logs a few times (referring to the cycle timeout) ...
```
[20:37:17][I][CYCLE:038]: 1: Cycle start
[20:37:17][I][custom:101]: The value of sensor is: 18.809999
[20:37:17][I][CN105:494]: widevane setting (extra select component) changed
[20:37:26][W][CN105:009]: Cycle timeout, reseting cycle...
[20:37:26][I][CYCLE:051]: 6: Cycle ended in 9.0 seconds (with timeout?: YES)
```
If the cycle was successful, the message is
```
[20:38:17][I][CN105:347]: Last heatpump data update successful!
```

echavet, 2024-04-24:
Hey!
could you test this please,

```yaml
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome@echavet/issue74
    refresh: 0s
```

and tell me if you are ok with the behavior ?

enomam, 2024-04-27:
I've captured more debug when using the HA sensor, rather that mqtt. Still not seeing the temperature update on the indoor unit. I've taken a screen from HomeAssistant too. It shows the temperature sensor at 19.5, but the aircon is still saying 19.0.

```
[14:33:36][I][CYCLE:038]: 1: Cycle start
[14:33:36][D][CYCLE:313]: 2a: Sending settings request (0x02)
[14:33:36][D][CN105:325]: creating Info packet
[14:33:36][D][CN105:084]: writing packet...
[14:33:37][D][CYCLE:265]: 2b: Receiving settings response
[14:33:37][D][CN105:267]: Climate mode is: 0
[14:33:37][D][CN105:268]: Climate action is: 0
[14:33:37][I][CN105:494]: widevane setting (extra select component) changed
[14:33:37][D][select:015]: 'esphome-web-0f8818 Horizontal Vane': Sending state | (index 2)
[14:33:37][D][CYCLE:268]: 3a: Sending room °C request (0x03)
[14:33:37][D][CN105:325]: creating Info packet
[14:33:37][D][CN105:084]: writing packet...
[14:33:37][D][CYCLE:274]: 3b: Receiving room °C response
[14:33:37][D][CYCLE:277]: 4a: Sending status request (0x06)
[14:33:37][D][CN105:325]: creating Info packet
[14:33:37][D][CN105:084]: writing packet...
[14:33:37][D][CYCLE:295]: 4b: Receiving status response
[14:33:37][D][CYCLE:297]: 5a: Sending power request (0x09)
[14:33:37][D][CN105:325]: creating Info packet
[14:33:37][D][CN105:084]: writing packet...
[14:33:37][D][CYCLE:303]: 5b: Receiving Power/Standby response
[14:33:37][D][text_sensor:064]: 'Oscar AC - ESP01 Stage Sensor': Sending state 'IDLE'
[14:33:37][D][text_sensor:064]: 'Oscar AC - ESP01 Sub Mode Sensor': Sending state 'NORMAL'
[14:33:37][D][text_sensor:064]: 'Oscar AC - ESP01 Auto Sub Mode Sensor': Sending state 'AUTO_OFF'
[14:33:37][I][CYCLE:051]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)
[14:33:41][D][CN105:310]: sending a request for settings packet (0x02)
[14:33:41][I][CYCLE:038]: 1: Cycle start
[14:33:41][D][CYCLE:313]: 2a: Sending settings request (0x02)
[14:33:41][D][CN105:325]: creating Info packet
[14:33:41][D][CN105:084]: writing packet...
[14:33:42][D][CYCLE:265]: 2b: Receiving settings response
[14:33:42][D][CN105:267]: Climate mode is: 0
[14:33:42][D][CN105:268]: Climate action is: 0
[14:33:42][I][CN105:494]: widevane setting (extra select component) changed
[14:33:42][D][select:015]: 'esphome-web-0f8818 Horizontal Vane': Sending state | (index 2)
[14:33:42][D][CYCLE:268]: 3a: Sending room °C request (0x03)
[14:33:42][D][CN105:325]: creating Info packet
[14:33:42][D][CN105:084]: writing packet...
[14:33:42][D][CYCLE:274]: 3b: Receiving room °C response
[14:33:42][D][CYCLE:277]: 4a: Sending status request (0x06)
[14:33:42][D][CN105:325]: creating Info packet
[14:33:42][D][CN105:084]: writing packet...
[14:33:42][D][CYCLE:295]: 4b: Receiving status response
[14:33:42][D][CYCLE:297]: 5a: Sending power request (0x09)
[14:33:42][D][CN105:325]: creating Info packet
[14:33:42][D][CN105:084]: writing packet...
[14:33:42][D][CYCLE:303]: 5b: Receiving Power/Standby response
[14:33:42][D][text_sensor:064]: 'Oscar AC - ESP01 Stage Sensor': Sending state 'IDLE'
[14:33:42][D][text_sensor:064]: 'Oscar AC - ESP01 Sub Mode Sensor': Sending state 'NORMAL'
[14:33:42][D][text_sensor:064]: 'Oscar AC - ESP01 Auto Sub Mode Sensor': Sending state 'AUTO_OFF'
[14:33:42][I][CYCLE:051]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)
[14:33:44][D][sensor:093]: 'study_ha': Sending state 19.49000  with 1 decimals of accuracy
[14:33:44][I][custom:138]: The value of sensor is: 19.490000
[14:33:44][D][CN105:375]: sending remote temperature packet...
[14:33:44][D][CN105:084]: writing packet...
[14:33:44][D][Update-ACK:157]: FC 61 01 30 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5E 
[14:33:44][I][CN105:347]: Last heatpump data update successful!
[14:33:46][D][CN105:310]: sending a request for settings packet (0x02)
[14:33:46][I][CYCLE:038]: 1: Cycle start
[14:33:46][D][CYCLE:313]: 2a: Sending settings request (0x02)
[14:33:46][D][CN105:325]: creating Info packet
[14:33:46][D][CN105:084]: writing packet...
[14:33:47][D][CYCLE:265]: 2b: Receiving settings response
[14:33:47][D][CN105:267]: Climate mode is: 0
[14:33:47][D][CN105:268]: Climate action is: 0
[14:33:47][I][CN105:494]: widevane setting (extra select component) changed
[14:33:47][D][select:015]: 'esphome-web-0f8818 Horizontal Vane': Sending state | (index 2)
[14:33:47][D][CYCLE:268]: 3a: Sending room °C request (0x03)
[14:33:47][D][CN105:325]: creating Info packet
[14:33:47][D][CN105:084]: writing packet...
[14:33:47][D][CYCLE:274]: 3b: Receiving room °C response
[14:33:47][D][CYCLE:277]: 4a: Sending status request (0x06)
[14:33:47][D][CN105:325]: creating Info packet
[14:33:47][D][CN105:084]: writing packet...
[14:33:47][D][CYCLE:295]: 4b: Receiving status response
[14:33:47][D][CYCLE:297]: 5a: Sending power request (0x09)
[14:33:47][D][CN105:325]: creating Info packet
[14:33:47][D][CN105:084]: writing packet...
[14:33:47][D][CYCLE:303]: 5b: Receiving Power/Standby response
[14:33:47][D][text_sensor:064]: 'Oscar AC - ESP01 Stage Sensor': Sending state 'IDLE'
[14:33:47][D][text_sensor:064]: 'Oscar AC - ESP01 Sub Mode Sensor': Sending state 'NORMAL'
[14:33:47][D][text_sensor:064]: 'Oscar AC - ESP01 Auto Sub Mode Sensor': Sending state 'AUTO_OFF'
[14:33:47][I][CYCLE:051]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)
```
[image]

echavet, 2024-05-08:
> Hi,
> I make the same observation: sometimes the external temperature is not
> taken into account... by the climate entity.
> However, the pump takes into account the temperature value sent.
> I see this by activating the "decoder" debug:
>
> [16:48:03][D][CYCLE:274]: 3b: Receiving room °C response
> [16:48:03][D][Decoder:237]: [Room °C: 20.000000]
>
> 20°C is indeed the good value sent by the external sensor, but it is not
> displayed in the HA climate entity current temperature:
>
> [16:57:25][D][CN105:268]: Climate action is: 3
> [16:57:25][D][climate:396]: '${friendly_name}' - Sending state:
> [16:57:25][D][climate:399]: Mode: HEAT
> [16:57:25][D][climate:401]: Action: HEATING
> [16:57:25][D][climate:404]: Fan Mode: AUTO
> [16:57:25][D][climate:416]: Swing Mode: OFF
> [16:57:25][D][climate:419]: Current Temperature: 20.50°C
>
> I think the problem comes from updating the climate entity, which does not
> send HA the correct current room temperature information.
> A change in current room temperature does not always trigger an update of
> the HA climate component.
>
> I hope I'm not mistaken and that this can help identify the problem.
> I keep searching !

Thank you for your message.
I will have a look

echavet, 2024-05-09:
hey @enomam and @StartofCadelanne 
thank you for your analysis.

I may have found a potential reason in this temperature update issue.
In the `void CN105Climate::set_remote_temperature(float setting)` method, which is the one called when a temperature sensor update occurs, I was doing this:
 ```cpp
    // optimistic
    this->currentStatus.roomTemperature = setting;
```
instead of this:

```cpp
    // optimistic
    this->current_temperature = this->currentStatus.roomTemperature;
    this->publish_state();
```
resulting in hiding the temperature change in the status returned by the heatpump endin with no HA component update.

But I think it is not worth to be "optimistic" and to force this temperature setting in this method. Just removing 
`this->currentStatus.roomTemperature = setting;` should allow the next status request to update the HA component.

Could you test my modification before I merge to the main branch please?

You just have to compile and upload this version and of course check the behavior. 

```yaml
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome@echavet/issue74
    refresh: 0s

```

StartofCadelanne, 2024-05-09:
Hi @echavet 
Thanks, 
Same idea yesterday evening, I commented the line. 
Already tested all the night and it's much better. 
I think a part of the problem is solve : no more discrepency between heat pump room temperature and HA climate entity.

Here the last 10 hours on 4 x MSZ-HRxxVF units equiped with esp32 D1 Mini NodeMCU:
[image]
"Defaut temp" shows inconsistency between external temperature sensor and HA climate entity feedback. (Studio = RdC)
(This is only relevant if there are temperature changes of more than 0.5°C.)

I see that some temperature changes are not taken into account, and need next step change to be applied.
So I think the pump miss some packets.

When are the temperatures sent? Is it at a fixed moment in the cycle or upon receipt?

echavet, 2024-05-09:
Oh wow it's cool !

Depending on the unit, I can see in the original implementation that either the heat pump supports float values with a precision of 0.1 or multiples of 0.5, which may explain what you noticed. In the latter case, the temperature cannot be consistent between the two devices if they have different capabilities.

StartofCadelanne, 2024-05-09:
You're right. My heat pumps deal with 0.5 °C increments.
I explained badly in my previous post. What I wanted to say will be clearer in the following example that i found this afternoon :
[image]
Mitshu-Salon unit and Mitshu-Cuisine unit share the same external Zigbee temperature sensor (because both are located in the same big room).
"Defaut Temp Cuisine" and "Defaut Temp Salon" are HA custom sensors. They round the external temp sensor to 0.5 before compare with heat pump climate component to monitor the behavior.

When Zig ext. temp. sensor (in blue) reaches 20.3°C, Salon unit (in orange) catch the value and set from 20 to 20.5 °C 10 sec later, due to time cycle. OK, Very good !
But Cuisine unit (in purple) miss this value, waiting the following temperature step at 20.4 °C 15 min later.

StartofCadelanne, 2024-05-09:
I may have found something:
I diverted an external temperature sensor to be able to force its value manually for debugging (this saves me from waiting for the temperature to change).
Here is what I noticed:
When the temperature change occurs between 2 cycles, everything is fine 👍 :

```
[22:56:09][I][CYCLE:052]: 6: Cycle ended in 1.2 seconds (with timeout?:  NO)
[22:56:11][D][homeassistant.sensor:024]: 'sensor.zig_temp_studio_temperature': Got state 21.70
[22:56:11][D][sensor:094]: 'temp_sensor_stud': Sending state 21.70000  with 1 decimals of accuracy
[22:56:11][D][CN105:375]: sending remote temperature packet...
[22:56:11][D][CN105:084]: writing packet...
[22:56:11][D][WRITE:157]: FC 41 01 30 10 07 01 1A AB 00 00 00 00 00 00 00 00 00 00 00 00 B1 
[22:56:12][D][Header:091]: command: (61) data length: [10]<-- header
[22:56:12][D][READ:157]: FC 61 01 30 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5E 
[22:56:12][D][chkSum:077]: OK-> 5E=5E 
[22:56:12][D][Update-ACK:157]: FC 61 01 30 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5E 
[22:56:12][I][CN105:347]: Last heatpump data update successful!
[22:56:16][D][CN105:310]: sending a request for settings packet (0x02)
[22:56:16][I][CYCLE:038]: 1: Cycle start
```

But when the temperature changes during the cycle, for example, this is what happens:

```
[22:34:33][D][CYCLE:277]: 4a: Sending status request (0x06)
[22:34:33][D][CN105:325]: creating Info packet
[22:34:33][D][CN105:084]: writing packet...
[22:34:33][D][WRITE:157]: FC 42 01 30 10 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 77 
[22:34:33][D][homeassistant.sensor:024]: 'sensor.zig_temp_studio_temperature': Got state 21.70
[22:34:33][D][sensor:094]: 'temp_sensor_stud': Sending state 21.70000  with 1 decimals of accuracy
[22:34:33][D][CN105:375]: sending remote temperature packet...
[22:34:33][D][CN105:084]: writing packet...
[22:34:33][D][WRITE:157]: FC 41 01 30 10 07 01 1A AB 00 00 00 00 00 00 00 00 00 00 00 00 B1 
[22:34:33][D][Header:091]: command: (62) data length: [10]<-- header
[22:34:33][D][READ:157]: FC 62 01 30 10 06 00 00 00 00 00 01 01 54 00 00 42 00 00 00 00 BF 
[22:34:33][D][chkSum:077]: OK-> BF=BF 
[22:34:33][D][CYCLE:295]: 4b: Receiving status response
[22:34:33][D][Decoder:248]: [0x06 is status]
```
It seems that the unit ignore the temperature packet because it's already dealing with a status request.

What do you think about this ? Is it possible to add a step into the cycle dedicated to external temperature update  ?

echavet, 2024-05-10:
Interesting. Thank you. That's a good log.
I need to study that ..

StartofCadelanne, 2024-05-10:
Hey ! Thanks for reopened this. Great !
Anyway, it will not be simple. As I understand, it souldn't be inside the cycle because the answer is not a data packet but a setting packet. 
May sent after the cycle like new settings, but how to differentiate new settings and new external temp ? Seems to be the same feedback...

echavet, 2024-05-11:
yes exactly, but it does not really matters as the status should finally return the ext temperature. 
I think the important thing is to send the packet with the right timing (probably outside a full cycle). 
I did like you've suggested to get the temperature from a manual input. I will debug that as soon as possible to understand what's happening. Because you right: it does not always update the unit

echavet, 2024-05-11:
There are 2 things to notice in this issue:
1. It takes time for the unit to take new temperature into account (about 5 to 30 seconds). So we shouldn't expect a change to be immediately visible on the UI component. Because we must wait for the unit to handle the value AND to return it in the next status response.
2. If you use a heartbeat filter like that:
```yaml
 - platform: homeassistant
    id: ha_cdeg_sejour_et_cuisine
    entity_id: sensor.temperature_sejour
    internal: true
    on_value:
      then:
        - lambda: |-
            id(esp32_clim).set_remote_temperature(x);
    filters:
      - heartbeat: 1min
```
You must note that the on_value won't occur more often than **1 time per minute** (`heartbeat: 1min`).

To be sure there is no collision in requests I will change the behavior to send remote temp only a the end of a full cycle.

echavet, 2024-05-11:
> Hey ! Thanks for reopened this. Great ! Anyway, it will not be simple. As I understand, it souldn't be inside the cycle because the answer is not a data packet but a setting packet. May sent after [...]

Actually I didn't want to close the issue. But when I commit a branch connected to this issue, github automatically closes the issue!

closed 2024-05-11 by echavet

StartofCadelanne, 2024-05-11:
Hi @echavet 
Wow ! Good Job !
Tested manualy with 1 unit :

- [x] external température change during a cycle --> Ok
- [x] external temperature change between two cycles --> Ok
- [x] both external temperature and new HP setting change in same time during a cycle -- OK
- [x] both external temperature and new HP setting change in same time between two cycles -- OK

Good interlocks, seems robust. No sync issue anymore.

Now updated in my 6 units ! Let's see over the night the results in real environment. I will post here tomorrow morning.
Congrats and thanks 👍

StartofCadelanne, 2024-05-12:
Results over the 16 last hours on 6 x MSZ-HR units:
[image]
Perfect !

commit 70a70d6d (2024-05-09): External temperature updates don't always persist on the unit
Fixes #74
commit 0276f693 (2024-05-11): External temperature updates don't always persist on the unit
Fixes #74