echavet/MitsubishiCN105ESPHome #85: Cannot connect to heatpump - tested with D1 mini,

HansVanEylen, 2024-05-01:
I've been trying to connect to my MSZ-AP25VGK (wall unit with built-in WiFi) using 

- a D1 mini (https://www.amazon.com.be/-/nl/dp/B0CJJ8L1DG?psc=1&ref=ppx_yo2ov_dt_b_product_details)
- an ESP32 ([MH-ET Live ESP32 Development Board WiFi+Bluetooth Ultra-Low Power Consumption Dual Core ESP-32 ESP-32S ESP 32])
- an Atom Echo (ESP32-Pico) I had lying around

The D1 mini was setup correctly in ESPHome (2024.4.2) but the logs showed

[E][CN105:032]: --> Heatpump did not reply: NOT CONNECTED <--
[I][CN105:033]: Trying to connect again...
[W][CN105:157]: Heatpump has not replied for 40 s
[I][CN105:158]: We think Heatpump is not connected anymore
[I][CN105:093]: setupUART() with baudrate 2400

Both ESP32 devices gave similar output:

[16:44:29][I][app:100]: ESPHome version 2024.4.2 compiled on May  1 2024, 16:31:16
[16:44:29][E][CN105:317]: sync impossible: heatpump not connected
[16:44:29][E][CN105:317]: sync impossible: heatpump not connected

but when I powered them on, the horizontal vane of the MSZ opened and closed, so there must be something going on. I tried three different baud rates : 2400, 4800 and 9600. All gave the same result.

Config for the D1 mini (I removed the wifi setup and api key info):

```
substitutions:
  name: acbureaud1mini
  friendly_name: AC Bureau D1 mini

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

esp8266:
  board: d1_mini

uart:
  id: HP_UART
  baud_rate: 9600
  tx_pin: 1
  rx_pin: 3

external_components:
  - source: github://echavet/MitsubishiCN105ESPHome
#    refresh: 0s

# Enable Web server.
web_server:
  port: 80

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

logger:
  hardware_uart: UART1
  level: INFO

# Create a button to restart the unit from HomeAssistant. Rarely needed, but can be handy.
button:
  - platform: restart
    name: "Restart ${friendly_name}"

climate:
  - platform: cn105
    name: "${friendly_name}"
    icon: mdi:heat-pump
    visual:
      min_temperature: 15
      max_temperature: 31
      temperature_step:
        target_temperature: 1
        current_temperature: 0.1
    compressor_frequency_sensor:
      name: ${name} Compressor Frequency
    vertical_vane_select:
      name: ${name} Vertical Vane
    horizontal_vane_select:
      name: ${name} Horizontal Vane
    remote_temperature_timeout: 30min
    debounce_delay : 500ms
    update_interval: 4s
```

Config for the ESP32 (in this case the Atom):

```
esphome:
  name: acbureau2
  friendly_name: ACBureau2

esp32:
  board: esp32doit-devkit-v1
  framework:
    type: esp-idf   

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: GPIO32
  rx_pin: GPIO26

external_components:
  - source: github://echavet/MitsubishiCN105ESPHome@v1.2.2

captive_portal:

# Enable Web server.
web_server:
  port: 80

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

climate:
  - platform: cn105
    name: "${friendly_name}"
    icon: mdi:heat-pump
    visual:
      min_temperature: 15
      max_temperature: 31
      temperature_step:
        target_temperature: 1
        current_temperature: 0.1
    compressor_frequency_sensor:
      name: ${name} Compressor Frequency
    vertical_vane_select:
      name: ${name} Vertical Vane
    horizontal_vane_select:
      name: ${name} Horizontal Vane
    remote_temperature_timeout: 30min
    debounce_delay : 500ms
    update_interval: 4s

logger:
    level: INFO
```

I did read https://github.com/echavet/MitsubishiCN105ESPHome/issues/70, so I will be getting a voltage regulator for the D1 mini, but I thought this wouldn't be necessary for the ESP32 devices?
Is there anything else I'm missing?

thanks,
Hans

mfinnz, 2024-05-01:
I'm having the same problem still after adding a 3.3V regulator to a D1 mini, as documented in #70 

I'm getting exactly the same result, it is getting no reply from the heatpump. If I use the code from GeoffDavis repository set to 2400 baud I can send commands and the heatpump responds so I know that is the correct baud rate and at least the TX line is working.

mfinnz, 2024-05-02:
I've just now added 10kOhm resistors as above, between the RX and TX lines to the 5V line, but no change in behaviour. On the D1 Mini I am measuring 3.56v across the RX pin and 3.3V across TX, which seems a little odd. I've triple checked my connections and it all seems correct.

echavet, 2024-05-02:
> I've just now added 10kOhm resistors as above, between the RX and TX lines to the 5V line, but no change in behaviour. On the D1 Mini I am measuring 3.56v across the RX pin and 3.3V across TX, which seems a little odd. I've triple checked my connections and it all seems correct. I've tried a third D1 Mini, but not luck, so I doubt the devices are faulty. I also gave mitsubishi2mqtt a go, and it can't connect either. So it's not the code, but hoping we can pool info and solve it

Well, if you're sure that:the board connectors (pins) are okay on both sides,your cable is good (you have performed an electrical continuity test),it doesn't work with other controllers (and you have verified you're using the correct Rx and Tx pin numbers),your unit is known to work with any of the Swicago implementations,then I'm afraid you don't have many options left!What happened to me once, when I was using a specific connection scheme with both a heat pump and a Raspberry Pi (for collecting logs) connected to two different UARTs on the controller, was that I couldn't get the controller to read the heat pump responses either. My error was using the Raspberry Pi's ground for the controller instead of the heat pump's ground. It's really important for both sides of the UART to share the same ground. I think the most crucial thing is to ensure a nice and stable ground connection between the two. So, I would try using a pull-down resistor rather than a pull-up. However, I must admit my knowledge of hardware electronics is very limited, so this could also be a terrible piece of advice!

echavet, 2024-05-02:
I have tested 👍 these setup:
```yaml
esp8266:
  board: d1_mini
uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3
```

mfinnz, 2024-05-02:
Thanks for your suggestions. I have reversed the connection for the 10kOhm resistors so they connect to the GND as a pull down resistor, but it's still the same. The board I am using is this one [https://www.aliexpress.com/item/1005005121150737.html](url) I have tried 3 of them, so unlikely it's a faulty board.

I may end up just getting ESP32 board as others have had problems with ESP8266, or seeing if I can order the one you use.

I am using this setup:

```
# For ESP8266 Devices
esp8266:
  board: d1_mini

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3
```

Also I use this for the components, I thought that means I am using the latest, v 1.3? If not, do I just download the latest release zip file and how do I install it? I thought it did it at install/build time using the external component, but I'm quite new to this.

```
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome
    refresh: 0s

```

HansVanEylen, 2024-05-02:
Got it working at last !

I used the Atom Echo and carefully reviewed the wiring again (like for the 99th time...) and it was recognized immediately. I suspect there may have been contact issues with the dupont connectors previously.
I'm going to run it like this for some time and then get rid of the dupont connectors by trim the wires down to a manageable length, soldering them with heat shrinking tube around. For the other heatpumps I will use the Atom Lite without speaker (https://www.tinytronics.nl/nl/development-boards/microcontroller-boards/met-wi-fi/m5stack-atom-lite-esp32-development-board)

The equipment used for this test setup:

- the Atom Echo https://shop.m5stack.com/products/atom-echo-smart-speaker-dev-kit (had this lying around)
- Pa 2.0Mm Pap to Dupont female Connector PAP-04V-S PAP-05V-S- I. This is the exact CN105 connector. (https://nl.aliexpress.com/item/1005005562174022.html?spm=a2g0o.order_detail.order_detail_item.3.7d156d76f6f5wz&gatewayAdapt=glo2nld)
- Micro JST PH 2.0 5-pin connectors which fit the Atom Echo perfectly if you carefully cut off one pin (https://www.amazon.com.be/-/nl/dp/B08RCX4B84?ref=ppx_yo2ov_dt_b_product_details&th=1)

For now it looks like this:

[image]

Thanks for the work you put into this ! It means I will be able to dump the unreliable Melcloud service.

mfinnz, 2024-05-03:
So I've tried various things all morning and decided to give up on this D1 Mini I have. I've ordered both Atom Lite and a generic ESP32 board. Will give them both a go in a few weeks when they arrive and fingers crossed it all come right. Thanks for the help so far

echavet, 2024-05-07:
Try to remove the 
> hardware_uart: UART1
From your log section

Look at the corresponding UART numbers and their Rx an tx pin numbers:

https://docs.espressif.com/projects/esp-at/en/latest/esp32/Get_Started/Hardware_connection.html

c-mholm, 2024-05-07:
Thanks for your help. I removed 

```yaml
Hardware_uart: UART1
```

And added
```yaml
logger:
  baud_rate: 0
```

Something changed in the writeback in the log, but still no communication.

Then I changed 
```yaml
tx_pin: GPIO17
rx_pin: GPIO16
```
After googling ESP32 WROOM 32 pin out diagram.
To

```yaml
tx_pin: GPIO1
rx_pin: GPIO3
```

Now its working, woohoo :)

marsavela, 2024-05-10:
Same issue here with the LOLIN Wemos D1 Mini V4.0.0

I have not been successful with any config.

echavet, 2024-05-11:
> Same issue here with the LOLIN Wemos D1 Mini V4.0.0
> 
> I have not been successful with any config.

Yes it is known that this controller cannot connect with heatpump

marsavela, 2024-05-12:
> > Same issue here with the LOLIN Wemos D1 Mini V4.0.0
> > I have not been successful with any config.
> 
> Yes it is known that this controller cannot connect with heatpump

👍 

I didn't know that. 🫤

I switched it with an AZDelivery board and everything works as expected.

drawks, 2024-05-17:
> Same issue here with the LOLIN Wemos D1 Mini V4.0.0
> 
> I have not been successful with any config.

The issue with the d1 mini v4 compared to the earlier v1.0.0 board is that there is no resistor on the board between the rx pin and the ch340c usb controller chip. Without this resistor when the AC unit tries to pull this line to ground it is unable to get the voltage to drop low enough because current backfeeds from the usb controller.

You can modify the board by lifting pin 2 of the usb controller and either leaving it disconnected along with adding a 10k pull up resistor between the RX(GPIO3) and VBUS *OR* you can put a 470ohm resistor between the lifted pin 2 on the usb controller and the RX(GPIO3) pin to match the wiring of the earlier v1 board

It is also worth noting that you cannot easily use the alternate UART pins on GPIO 13/15 without both providing a pull resistor on those pins but also ensuring that GPIO15 is not pulled up during startup or the board will not boot.

I'm just data dumping here, because I've been struggling with a bunch of these v4 boards that bought in bulk and it was maddening see many reports of them not working, but without any concrete advice on how to work around their failing

closed 2024-05-21 by echavet

mfinnz, 2024-06-08:
Very happy to report that this is finally working for me. All I did was order a ESP32 board, connected it to the VIN (5V) and GND with GPIO16 and 17 and it's working perfectly. No resisters or voltage regulator needed. This is the board I ordered

https://www.aliexpress.com/item/1005006422498371.html

eikowagenknecht, 2024-06-20:
> > Same issue here with the LOLIN Wemos D1 Mini V4.0.0
> > I have not been successful with any config.
> 
> Yes it is known that this controller cannot connect with heatpump

I have a Lolin V3.1.0 and can't seem to be able to get it working either. Is that controller known to be not working as well?

echavet, 2024-06-20:
Yes ! I'm not surprised, the UART communication between the chip and the heat pump is quite sensitive. It requires a stable and very clean power supply. Ultimately, few chips are clean enough to establish reliable communication.

drawks, 2024-06-20:
> > > Same issue here with the LOLIN Wemos D1 Mini V4.0.0
> > > I have not been successful with any config.
> > 
> > Yes it is known that this controller cannot connect with heatpump
> 
> I have a Lolin V3.1.0 and can't seem to be able to get it working either. Is that controller known to be not working as well?

This board also has a bare connection between its USB controller and the uart0 Rx pin, it should work with a small rewire as I detailed in my previous comment.

disruptivepatternmaterial, 2024-06-20:
> > > Same issue here with the LOLIN Wemos D1 Mini V4.0.0
> > > I have not been successful with any config.
> > 
> > 
> > Yes it is known that this controller cannot connect with heatpump
> 
> I have a Lolin V3.1.0 and can't seem to be able to get it working either. Is that controller known to be not working as well?

Let's walk through some logic here.

A 2000$ up to 30,000$ HVAC system and people keep wanting to save 5$ on controllers. Does this make sense to anyone? Hell, it is a free country, as they say...

Stop buying junk from rando's on the internet: the adafruit, m5, and lilygo are more money, but we are talking less than 10$ more and they just work.

I use ONLY m5 because they by and large contain the same plugs, which makes them easy to use, they mostly contain battery controls, and come in nice cases.

I have never had an issue with an m5 device. I use the stamps (orange ones). I even have other sensors on them and they never fail.

eikowagenknecht, 2024-06-20:
@drawks
Thanks for the clarification!

@disruptivepatternmaterial
You‘re making the wrong assumption here (that I bought the Lolin board to save money). For me it‘s simply that I don‘t know much about these boards and the Lolin ones were recommended somewhere.

So thanks for the product recommendation!

drawks, 2024-06-20:
@disruptivepatternmaterial I'm not sure the issues of this project is the place to debate the relative value proposition of the various microcontrollers. They are all very similar, cost has very little to do with it. The entire market of knock off d1 controllers are trace and component identical to the more expensive original wemo design, they contain the same flaw as that design.

It is nice that this project does maintain a list of controllers which are known to work with no additional circuitry, beyond that I don't think it contributes anything to the project to imply that people are having problems because they are "buying junk from randos on the internet".

There are plenty of 15 dollar parts that don't work and there are plenty of 3 dollar ones that do.