echavet/MitsubishiCN105ESPHome #86: Component esphome.coroutine took a long time for an operation (75 ms)

c-mholm, 2024-05-08:
I get this yellow warning/error every cycle the code loops. I tried to google it but seems like there is no universal fix but maybe something that needs to be addressed in the Mitsubishi code for ESPHome? Anyone else getting this?

esp32dev
Firmware: [MitsubishiCN105ESPHome](https://github.com/echavet/MitsubishiCN105ESPHome)
Connected to GPIO1 and GPIO3

[image]

echavet, 2024-05-08:
This is "normal" behaviour. The component should not block the esphome framework for a too long period but if you keep the logs active you can't deal without this warning. Actually I couldn't!!!

However, even without any log, some chips take too much time according to the esphome framework!

I could try to optimise again and again by limiting the number of operations at each loop. But I think it is not worth the pain.

c-mholm, 2024-05-08:
Ok. That’s what I thought. Read about a lot of different implementations with the same message. Everything seems to work so we just have to live with it. My ocd is just not happy with yellow text in the log :)

closed 2024-05-08 by c-mholm