echavet/MitsubishiCN105ESPHome #87: Weird option in Horizontal Vane (widevane)

jcugat, 2024-05-08:
Hey @echavet first of all thanks for this component! I just started using it but it's been great so far.

I have it running with my unit MSZ-AY25VGK and noticed something odd with the Horizontal Vane (widevane) option. From all the possible values, there is one called `←→` which is defined here https://github.com/echavet/MitsubishiCN105ESPHome/blob/c6eb0cc0ecef0597f605b6da09f56ddd3cd1ec63/components/cn105/Globals.h#L79-L80

But that option doesn't seem to do anything for my unit, is this a mode only supported in some models? Do you know if there is any option to "hide" it from home assistant?

Not sure if related, but I have a lot of logs spammed like this on my Wemos D1 mini:

```
[14:55:00][I][CYCLE:038]: 1: Cycle start
[14:55:00][I][CN105:494]: widevane setting (extra select component) changed
[14:55:01][I][CYCLE:051]: 6: Cycle ended in 0.9 seconds (with timeout?:  NO)
[14:55:05][I][CYCLE:038]: 1: Cycle start
[14:55:05][I][CN105:494]: widevane setting (extra select component) changed
[14:55:06][I][CYCLE:051]: 6: Cycle ended in 0.9 seconds (with timeout?:  NO)
[14:55:10][I][CYCLE:038]: 1: Cycle start
[14:55:10][I][CN105:494]: widevane setting (extra select component) changed
[14:55:10][I][CYCLE:051]: 6: Cycle ended in 0.9 seconds (with timeout?:  NO)
[14:55:14][I][CYCLE:038]: 1: Cycle start
[14:55:15][I][CN105:494]: widevane setting (extra select component) changed
[14:55:15][I][CYCLE:051]: 6: Cycle ended in 0.9 seconds (with timeout?:  NO)
```

It also mentions `widevane` all the time, as if somehow that setting was being changed all the time, but I'm not changing anything through the IR remote nor Melcloud.

Thanks!

SixFive7, 2024-05-08:
Hi,

I have three MSZ-AP42VGK units. These units also do not support the horizontal fan split "←→" setting. Whenever I set it to that configuration the unit just ignores it and stays on the previous setting. It would be nice if I could hide this unsupported value for my specific units.

Looking at geoffdavis/esphome-mitsubishiheatpump's [advanced configuration](https://github.com/geoffdavis/esphome-mitsubishiheatpump#advanced-configuration) section it has an "supports" section:
[image]

It would be nice to port that capability so we can clean up our interface in ESPHome and Home Assistant by hiding hardware incompatible options if applicable.

SixFive7, 2024-05-08:
Concerning the Wemos D1 mini (which I also have installed); I can confirm I am also seeing log spam. This happens always. Be it in heating, cooling or even off mode. I just assumed this was normal, but thinking about it more I realize all other operations are not continuously being reset. Just the widevane setting is. This is the log my three (MSZ-AP42VGK with a Wemos D1 mini) units are giving me:

[image]

echavet, 2024-05-09:
> to that configuration the unit

hi @SixFive7 
could you share your complete yaml configuration and a screen capture of your component. I might be missing something but my heatpump does not support horizontal vane either. So don't define the horizontal select option. And I only set the supported options in the `SUPPORTS` section. And I don't see any UI options for unsupported options.

echavet, 2024-05-09:
To everyone:
it would be easier for me to analyse this issue if you could provide 
- your complete yaml 
- the date when you uploaded your firmware

SixFive7, 2024-05-09:
Hi Eric,

I think there is a misunderstanding. There are two issues in this issue. Issue number 1 is limited widevane support. My MSZ-AP42VGK does support vertical and horizontal vane positions. These both work perfectly fine. Except for the `←→` option. My unit can send air to the very left, to the left, through the middle, to the right and to the very right. It can also swing. What it cannot do is split the airflow by sending half of the machine to the left and the other half to the right (which I assume is what the `←→` means).
In other words: I can set my horizontal vane to 0x01, 0x02, 0x03, 0x04, 0x05, and 0x0c. But when I set it to 0x08 the unit just ignores the setting. It would be nice to add some extra yaml config capability where I can define that my unit cannot handle 0x08 so that it hides `←→` in both the ESPHome component and Home Assistant.
Right now we can configure in yaml if we have widevane support at all or not.

What I don't understand is the rationale behind https://github.com/echavet/MitsubishiCN105ESPHome/pull/88. Why did you change the label of 0x08 from `←→` to `→→` (the same as 0x05)? I hope not because of me?

Code reference:
```
static const uint8_t VANE[7] = { 0x00,  0x01, 0x02, 0x03, 0x04, 0x05, 0x07 };
static const char* VANE_MAP[7] = { "AUTO", "↑↑", "↑", "—", "↓", "↓↓", "SWING" };
static const uint8_t WIDEVANE[7] = { 0x01, 0x02, 0x03, 0x04, 0x05, 0x08, 0x0c };
static const char* WIDEVANE_MAP[7] = { "←←", "←", "|", "→", "→→", "←→", "SWING" };
```

Completely unrelated (I think) is issue number 2. This pertains to the constant logging of changes to the widevane setting regardless of what setting it is on. To be clear, setting the widevane property works. I see my unit moving and all is good. Only, the log is getting spammed like it is continuously sending the widevane setting to the unit.

I am not sure what date I uploaded my code. But as per yesterday there was no update available.
<details>
<summary>This is my complete yaml</summary>

```yaml
substitutions:
  name: dansvloer-warmtepomp
  friendly_name: Dansvloer warmtepomp

esphome:
  name: ${name}
  friendly_name: ${friendly_name}

esp8266:
  board: d1_mini

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3

external_components:
  source: github://echavet/MitsubishiCN105ESPHome
  refresh: 0s

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: "${name}"
    password: !secret wifi_fallback_password
captive_portal:

logger:
  hardware_uart: UART1
  level: INFO
  logs:
    EVT_SETS : INFO
    WIFI : INFO
    MQTT : INFO
    WRITE_SETTINGS : INFO
    SETTINGS : INFO
    STATUS : INFO
    CN105Climate: WARN
    CN105: INFO
    climate: WARN
    sensor: WARN
    chkSum : INFO
    WRITE : WARN
    READ : WARN
    Header: INFO
    Decoder : INFO
    CONTROL_WANTED_SETTINGS: INFO
#  level: DEBUG
#  logs:
#    EVT_SETS : DEBUG
#    WIFI : INFO
#    MQTT : INFO
#    WRITE_SETTINGS : DEBUG
#    SETTINGS : DEBUG
#    STATUS : INFO
#    CN105Climate: WARN
#    CN105: DEBUG
#    climate: WARN
#    sensor: WARN
#    chkSum : INFO
#    WRITE : WARN
#    READ : WARN
#    Header: INFO
#    Decoder : DEBUG
#    CONTROL_WANTED_SETTINGS: DEBUG

api:
  encryption:
    key: !secret api_key
  services:
#    - service: set_remote_temperature
#      variables:
#        temperature: float
#      then:
#        - lambda: 'id(hp).set_remote_temperature(temperature);'
    - service: use_internal_temperature
      then:
        - lambda: 'id(hp).set_remote_temperature(0);'

ota:

web_server:
  local: true
  ota: false

time:
  - platform: homeassistant
    id: homeassistant_time

text_sensor:
  - name: ESPHome Version
    platform: version
  - platform: wifi_info
    ip_address:
      name: IP
    ssid:
      name: SSID
    bssid:
      name: BSSID

sensor:
  - name: Uptime
    platform: uptime

  - name: WiFi Signal
    platform: wifi_signal
    update_interval: 120s

button:
  - name: "Restart"
    platform: restart

climate:
  - name: "${friendly_name}"
    platform: cn105
    id: hp
    icon: mdi:heat-pump
    visual:
      min_temperature: 16
      max_temperature: 31
      temperature_step:
        target_temperature: 1
        current_temperature: 0.1
    compressor_frequency_sensor:
      name: Compressor Frequency
    vertical_vane_select:
      name: Vertical Vane
    horizontal_vane_select:
      name: Horizontal Vane
#    isee_sensor:
#      name: ISEE Sensor
    remote_temperature_timeout: 30min
    debounce_delay : 500ms
    update_interval: 4s
    stage_sensor:
      name: Stage Sensor
    sub_mode_sensor:
      name: Sub Mode Sensor
    auto_sub_mode_sensor:
      name: Auto Sub Mode Sensor
```
</details>

SixFive7, 2024-05-09:
> > to that configuration the unit
> 
> hi @SixFive7 could you share your complete yaml configuration and a screen capture of your component. I might be missing something but my heatpump does not support horizontal vane either. So don't define the horizontal select option. And I only set the supported options in the `SUPPORTS` section. And I don't see any UI options for unsupported options.
> 
> When I look at the code I can see this instruction `ESP_LOGI(TAG, "widevane setting (extra select component) changed");` is located in a logical `IF` test that should only be executed when `horizontal_vane_select_` is configured. So I think you might have defined the `horizontal_vane_select` in your yaml file.

My unit does support widevane settings. See ticket above. Here is a full screenshot of my component. I hope that clarifies things. :-)

[image]

echavet, 2024-05-09:
> Hi Eric,
> 
> I think there is a misunderstanding. There are two issues in this issue. Issue number 1 is limited widevane support. My MSZ-AP42VGK does support vertical and horizontal vane positions. These both work perfectly fine. Except for the `←→` option. My unit can send air to the very left, to the left, through the middle, to the right and to the very right. It can also swing. What it cannot do is split the airflow by sending half of the machine to the left and the other half to the right (which I assume is what the `←→` means). In other words: I can set my horizontal vane to 0x01, 0x02, 0x03, 0x04, 0x05, and 0x0c. But when I set it to 0x08 the unit just ignores the setting. It would be nice to add some extra yaml config capability where I can define that my unit cannot handle 0x08 so that it hides `←→` in both the ESPHome component and Home Assistant. Right now we can configure in yaml if we have widevane support at all or not.
> 
> What I don't understand is the rationale behind #88. Why did you change the label of 0x08 from `←→` to `→→` (the same as 0x05)? I hope not because of me?
> 
> Code reference:
> 
> ```
> static const uint8_t VANE[7] = { 0x00,  0x01, 0x02, 0x03, 0x04, 0x05, 0x07 };
> static const char* VANE_MAP[7] = { "AUTO", "↑↑", "↑", "—", "↓", "↓↓", "SWING" };
> static const uint8_t WIDEVANE[7] = { 0x01, 0x02, 0x03, 0x04, 0x05, 0x08, 0x0c };
> static const char* WIDEVANE_MAP[7] = { "←←", "←", "|", "→", "→→", "←→", "SWING" };
> ```
> 
> Completely unrelated (I think) is issue number 2. This pertains to the constant logging of changes to the widevane setting regardless of what setting it is on. To be clear, setting the widevane property works. I see my unit moving and all is good. Only, the log is getting spammed like it is continuously sending the widevane setting to the unit.
> 
> I am not sure what date I uploaded my code. But as per yesterday there was no update available.
> 
> This is my complete yaml

ok, I understand. 
1: the "←→" is just a copy/paste error. The "←→" does not exist. It should be "→→", representing the extreme right orientation.
2: This issue is also a copy/paste error 
`if (this->hasChanged(this->horizontal_vane_select_->state.c_str(), **settings.vane**, "select wideVane")) {`
`if (this->hasChanged(this->horizontal_vane_select_->state.c_str(), settings.wideVane, "select wideVane")) {`

SixFive7, 2024-05-09:
Hi Eric,

Thank you for the fast response.

For issue 1 I don't understand the solution. The current code in this issue (and now also the main) branch gives:
https://github.com/echavet/MitsubishiCN105ESPHome/blob/b494bcb61d2804468485494939cccd807b28ad3b/components/cn105/Globals.h#L79-L80
In here both 0x05 and 0x08 seem to have the same `→→` label. But my unit does not work with 0x08. Also, won't this give two identical options in the drop down menu and in home assistant?
If 0x08 doesn't work it should be removed altogether. If 0x08 does work on some units it should be set back to whatever it means. We translated `<>` (taken directly from [SwiCago/HeatPump](https://github.com/SwiCago/HeatPump?tab=readme-ov-file#controlling-the-heat-pump)) to `←→`. Why do you think 0x05 and 0x08 both should be `→→`? Do you have some information unknown to the SwiCago project?
If 0x08 horizontal vane mode really is only supported by just some units. We could optionally add extra yaml config code to hide hardware incompatible sub options like 0x08? But that would require extra code.

For issue 1 and 2 I have tested the current state of affairs by building against your branch:
```yaml
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome@echavet/issue87
    refresh: 0s
```
This is the result:

[image]

As you can see, the console log is very clean now. I take it the cycles logs are expected?
But I have two identical horizontal vane options are unexpected.

jcugat, 2024-05-12:
Thanks for taking a look and fixing the log spam @echavet, I can confirm that the `widevane setting` log is not spammed anymore.

Regarding the other issue with `←→`, it's exactly as @SixFive7 described. Our unit don't support this mode for the wide vane, and it's completely ignored if we set it. Would be great if it could be hidden somehow, either with a configuration in the yaml file, or by autodetecting the support (or lack of thereof) for it.

SixFive7, 2024-05-12:
To add: I've traced back the `0x08`/`<>` (now `←→`) option to the initial commit of https://github.com/SwiCago. There is no explanation of why that value is there. Furthermore I see no mention of it at all anywhere by anybody. I'm not sure if there even exists a unit that supports split horizontal vanes through this interface. All the unit documentation to be found in the https://github.com/SwiCago repository shows units with either no horizontal vane support, or units supporting the options `0x01` through `0x05` and `0x0c`. No clue as to what `0x08` is supposed to do anywhere.
I'd opt for removing the value `0x08` ( `←→`) altogether.

echavet, 2024-05-13:
Thank you very much for your message and research.
I will check that soon.

homegreen, 2024-05-14:
0x08 ( ←→) means something on my units (MSZ-FS series). It means to direct air away from the occupants in the room. Please don't remove it entirely (but config or auto-detecting if it is available makes sense).

SixFive7, 2024-05-14:
Thank you for reacting! Apologies for not finding your unit in my research.

PR #88 (2024-05-09): Weird option in Horizontal Vane (widevane)
fix the horiz waidevane const error →→
Fixes #87
commit 8319ab26 (2024-05-09): Weird option in Horizontal Vane (widevane)
Fixes #87
commit 62ab7d2a (2024-05-09): Weird option in Horizontal Vane (widevane)
Fixes #87
PR #91 (2024-05-09): Weird option in Horizontal Vane (widevane)
Fixes #87