echavet/MitsubishiCN105ESPHome #95: 134 broke all my devices...

disruptivepatternmaterial, 2024-05-10:
...I had to revert back to 133, not sure why, it is midnight here and need to get some sleep :) but something is up!

echavet, 2024-05-10:
Oh sh...!!

echavet, 2024-05-10:
@disruptivepatternmaterial,
as soon as possible, could you tell me more about that?
I don't understand what could happen because the difference between 1.3.3 ans 1.3.4 is just about vaneMap and wideVaneMap constants: https://github.com/echavet/MitsubishiCN105ESPHome/compare/v1.3.3...v1.3.4?diff=split&w=
I'd like to fix the pb ASAP if it's so bad.

closed 2024-05-10 by echavet

echavet, 2024-05-10:
ok I think it is fixed on v1.3.4b

disruptivepatternmaterial, 2024-05-10:
HAHAH yes sorry about that I wanted to post something before I went to bed. I am a tad clueless as to what the issue was because 2 of them eventually came back, but 2 required me to remove it and flash it from the usb. Let me try out what you have now and see what I find :)

echavet, 2024-05-12:
> HAHAH yes sorry about that I wanted to post something before I went to bed. I am a tad clueless as to what the issue was because 2 of them eventually came back, but 2 required me to remove it and flash it from the usb. Let me try out what you have now and see what I find :) 

😂
I think it was a (nullptr) side effect of my attempt to fix an issue about widevane.

PR #96 (2024-05-10): 134 broke all my devices...
Fixes #95
commit dc4055f6 (2024-05-10): 134 broke all my devices...
Fixes #95