echavet/MitsubishiCN105ESPHome #101: Thermostat value won't take

axyjo, 2024-05-18:
Hi! I just installed this on a Atom S3, and everything works well for the most part. However, I'm having some issues with getting the thermostat set point to persist. Whatever I do, it seems to reset back to 19 deg C. Has anyone else run into this?

I'm currently working around this by adding or subtracting a value to the room temperature sensor reading, but it'd certainly be nice to have the HA thermostat component work out of the box.

<details>
  <summary>My config</summary>
  
  ```yaml
  esphome:
  name: heat-pump
  friendly_name: Heat Pump

esp32:
  board: esp32-s3-devkitc-1
  framework:
    type: arduino

# Enable Home Assistant API
api:
  encryption:
    key: "..."
  services:
    - service: set_remote_temperature
      variables:
        temperature: float
      then:
# Select between the C version and the F version
# Uncomment just ONE of the below lines. The top receives the temperature value in C,
# the bottom receives the value in F, converting to C here.
        - lambda: 'id(hp).set_remote_temperature(temperature);'
#        - lambda: 'id(hp).set_remote_temperature((temperature - 32.0) * (5.0 / 9.0));'
    - service: use_internal_temperature
      then:
        - lambda: 'id(hp).set_remote_temperature(0);'
ota:
  password: "..."

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Heat-Pump Fallback Hotspot"
    password: "..."

captive_portal:

external_components:
  - source: github://echavet/MitsubishiCN105ESPHome@v1.3.5

time:
  - platform: homeassistant
    id: esptime

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: GPIO1
  rx_pin: GPIO8

climate:
  - platform: cn105
    id: hp
    name: "${friendly_name}"
    icon: mdi:heat-pump
    visual:
      min_temperature: 15
      max_temperature: 31
      temperature_step:
        target_temperature: 1
        current_temperature: 0.1
    compressor_frequency_sensor:
      name: ${name} Compressor Frequency
    # vertical_vane_select:
    #   name: ${name} Vertical Vane
    # horizontal_vane_select:
    #   name: ${name} Horizontal Vane
    # isee_sensor:
    #   name: ${name} ISEE Sensor
    stage_sensor:
      name: Stage Sensor
    sub_mode_sensor:
      name: Sub Mode Sensor
    auto_sub_mode_sensor:
      name: Auto Sub Mode Sensor
    remote_temperature_timeout: 30min
    debounce_delay : 500ms
    update_interval: 4s

# Enable logging
logger:
  level: DEBUG
  logs:
    EVT_SETS : DEBUG
    WIFI : INFO
    MQTT : INFO
    WRITE_SETTINGS : INFO
    SETTINGS : INFO
    STATUS : DEBUG
    CN105Climate: WARN
    CN105: INFO
    climate: WARN
    sensor: WARN
    chkSum : INFO
    WRITE : WARN
    READ : WARN
    Header: INFO
    Decoder : INFO
    CONTROL_WANTED_SETTINGS: INFO

#G41=Button
binary_sensor:
  - platform: status
    name: "Node Status"
    id: system_status

sensor:
  - platform: homeassistant
    id: current_temp
    entity_id: climate.thermostat
    attribute: current_temperature
    internal: true
    on_value:
      then:
        - lambda: |-
            id(hp).set_remote_temperature(x + 2);
    filters:
      - heartbeat: 1min
  ```
</details>

echavet, 2024-05-20:
Hello @axyjo,

It seems to me your log are good.

I feel a little confused reading your message. Because you write:

_I'm having some issues with getting the **thermostat set point** to persist. Whatever I do, it seems to reset back to 19 deg C. _

And you write later: 

The set_remote_temperature(x+2)

And then you talk about the wanted settings...

I'm not sure what you're expecting but it seems to me that you're confusing the set point with the external room temperature.

The former can be set with the UI climate component, while the later can be set by calling the set_remote_temperature() method.

So when you set the room temperature by calling set_remote_temperature(x+2) method, you shouldn't expect to see the wanted settings changing because this is not a wanted settings property. As it is not the set point.  

Am I wrong ?

axyjo, 2024-05-20:
@echavet 

To be clear, I'm setting the temperature using the climate component -- the `set_remote_temperature` is just a hack I'm using for comfort reasons. Here's a video of what it looks like:

https://github.com/echavet/MitsubishiCN105ESPHome/assets/130490/4433549d-40be-471c-8655-a39ea25e3722

echavet, 2024-05-21:
ok, thanks fo the video it's great.

It looks like the heatpump don't want the new setpoint. Logs are running as expected : wanted settings are sent, update cycle is deferred to wait for heatpump ACK. Ack is received which means the heatpump did receive the new setpoint. Then the cycle starts again. And theheatpump sends his internal properties. setpoint seems to have been ignored.

I need to work on this. But I need a little help please, as I'm a little surprised by this behavior. 

- Do you have this behavior all the time or only sometimes? 
- As the user manipulating the UI, what do you feel is happening?

axyjo, 2024-05-21:
> Do you have this behavior all the time or only sometimes?

It happens all the time

> As the user manipulating the UI, what do you feel is happening?

For me, it feels like there's some kind of communication failure between the heat pump and the ESP32.

Now I'm wondering if there's a parameter on the heat pump settings that I haven't gotten set correctly.

echavet, 2024-05-21:
ok, I can't reproduce the issue in my home.
I'm gonna need your help if you don't mind, to diagnose the issue.

Could you please set the loggers `SETTINGS` and `WRITE_SETTINGS` to DEBUG instead if INFO, and post here the result with the same use case?
```yaml
# Enable logging
logger:
level: DEBUG
logs:
  ...
  SETTINGS : DEBUG                 # usefull to debug the received settings from the heatpump 
  WRITE_SETTINGS: DEBUG    # usefull to debug packet before we send them to the heatpump 
  ...
```

> Now I'm wondering if there's a parameter on the heat pump settings that I haven't gotten set correctly.

What is your heatpump model?
It would be a good idea to see if there are successful use of the same model by other users.
If you were facing a communication failure, I think it could be in the way the setpoint temp is encoded. This is a good lead.

Before we send the setpoint to the heatpump we encode it depending on a tempMode bool variable which is set in the `getSettingsFromResponsePacket()` method while receiving a packet from the heatpump.
 It results in using one or another temperature encoding protocole. Perhaps one of them has not been tested and perhaps this the one you are having trouble with.

```cpp
   if (wantedSettings.temperature != -1) {
        if (!tempMode) {
            ESP_LOGD(TAG, "temperature (tempmode is false) -> %f", getTemperatureSetting());
            packet[10] = TEMP[lookupByteMapIndex(TEMP_MAP, 16, getTemperatureSetting(), "temperature (write)")];
            packet[6] += CONTROL_PACKET_1[2];
        } else {
            ESP_LOGD(TAG, "temperature (tempmode is true) -> %f", getTemperatureSetting());
            float temp = (getTemperatureSetting() * 2) + 128;
            packet[19] = (int)temp;
            packet[6] += CONTROL_PACKET_1[2];
        }
    }
```

My packet is encoded using the 19th byte (A8) value:

> [15:16:30][D][EVT_SETS:057]: [wantedSettings]-> [power: -, target °C: 20.0, mode: -, fan: -, vane: -, wvane: -, hasChanged ? -> YES, hasBeenSent ? -> YES]
> [15:16:30][D][**WRITE_SETTINGS**:157]: FC 41 01 30 10 01 **04** 00 00 00 00 00 00 00 00 00 00 00 00 **A8** 00 D1 
> 

which means in my case the `tempMode` is set to `true`.
I'd like to know hows it is encoded in your case. You will be able to get the packet with the `WRITE_SETTINGS` logger set to `DEBUG`.

axyjo, 2024-05-22:
> What is your heatpump model?

I'm using a PVA-A24AA7 ducted heat pump

Looking at my `WRITE_SETTINGS` logs, I also see the 19th byte being changed (I believe `A2` encodes to 17 degC):

`[20:32:48][D][WRITE_SETTINGS:157]: FC 41 01 30 10 01 04 00 00 00 00 00 00 00 00 00 00 00 00 A2 00 D7 `

<details>
<summary>Full snippet</summary>

```
[20:32:47][D][SETTINGS:098]: [current]-> [power: ON, target °C: 19.0, mode: COOL, fan: AUTO, vane: SWING, wvane: -]
[20:32:47][D][SETTINGS:098]: [received]-> [power: ON, target °C: 19.0, mode: COOL, fan: AUTO, vane: SWING, wvane: -]
[20:32:47][D][EVT_SETS:057]: [wanted]-> [power: -, target °C: -1.0, mode: -, fan: -, vane: -, wvane: -, hasChanged ? ->  NO, hasBeenSent ? ->  NO]
[20:32:47][D][SETTINGS:079]: [climate]-> [mode: COOL, target °C: 19.0, fan: AUTO, swing: VERTICAL]
[20:32:47][D][CYCLE:270]: 3a: Sending room °C request (0x03)
[20:32:47][D][control:029]: espHome control() interface method called...
[20:32:47][I][control:045]: Setting heatpump setpoint : 17.0
[20:32:47][D][EVT_SETS:082]: clim.control() -> User changed something...
[20:32:47][D][EVT_SETS:057]: [control (wantedSettings)]-> [power: -, target °C: 17.0, mode: -, fan: -, vane: -, wvane: -, hasChanged ? -> YES, hasBeenSent ? ->  NO]
[20:32:48][D][CYCLE:276]: 3b: Receiving room °C response
[20:32:48][D][CYCLE:279]: 4a: Sending status request (0x06)
[20:32:48][D][CYCLE:297]: 4b: Receiving status response
[20:32:48][D][CYCLE:299]: 5a: Sending power request (0x09)
[20:32:48][D][CYCLE:305]: 5b: Receiving Power/Standby response
[20:32:48][D][text_sensor:064]: 'Stage Sensor': Sending state 'GENTLE'
[20:32:48][D][text_sensor:064]: 'Sub Mode Sensor': Sending state 'NORMAL'
[20:32:48][D][text_sensor:064]: 'Auto Sub Mode Sensor': Sending state 'AUTO_OFF'
[20:32:48][I][CYCLE:052]: 6: Cycle ended in 0.9 seconds (with timeout?:  NO)
[20:32:48][I][EVT_SETS:013]: checkPendingWantedSettings - wanted settings have changed, sending them to the heatpump...
[20:32:48][I][CN105:250]: sending wantedSettings..
[20:32:48][D][EVT_SETS:057]: [wantedSettings]-> [power: -, target °C: 17.0, mode: -, fan: -, vane: -, wvane: -, hasChanged ? -> YES, hasBeenSent ? -> YES]
[20:32:48][D][WRITE_SETTINGS:157]: FC 41 01 30 10 01 04 00 00 00 00 00 00 00 00 00 00 00 00 A2 00 D7 
[20:32:48][I][CYCLE:032]: Defering cycle trigger of 1500 ms
[20:32:48][D][Update-ACK:157]: FC 61 01 30 10 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 5E 
[20:32:48][I][CN105:357]: Last heatpump data update successful!
[20:32:54][I][CYCLE:038]: 1: Cycle start
[20:32:54][D][CYCLE:313]: 2a: Sending settings request (0x02)
[20:32:54][D][CYCLE:267]: 2b: Receiving settings response
[20:32:54][D][SETTINGS:098]: [current]-> [power: ON, target °C: 19.0, mode: COOL, fan: AUTO, vane: SWING, wvane: -]
[20:32:54][D][SETTINGS:098]: [received]-> [power: ON, target °C: 19.0, mode: COOL, fan: AUTO, vane: SWING, wvane: -]
[20:32:54][D][EVT_SETS:057]: [wanted]-> [power: -, target °C: -1.0, mode: -, fan: -, vane: -, wvane: -, hasChanged ? ->  NO, hasBeenSent ? ->  NO]
[20:32:54][D][SETTINGS:079]: [climate]-> [mode: COOL, target °C: 17.0, fan: AUTO, swing: VERTICAL]
[20:32:54][D][CYCLE:270]: 3a: Sending room °C request (0x03)
```

</details>

Hopefully the details above help! I'm not super familar with the protocol here, but maybe there's something in the update-ack message that can be of assistance?

echavet, 2024-05-22:
After the wanted settings are sent to your heatpump, the next cycle setting request `([D][CYCLE:313]: 2a: Sending settings request (0x02))` should receive a response with the updated "17°" target temperature. But it does not. Instead, 19° is received. This means your heatpump did not get the updated value:

`[20:32:54][D][SETTINGS:098]: [received]-> [power: ON, target °C: 19.0, mode: COOL, fan: AUTO, vane: SWING, wvane: -]` 

The problem is that I don't have any idea why this is happening. 
I could try for you, in a specific branch, to change the target temperature encoding method to force using the other option, but I'm not sure this is a good idea. This could lead to a pb with your heatpump. I've never seen that before, so I've never tested that. You would be the first.

If you want to try just tell me and I'll publish the modification in the branch.

Of course there is still a possibility that my code has a bug that only occurs in your case. The best way to test that is to check with another esphome implementation like the this one : https://github.com/geoffdavis/esphome-mitsubishiheatpump.

The other option is to perform a reverse engineering with a kumo cloud official device and see how different the target temperature is encoded. But it is complicated.

Tell me what you wan to do...

echavet, 2024-05-24:
a new implementation has been committed to the esphome main project.
This is not my implementation. But you could test it as soon as it is available.
It would be interesting to know if you still have the issue with this version.

look here for more info: https://github.com/echavet/MitsubishiCN105ESPHome/issues/100

axyjo, 2024-05-25:
Woah nice -- I'll try to give this a shot over the weekend.

echavet, 2024-05-25:
> Woah nice -- I'll try to give this a shot over the weekend.

Don't forget to share your feedback please...

echavet, 2024-06-10:
> Woah nice -- I'll try to give this a shot over the weekend.

hey @axyjo,
did you try the mitsubishi-uart component?

axyjo, 2024-07-10:
Finally got a chance to try out `mitsubishi-uart` -- it's exhibiting the same problem, so I don't think this is related to your code!

closed 2024-07-10 by axyjo