echavet/MitsubishiCN105ESPHome #115: Support for wide vane mode

Quarks-1, 2024-06-08:
I have an MLZ-KP18NA2 ductless system which has the vane settings of 1-5 vertical, vertical swing, and wide vane.  Wide vane is when the internal horizontal vane will swing left and right or stay stationary, but it cannot be selected which angle you want.  Would it be possible to add this as a setting? Currently it is only controllable through the remote for me even if I choose 'both' or 'horizontal' swing modes on the esp device. Thanks!

echavet, 2024-06-09:
Hey @Quarks-1,
My heatpumps do not support the widevane. 
So I cannot test this feature.

I'm surprised you're are the first person to face this issue.

Could you be more specific please ? I'm not sure to understand well. What is the combination of settings you can set with IR and not with the esp?

Could you also provide your complete yaml?

What I understand is that you would like to activate the horizontal swing with a specific angle, but you can't?

Are you using these attributes in the climate section?

```yaml
climate:
  vertical_vane_select:
    name: vertical vane orientation       
  horizontal_vane_select:
    name: horizontal vane Orientation
```

If so, did you notice the device in HA then provides separate select UI to set these options?
I know it's weird to have to use separate UI select components, but I didn't find a way to integrate the features in the UI climate component.

Quarks-1, 2024-06-10:
So some updates, I did some digging into the code and it seems that widevane is the same as horizontal vane,  My YAML does have the code you commented, however I have found that when I change the horizontal vane setting in HA it will always default back to ```|``` automatically.  I think this may be due to my HVAC only supporting an 'ON' or 'OFF' state for the wide (horizontal) vane rather than all the angles and modes that a vertical vane has.  On my remote it is a button labeled "WIDE VANE" and has no visual indication on the remote screen whether or not it is on or off, you can only tell by looking at the HVAC itself.  It also seems to be unable to remember if it was on or off since when you turn on the HVAC from it being off, it will always default to wide vane being off.

echavet, 2024-06-10:
Yes,
Widevane is the horizontal vane.
And when it rolls back to default, this means that:
- Either the heat pump does not support the setting
- or the protocol is not adapted to the heatpump model.
I don't know what to do to help!
Perhaps you could provide the logs (of the settings received by the esp),
after you set the widevane on with your IR remote control.

closed 2024-07-18 by echavet