echavet/MitsubishiCN105ESPHome #116: Repeating log message: auto mode sub mode caution: value 64 not found

drewkeller, 2024-06-17:
Every 5 seconds or so, I get this log message....
```
[02:07:07][W][lookup:191]: auto mode sub mode caution: value 64 not found, returning value at index 0
[02:07:12][W][lookup:191]: auto mode sub mode caution: value 64 not found, returning value at index 0
[02:07:17][W][lookup:191]: auto mode sub mode caution: value 64 not found, returning value at index 0
```

I have 3 MSZ-GS09NA units with similar setups (new installation, Wemos D1 mini, Home Assistant) and all of them are logging this message.

What additional info do you need?

echavet, 2024-06-17:
thank you for your message.
@disruptivepatternmaterial did work on these values.
```
static const uint8_t AUTO_SUB_MODE[4] = { 0x00, 0x01, 0x02, 0x03 };
static const char* AUTO_SUB_MODE_MAP[4] = { "AUTO_OFF","AUTO_COOL", "AUTO_HEAT", "AUTO_LEADER" };
```
it looks like 0x40 (64) is an unknown option.

You can prevent logger from logging it by setting logger `lookup` to ERROR in your yaml

drewkeller, 2024-06-18:
Thanks. I played around a bit with the settings and couldn't get that bit 6 to clear. I did get a value 65 when I set the mode from cool to auto, if that's interesting for any research.

Another question (hope it's ok here) .... I was trying to figure out how to set up the remote sensor, and this isn't working.
```
  - platform: homeassistant
    name: "Temperature Sensor From Home Assistant"
    id: current_temp
    entity_id: sensor.basement_thermostat_temperature
    on_value:
      then:
        - lambda: 'id(hp).set_remote_temperature(x);'
```

Result in the logs (at startup)
```
22:41:03 | [I] | [STATUS:121] | [received       ]-> [room C°: 0.0, operating: YES, compressor freq:  0 Hz]
22:41:03 | [I] | [STATUS:121] | [current        ]-> [room C°: 23.0, operating: YES, compressor freq:  0 Hz]
22:41:03 | [I] | [STATUS:121] | [received       ]-> [room C°: 0.0, operating: NO , compressor freq:  0 Hz]
22:41:03 | [I] | [STATUS:121] | [current        ]-> [room C°: 0.0, operating: YES, compressor freq:  0 Hz]
22:41:52 | [I] | [CN105:370] | Last heatpump data update successful!
```

The result is, the climate control shows 32F as the temperature (should be approx 73.4F)

echavet, 2024-06-18:
Hello @drewkeller,

I need to see your complete yaml file.

I also need a larger part of your logs, covering the whole period which matches the sent commands. You can use the markdown <details><summary> tag to expose them nicely in your messages.

I need to see the wantedSettings sent to the heatpump and the received settings coming after the next info request.

Without that it is very difficult for me to figure out the scenario you're dealing with.

Thanks for your comprehension

drewkeller, 2024-06-18:
Just to clarify, setting "lookup" to log level ERROR does suppress the warnings I was seeing before.

Here is my full config
<details>

``` yaml
substitutions:
  name: "basement-heatpump"
  friendly_name: Basement HeatPump

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp8266:
  board: d1_mini

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3

 # Default logging level
logger:
  hardware_uart: UART1 # This line can be removed for ESP32 devices
  level: INFO
  logs:
    EVT_SETS : INFO
    WIFI : INFO
    MQTT : INFO
    WRITE_SETTINGS : INFO
    SETTINGS : INFO
    STATUS : INFO
    CN105Climate: WARN
    CN105: INFO
    climate: WARN
    sensor: WARN
    chkSum : INFO
    WRITE : WARN
    READ : WARN
    Header: INFO
    Decoder : INFO
    CONTROL_WANTED_SETTINGS: INFO
    CYCLE: WARN
    lookup: ERROR

 # Enable Home Assistant API
api:
  #encryption:
  #  key: !secret api_key
  services:
    - service: set_remote_temperature
      variables:
        temperature: float
      then:
        # Select between the C version and the F version
        # Uncomment just ONE of the below lines. The top receives the temperature value in C,
        # the bottom receives the value in F, converting to C here.
        #- lambda: 'id(hp).set_remote_temperature(temperature);'
        - lambda: 'id(hp).set_remote_temperature((temperature - 32.0) * (5.0 / 9.0));'
    - service: use_internal_temperature
      then:
        - lambda: 'id(hp).set_remote_temperature(0);'

 # Allow Over-The-Air updates
ota:

 # Allow provisioning Wi-Fi via serial (requires non-zero logging)
improv_serial:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  manual_ip:
    # Set this to the IP of the ESP
    static_ip: 192.168.1.61
    # Set this to the IP address of the router. Often ends with .1
    gateway: 192.168.1.1
    # The subnet of the network. 255.255.255.0 works for most home networks.
    subnet: 255.255.255.0
  # Set up a wifi access point
  ap:
    ssid: "${friendly_name}"

 # In combination with the `ap` this allows the user 
 # to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp8266.yaml@main
  import_full_config: true

 # To have a "next url" for improv serial
web_server:

external_components:
  - source: github://echavet/MitsubishiCN105ESPHome

climate:
  - platform: cn105
    id: hp
    name: Settings
    icon: mdi:heat-pump
    visual:
      min_temperature: 15
      max_temperature: 31
      temperature_step:
        target_temperature: 1
        current_temperature: 0.1
    compressor_frequency_sensor:
      name: Compressor Frequency
    vertical_vane_select:
      name: Vertical Vane
    #horizontal_vane_select:
    #  name: Horizontal Vane
    #isee_sensor:
    #  name: ISEE Sensor
    remote_temperature_timeout: 30min
    debounce_delay : 500ms
    update_interval: 4s

 # Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

 # Text sensors with general information.
text_sensor:
  # Expose ESPHome version as sensor.
  - platform: version
    name: ESPHome Version
  # Expose WiFi information as sensors.
  - platform: wifi_info
    ip_address:
      name: IP
    ssid:
      name: SSID
    bssid:
      name: BSSID

 # Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: WiFi Signal
    update_interval: 60s

  - platform: homeassistant
    name: "Temperature Sensor From Home Assistant"
    id: current_temp
    entity_id: sensor.basement_thermostat_temperature
    on_value:
      then:
        - lambda: 'id(hp).set_remote_temperature(x);'
```

</details>

There isn't much in the way of logging (this is all that was in the log after 20 minutes), so I imagine there are some things that would need log level adjustments to get more. I re-enabled the internal sensor and disabled the HA automation that I have set up that calls the service. Which log items would output "the wantedSettings and the received settings coming after the next info request"?

<details>

```

[03:19:04][I][app:100]: ESPHome version 2024.5.5 compiled on Jun 18 2024, 03:18:33
[03:19:04][I][app:102]: Project esphome.web version 1.0
[03:19:14][I][STATUS:121]: [received       ]-> [room C°: 0.0, operating: YES, compressor freq:  0 Hz]
[03:19:14][I][STATUS:121]: [current        ]-> [room C°: 23.0, operating: YES, compressor freq:  0 Hz]
[03:19:15][I][STATUS:121]: [received       ]-> [room C°: 0.0, operating: NO , compressor freq:  0 Hz]
[03:19:15][I][STATUS:121]: [current        ]-> [room C°: 0.0, operating: YES, compressor freq:  0 Hz]
[03:23:47][I][CN105:370]: Last heatpump data update successful!
[03:23:48][I][ota:117]: Boot seems successful, resetting boot loop counter.
[03:26:37][I][CN105:370]: Last heatpump data update successful!
[03:27:26][I][CN105:370]: Last heatpump data update successful!
[03:38:26][I][CN105:370]: Last heatpump data update successful!

```

</details>

Here's the automation that I have set up and seems to work (I disabled it for the "internal" sensor method above). The sensor is a Third Reality 3RTHS24BZ which, when I have the automation enabled, looks like it reports updates every 5 minutes.

<details>

``` yaml

alias: Update Basement HeatPump Temperature
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.basement_thermostat_temperature
condition: []
action:
  - service: esphome.basement_heatpump_set_remote_temperature
    metadata: {}
    data:
      temperature: |
        {{ states('sensor.basement_thermostat_temperature') }}
mode: single

```

</details>

echavet, 2024-06-18:
Thank you.
Yes it's sure there is not enough log details to make any diagnostic.
I'm using theses log settings:
```yaml
logger:    
  level: DEBUG
  logs:
    EVT_SETS : DEBUG
    WIFI : INFO
    MQTT : INFO
    WRITE_SETTINGS : INFO
    SETTINGS : DEBUG
    STATUS : INFO
    CN105Climate: WARN
    CN105: DEBUG
    climate: WARN
    sensor: WARN
    chkSum : INFO
    WRITE : WARN
    READ : WARN
    ACK : DEBUG
    Header: INFO
    Decoder : INFO
    CONTROL_WANTED_SETTINGS: INFO    
    testMutex: DEBUG
    UPDT_ITVL: DEBUG
    CYCLE: DEBUG
    text_sensor : ERROR
```

You don't need to use both automation and esphome homeassistant sensor to update the temperature. You only need one or the other of the two different ways to set the remote temperature.
Setting the remote temperature with both mechanisms is redundant and overload the chip for nothing.

drewkeller, 2024-06-18:
I ran the logs for a few minutes using your settings, is this enough?

<details>
```

[07:57:08][I][app:100]: ESPHome version 2024.5.5 compiled on Jun 18 2024, 07:56:37
[07:57:08][I][app:102]: Project esphome.web version 1.0
[07:57:08][C][wifi:580]: WiFi:
[07:57:08][C][wifi:408]:   Local MAC: [redacted]
[07:57:08][C][wifi:413]:   SSID: [redacted]
[07:57:08][C][wifi:416]:   IP Address: 192.168.1.61
[07:57:08][C][wifi:419]:   BSSID: [redacted]
[07:57:08][C][wifi:421]:   Hostname: 'basement-heatpump'
[07:57:08][C][wifi:423]:   Signal strength: -52 dB ▂▄▆█
[07:57:08][C][wifi:427]:   Channel: 9
[07:57:08][C][wifi:428]:   Subnet: 255.255.255.0
[07:57:08][C][wifi:429]:   Gateway: 192.168.1.1
[07:57:08][C][wifi:430]:   DNS1: 0.0.0.0
[07:57:08][C][wifi:431]:   DNS2: 0.0.0.0
[07:57:08][C][logger:185]: Logger:
[07:57:08][C][logger:186]:   Level: DEBUG
[07:57:08][C][logger:188]:   Log Baud Rate: 115200
[07:57:08][C][logger:189]:   Hardware UART: UART1
[07:57:08][C][logger:193]:   Level for 'EVT_SETS': DEBUG
[07:57:08][C][logger:193]:   Level for 'WIFI': INFO
[07:57:08][C][logger:193]:   Level for 'MQTT': INFO
[07:57:08][C][logger:193]:   Level for 'WRITE_SETTINGS': INFO
[07:57:08][C][logger:193]:   Level for 'SETTINGS': DEBUG
[07:57:08][C][logger:193]:   Level for 'STATUS': INFO
[07:57:08][C][logger:193]:   Level for 'CN105Climate': WARN
[07:57:08][C][logger:193]:   Level for 'CN105': DEBUG
[07:57:08][C][logger:193]:   Level for 'climate': WARN
[07:57:08][C][logger:193]:   Level for 'sensor': WARN
[07:57:08][C][logger:193]:   Level for 'chkSum': INFO
[07:57:08][C][logger:193]:   Level for 'WRITE': WARN
[07:57:08][C][logger:193]:   Level for 'READ': WARN
[07:57:08][C][logger:193]:   Level for 'ACK': DEBUG
[07:57:08][C][logger:193]:   Level for 'Header': INFO
[07:57:08][C][logger:193]:   Level for 'Decoder': INFO
[07:57:08][C][logger:193]:   Level for 'CONTROL_WANTED_SETTINGS': INFO
[07:57:08][C][logger:193]:   Level for 'testMutex': DEBUG
[07:57:08][C][logger:193]:   Level for 'UPDT_ITVL': DEBUG
[07:57:08][C][logger:193]:   Level for 'CYCLE': DEBUG
[07:57:08][C][logger:193]:   Level for 'text_sensor': ERROR
[07:57:08][C][uart.arduino_esp8266:118]: UART Bus:
[07:57:08][C][uart.arduino_esp8266:119]:   TX Pin: GPIO1
[07:57:08][C][uart.arduino_esp8266:120]:   RX Pin: GPIO3
[07:57:08][C][uart.arduino_esp8266:122]:   RX Buffer Size: 256
[07:57:08][C][uart.arduino_esp8266:124]:   Baud Rate: 2400 baud
[07:57:08][C][uart.arduino_esp8266:125]:   Data Bits: 8
[07:57:08][C][uart.arduino_esp8266:126]:   Parity: EVEN
[07:57:08][C][uart.arduino_esp8266:127]:   Stop bits: 1
[07:57:08][C][uart.arduino_esp8266:129]:   Using hardware serial interface.
[07:57:08][C][uptime.sensor:031]: Uptime Sensor 'Uptime'
[07:57:08][C][uptime.sensor:031]:   Device Class: 'duration'
[07:57:08][C][uptime.sensor:031]:   State Class: 'total_increasing'
[07:57:08][C][uptime.sensor:031]:   Unit of Measurement: 's'
[07:57:08][C][uptime.sensor:031]:   Accuracy Decimals: 0
[07:57:08][C][uptime.sensor:031]:   Icon: 'mdi:timer-outline'
[07:57:08][C][homeassistant.time:010]: Home Assistant Time:
[07:57:08][C][homeassistant.time:011]:   Timezone: 'CST6CDT,M3.2.0,M11.1.0'
[07:57:08][C][version.text_sensor:021]: Version Text Sensor 'ESPHome Version'
[07:57:08][C][version.text_sensor:021]:   Icon: 'mdi:new-box'
[07:57:08][C][captive_portal:088]: Captive Portal:
[07:57:08][C][web_server:173]: Web Server:
[07:57:08][C][web_server:174]:   Address: 192.168.1.61:80
[07:57:08][C][mdns:115]: mDNS:
[07:57:08][C][mdns:116]:   Hostname: basement-heatpump
[07:57:08][C][ota:096]: Over-The-Air Updates:
[07:57:08][C][ota:097]:   Address: 192.168.1.61:8266
[... 1137 lines cut ...]
[08:00:19][D][SETTINGS:100]: [received       ]-> [power: ON , target °C: 22.0, mode: COOL  , fan: AUTO  , vane: AUTO  , wvane: |     ]
[08:00:19][D][EVT_SETS:059]: [wanted         ]-> [power: -  , target °C: -1.0, mode: -     , fan: -     , vane: -     , wvane: -     , hasChanged ? ->  NO, hasBeenSent ? ->  NO]
[08:00:19][D][SETTINGS:074]: [climate]-> [mode: COOL, target °C: 22.0, fan: AUTO, swing: OFF]
[08:00:19][D][CN105:218]: setting action to -> 4
[08:00:19][D][CN105:267]: Climate mode is: 2
[08:00:19][D][CN105:268]: Climate action is: 4
[08:00:19][D][CYCLE:285]: 3a: Sending room °C request (0x03)
[08:00:19][D][CN105:320]: creating Info packet
[08:00:19][D][CN105:084]: writing packet...
[08:00:19][W][component:237]: Component esphome.coroutine took a long time for an operation (71 ms).
[08:00:19][W][component:238]: Components should block for at most 30 ms.
[08:00:19][D][CYCLE:291]: 3b: Receiving room °C response
[08:00:19][D][CYCLE:294]: 4a: Sending status request (0x06)
[08:00:19][D][CN105:320]: creating Info packet
[08:00:19][D][CN105:084]: writing packet...
[08:00:19][D][CYCLE:312]: 4b: Receiving status response
[08:00:19][D][CYCLE:316]: 5a: Sending power request (0x09)
[08:00:19][D][CN105:320]: creating Info packet
[08:00:19][D][CN105:084]: writing packet...
[08:00:20][D][CYCLE:331]: 5b: Receiving Power/Standby response
[08:00:20][W][lookup:191]: auto mode sub mode caution: value 64 not found, returning value at index 0
[08:00:20][I][CYCLE:051]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)

```
</details>

drewkeller, 2024-06-19:
The Home Assistant sensor seems to be working now. I thought I had tried this before, but I applied the conversion from Fahrenheit and that seems to be what fixed it. In the process, I also figured out how to write a script to do the conversion in one place for either method (but have not figured out why the log output from the script is not working). 

``` yaml

script:
  - id: set_current_temperature
    parameters: 
      # temperature in F
      temperature: float
    then:
      - lambda: 'id(hp).set_remote_temperature((temperature - 32.0) * (5.0 / 9.0));'
      - logger.log:
          level: INFO
          format: "Setting current temperature: %.1fF"
          args: [ 'float(temperature)' ]

api:
  #encryption:
  #  key: !secret api_key
  services:
    # If using this service, disable the Home Assistant temperature sensor below.
    - service: set_remote_temperature
      variables:
        temperature: float
      then:
        - lambda: 'id(set_current_temperature)->execute(temperature);'

sensor:
  # If using this sensor, don't use the service above.
  - platform: homeassistant
    name: "Temperature Sensor From Home Assistant"
    id: current_temp
    entity_id: sensor.basement_thermostat_temperature
    internal: true
    on_value:
      then:
        - lambda: 'id(set_current_temperature)->execute(x);'
        #- lambda: 'id(hp).set_remote_temperature((x - 32.0) * (5.0 / 9.0));'

```

drewkeller, 2024-06-20:
Just to follow up...

With SETTINGS set to DEBUG, I can see the temperature getting set as below.
```
[D]	[homeassistant.sensor:024]	'sensor.basement_thermostat_temperature': Got state 74.30
[D]	[SETTINGS:067]	setting remote temperature to 23.500002
[I]	[main:174]	Setting current temperature: 74.3F
```

When the heatpump reports its current temperature, it is in increments of .5C (.9F), so you may notice a difference between the heatpump and your sensor, if you're using sensors that show .1F resolution.

closed 2024-06-20 by drewkeller

echavet, 2024-06-20:
> When the heatpump reports its current temperature, it is in increments of .5C (.9F), so you may notice a difference between the heatpump and your sensor, if you're using sensors that show .1F resolution.

Yes that's normal.