echavet/MitsubishiCN105ESPHome #118: Inconsistent Heatpump Checksum Failures

GEMISIS, 2024-06-19:
Hey folks,
Having some issues with a model MSZ-GL06NA model unit where I'm getting inconsistent checksum errors. You can see in this screenshot that at times the checksum is succeeding, but most of the time it is failing:

[image]

I have another one of these units that's working fine with the heatpump implementation at https://github.com/geoffdavis/esphome-mitsubishiheatpump but this one has been a bit more stubborn? Also seeing a lot of cycle timeouts, but I wonder if this is related to the same issue?

echavet, 2024-06-19:
Hi @GEMISIS 
Yes it is related.

I've had the same problem with my first tests with ESP8266 chips. 
This was solved by using an external voltage regulator.
I'm now using esp32 chips which do not suffer from these kinds of problems.

You could switch your materials from one to another to identify if it comes from the cables ou the heatpump itself.

I've had such experience too. The same material was working on 2 of my heatpumps but not on a friend of mine's one. Until I decided to use a voltage regulator.

GEMISIS, 2024-06-19:
Do you have more details on what's going on here?

I ask because I have the exact same AC units, exact same ESP8266 boards, and exact same wiring on two units, and one is working fine while the other is not. I've double checked the voltages on both with a multimeter going into the boards, and they match up almost exactly 1 to 1, so something seems extremely off on this. I've seen this mentioned around other forums, and I'm wondering if some units have a *slightly* different format that maybe needs to be accounted for?

phidauex, 2024-06-20:
A number of people are finding issues with the ESP8266 boards - the issue isn’t a difference in the unit protocol (that is very consistent, and has to be to support all of Mitsubishi’s own accessories), but in small differences in the internal voltage regulators, or just random noise in the units that is slightly worse in some cases than others.

Given the low cost of the devices, switching to an ESP32S3 based unit sounds prudent if you are having issues - I don’t think the problems you are seeing can be fixed in the library.

GEMISIS, 2024-06-20:
Interesting, I've got 6 of these heads in my house and have 4 of them setup with the same ESP8266 boards, and only one of them isn't working right now (though the other units are different models). Does the ESP32S3 have a 5v input for power? Or do I need to build out the regulator circuit for it still? Trying to avoid needing to build out more than necessary where possible  😂

GEMISIS, 2024-06-20:
Alright so super weird thing just happened: Latest esphome broke the previous setup I had using the geoffdavis repo, and so I went ahead and tried flashing this over to the unit again, and now it's suddenly all working?

I'm still seeing some occasional checksum failures, but the unit works exactly as expected now (including with the remote) 😮 The only minor change is that the update interval is now much lower (500ms instead of the default 4s), and so I wonder if maybe it's causing it to "fail faster" on the occasions when the checksum is failing and thus finding a better state more quickly?

In either case, this is super odd, but I guess we can close this out if we want? or maybe it's worth poking around a bit more? I'm a bit nervous to do so since it's working now, but maybe i'll hook up a debug cable to one for fun later and take a peek lol

echavet, 2024-06-21:
> Alright so super weird thing just happened: Latest esphome broke the previous setup I had using the geoffdavis repo, and so I went ahead and tried flashing this over to the unit again, and now it's [...]

Hello @GEMISIS,
thank you for your message. 
Could you provide you yaml file please? I'd like to check wether what you're saying comes from an issue or your config file?

The UART link with the heatpump is very sensitive to noise and voltage variations. Depending on the heatpump internal power supply, there might bee some different behaviors. Any variation may corrupt data and cause checksum error. 
Having a short update interval should not cause any trouble (as far as I imagine!!), because the internal loop is managing a logical cycle (1a/1b, 2a... 6) in which each step if coming after the previous one. 
You can notice that this component (cn105) do not implement the PollingComponent but directly its underlying Component class.  So in this context, if the updateInterval is too short, this won't have any consequences, as the loop will keep processing its current cycle. But as soon as the cycle ends, the next one will begin with no delay. But there are some internal timeout constants which values depends on this update interval. 
One way to set it up correctly is to run it once and to check the cycle duration that is logged. Then you just have to set the update_interval a little higher than the cycle duration.

GEMISIS, 2024-06-21:
Sure, here's the new configuration I'm using that's working fine now:

```
substitutions:
  name: master-bedroom-ac
  friendly_name: Master Bedroom AC

esphome:
  name: master-bedroom-ac
  friendly_name: Master Bedroom AC

# Enable Home Assistant API
api:
  encryption:
    key: "secret"

ota:
  - platform: esphome
    password: "secret"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "MasterBedroomAC Fallback Hotspot"
    password: "secret"

captive_portal:
    
esp8266:
  board: d1_mini

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: TX
  rx_pin: RX

external_components:
  - source: github://echavet/MitsubishiCN105ESPHome

# Disable logging
logger:
  baud_rate: 0

# Sensors with general information.
sensor:
  # Uptime sensor.
  - platform: uptime
    name: "${friendly_name} Uptime"

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: "${friendly_name} WiFi Signal"
    update_interval: 60s

climate:
  - platform: cn105
    name: "${friendly_name}"

    # Optional
    update_interval: 500ms
```

Note again that nothing on the hardware side changed, and I still see both checksum failures and cycle timeouts, but the unit is now receiving proper information and works perfectly. The only change between this configuration and the old one is the update_interval is set to 500ms whereas before it was set to 4s (I had also tried 20s which did not work either).

echavet, 2024-06-28:
I'm closing this one because I think the issue comes from hardware

closed 2024-06-28 by echavet