echavet/MitsubishiCN105ESPHome #119: Feature Request: Add vane positions/powerful/eco modes via additional componets

flybrys, 2024-06-21:
I just stumbled across this fork after some issues with 2024.6 and it has fixed all the issues I had with the original esphome-mitsubishiheatpump. I can now also use an ESP32 and enable Bluetooth Proxy where I needed a separate repeater before. Thank you for your efforts!! If you had a donation link I'd buy you a coffee.

I am using a similar but different custom component for my Panasonic unit which uses the switch and select components to be able to specify an exact vane position and whether to use the Eco or Boost modes. Reference: https://github.com/DomiStyle/esphome-panasonic-ac

Setting the vane position would be super helpful, as the unit blows air on us while watching TV and I need to manually position the vanes after using auto mode.

As the interest in this component has fired up again, I thought I'd try my luck to see if it could be added somehow?

Cheers

flybrys, 2024-06-21:
Just noticed there is a pull request in the previous component to achieve this but not yet merged.
https://github.com/geoffdavis/esphome-mitsubishiheatpump/pull/109

So hopefully that means something similar could be achieved!

eikowagenknecht, 2024-06-22:
As far as I can see, the vane positions can already be controlled, as shown in the example here: https://github.com/echavet/MitsubishiCN105ESPHome?tab=readme-ov-file#climate-component-full-example

It looks like this in HA: 
[image]
[image]

flybrys, 2024-06-22:
Yep, I just found it also. Didn't realise it was already there until I stumbled upon the 2 extra config options in the full example.
Still would be good to get Eco/Boost somehow though. Would that even be possible?

eikowagenknecht, 2024-06-22:
I would think probably not, at least I can not see eco or boost modes referenced anywhere in the underlying library (https://github.com/SwiCago/HeatPump). Also it's not available on MelCloud as far as I can see.

flybrys, 2024-06-22:
Fair enough. Thanks for the replies.
I'll close this as not required. Thanks to all for a great integration!

closed 2024-06-22 by flybrys