echavet/MitsubishiCN105ESPHome #121: Wrong Current temperature reported.

Pierre-33, 2024-06-24:
Hi
I'm using an external sensor for temperature monitoring like in your example:

```yaml
sensor:
  - platform: homeassistant
    name: "Temperature Sensor From Home Assistant"
    entity_id: sensor.clim_salon_temperature_sensor
    on_value:
      then:
        - lambda: 'id(hp).set_remote_temperature(x);'
```

This seems to work as the temperature display in Home Assistant for my heatpump is the same as the one of my sensor (26°), but as soon as a I turn on the heatpump, the temperature drop to 25° (my external sensor stay at 26°), and a few seconds after I turn off the heatpump it get back to 26.

I used the same setup with the original geoffdavis component that worked like a charm, but maybe yours need a slightly different setup?

Pierre-33, 2024-06-24:
Here is the complete Yaml

```yaml
esphome:
  name: clim-bureau
  friendly_name: Clim Bureau

esp8266:
  board: esp01_1m

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3

# Enable logging
logger:
  baud_rate: 0

# Enable Home Assistant API
api:
  encryption:
    key: xxx

ota:
  platform: esphome
  password: xxx

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Clim-Bureau Fallback Hotspot"
    password: xxx

captive_portal:

external_components:
  - source: github://echavet/MitsubishiCN105ESPHome

climate:
  - platform: cn105
    id: hp
    name: None
    update_interval: 4s
    compressor_frequency_sensor:
      name: Compressor Frequency
    vertical_vane_select:
      name: Vertical Vane
    horizontal_vane_select:
      name: Horizontal Vane
    #remote_temperature_timeout: 60min
    debounce_delay : 500ms

sensor:
  - platform: homeassistant
    name: "Temperature Sensor From Home Assistant"
    id: current_temp
    internal: true
    entity_id: sensor.clim_bureau_temperature_sensor
    on_value:
      then:
        - lambda: 'id(hp).set_remote_temperature(x);'

button:
  - platform: restart
    name: "Restart"
    
```

I say the topic about logger granularity in the documentation but I don't know which part would be relevant in debug and I'm affraid activating everything could be overwhelming. Any particular component you want in debug?

echavet, 2024-06-25:
Ok thanks,
First try please to set these options in the climate component. Choose the values you want for min and max. But set the target and current temperature like in this example:

```yam
    visual:
      min_temperature: 18
      max_temperature: 25
      temperature_step:
        target_temperature: 0.5
        current_temperature: 0.1
```

Then set the logger:
```
level: DEBUG
logs:
EVT_SETS : DEBUG
WIFI : INFO
MQTT: INFO
WRITE_SETTINGS: INFO
SETTINGS: DEBUG
STATUS: INFO
CN105Climate: WARN
CN105: INFO
climate: WARN
sensor: WARN
chkSum: INFO
WRITE : WARN
READ : WARN
ACK : DEBUG
Header: INFO
Decoder : DEBUG
CONTROL WANTED_SETTINGS: INFO 
UPDT ITVL: DEBUG
CYCLE: DEBUG
```

Pierre-33, 2024-06-25:
Here is a video of the issue with the logging option you requested :

https://github.com/echavet/MitsubishiCN105ESPHome/assets/81322563/76a7f3bf-af9c-4f9a-8d20-8612b42035be

Protoncek, 2024-06-27:
Hi guys!
I can confirm pretty much similar condition. My behaviour is slightly different, though: my climate cools one degree  lower than set: if i set cooling to 25 it cools to 24, if i set to 26 it cools to 25 etc... only difference from Pierre is that in my case it seems that reported temperature doesn't display 1 degree lower, it really cools down to shown temperature. My logger is pretty much the same as above from Pierre, i also get "wanted" target  -1°C.

Protoncek, 2024-06-27:
Room is definitely not too big… interesting is that i have more climates in my house (in bigger rooms, too) and this is happening only on this one… today i moved xiaomi sensor on other side of the room to see if there will be any difference. Interesting is that difference is very accurate at 1 degree…

I agree with you that integration seems to work properly, since current temperature IS changing; 

I do wonder, however…is there any way to know if climate works on internal sensor or external one? Yaml has “fail-safe” function: if there are no data from external sensor for 30 minutes it switches to internal, but how (if at all) can i know that it did switch to internal?

echavet, 2024-06-27:
Well you can check at the logs for that info it is a fall back situation.
As soon as the heatpump receives a new external temperature it is going back to external mode.

Pierre-33, 2024-06-28:
Ok, so all those discussions about overshooting made me remember that last summer to compensate overshooting I actually wrote an automation that reduce of 1 degree the temperature I feed to the heatpump... so the temperature sensor you saw in my video is not the one I feed to the heat pump and everything is actually working exactly like it should.

**TLDR: This is a mistake of mine and their is not actual issue.**

Since their is a lot of other comment talking about something apparently similar, I let you decide wether you want to close the issue or not.

Thanks for your great integration and your quick answer and sorry about the fake issue :-/

Protoncek, 2024-06-28:
I did some further testing...
First, i installed Geoff's FW, just to double-check: result was also 1 degree lower temperature with external sensor. 

Then i switched to internal sensor for one day. Result was ALSO 1 degree to low! So, i can't think of anything else than poor internal calibrating or something in this direction. Especially since i have another (newer model) climate also from Mitsubishi which cools to decimal of degree correct, while epshome FW is exactly the same (checked, double-checked, triple-checked...)

Now i think I'll "recalibrate" my homeassistant sensor in esphome with filter --> offset.

closed 2024-07-18 by echavet