echavet/MitsubishiCN105ESPHome #122: Current Temperature set to 0 C after attempting remote temperature

bunjg, 2024-06-26:
I've tried to add remote temperature using the sensor automation and the api service and both result in setting my heat pump current temperature to 0 C/32 F.  I could call the api service, but it would revert back to 0 C seconds later.  Heat pump works correctly, in auto mode the cooling will turn on below the actual room temperature.  I rolled back to the code without the remote temperature code and it works (with internal temperature) normally now.

 **YAML**
```
substitutions:
  name: esphome-web-7149dc
  friendly_name: Bedroom Heat Pump

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp32:
  board: esp32dev
  framework:
    type: esp-idf

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: GPIO17
  rx_pin: GPIO16

# Enable logging
logger:
  level: DEBUG
  logs:
    SETTINGS: DEBUG
	WRITE_SETTINGS: DEBUG

# Enable Home Assistant API
api:

# Get temperature from Home Assistant
sensor:
  - platform: homeassistant
    name: "Temperature Sensor From Home Assistant"
    entity_id: sensor.bedroom_button_temperature
    on_value:
      then:
        - lambda: 'id(hp).set_remote_temperature(x);'
    filters: 
      - heartbeat: 10s
      
# Allow Over-The-Air updates
ota:
  - platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:

external_components:
  - source: github://echavet/MitsubishiCN105ESPHome

climate:
  - platform: cn105
    id: hp
    name: "Bedroom Heat Pump"
    icon: mdi:heat-pump
    update_interval: 4s
    visual:
      min_temperature: 15
      max_temperature: 31
      temperature_step:
        target_temperature: 1
        current_temperature: .1
```
 
I've tried the same  sensor automation code on a esp8266 unit with the same results.  Any thoughts on how to resolve?

bunjg, 2024-06-26:
I started typing this up a couple days ago and now see that issue #121 is similar to mine.  I'll try adding "id" to my sensor code and update the logger info, then report back.

Another oddity, I had to completely wipe the esp32 in order to get it working again.  Just removing the sensor automation and re-installing didn't resolve HA reporting 0 C current temperature.  I'm new to esp32, but that didn't make sense to me.

phidauex, 2024-06-26:
Might be a longshot, but are you sure your automation isn't sending the wrong datatype, like a string? I've always had an input variable on the temperature update service that converts the value to float before sending it to the heatpump. You might be able to check your automation traces to see what is being sent to the service as well.

```yaml
  services:
    - service: set_remote_temperature
      variables:
        temperature: float
      then:
        - lambda: 'id(hp).set_remote_temperature(temperature);'
    - service: use_internal_temperature
      then:
        - lambda: 'id(hp).set_remote_temperature(0);'
 ```

bunjg, 2024-06-29:
Still no luck.  I used a ESP8266 unit this time and I can see in the logs that 0 C was stated as the current temperature, but it never re-connected with Home Assistant.  I had to wipe it clean and start with a new build without the external temperature code.

```
substitutions:
  name: esphome-web-5284e0
  friendly_name: Leah Heat Pump

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp8266:
  board: esp01_1m
  
uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3

# Enable logging
logger:
  hardware_uart: UART1
  level: DEBUG
  logs:
    EVT_SETS : DEBUG
    WIFI : INFO
    MQTT: INFO
    WRITE_SETTINGS: INFO
    SETTINGS: DEBUG
    STATUS: INFO
    CN105Climate: WARN
    CN105: INFO
    climate: WARN
    sensor: WARN
    chkSum: INFO
    WRITE : WARN
    READ : WARN
    ACK : DEBUG
    Header: INFO
    Decoder : DEBUG
    CONTROL WANTED_SETTINGS: INFO 
    UPDT ITVL: DEBUG
    CYCLE: DEBUG  

# Enable Home Assistant API
api:

# Allow Over-The-Air updates
ota:
  - platform: esphome

#Pull room temperature from Home Assistant
sensor:
  - platform: homeassistant
    name: "Temperature Sensor From Home Assistant"
    id: current_temp
    internal: true
    entity_id: sensor.esphome_web_f6207c_leah_room_temperature
    on_value:
      then:
        - lambda: 'id(hp).set_remote_temperature((x);'
    filters: 
      - heartbeat: 10s		

# Allow provisioning Wi-Fi via serial
#improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp8266.yaml@main
  import_full_config: true

# To have a "next url" for improv serial
web_server:

external_components:
  - source: github://bunjg/MitsubishiCN105ESPHome

climate:
  - platform: cn105
    id: hp
    name: "Leah Heat Pump"
    icon: mdi:heat-pump
    internal: true
    visual:
      min_temperature: 15
      max_temperature: 31
      temperature_step:
        target_temperature: 1
        current_temperature: 0.1
```

I noticed in the logs that the external temperature was coming in as Fahrenheit, so I tried to convert it in the lambda line:
`        - lambda: 'id(hp).set_remote_temperature((x - 32.0) * (5.0 / 9.0))`
 Still no success, but I can see that the conversion worked correctly.

[CN105_Log_C.txt](https://github.com/user-attachments/files/16038420/CN105_Log_C.txt)
[CN105_Log_F.txt](https://github.com/user-attachments/files/16038421/CN105_Log_F.txt)

echavet, 2024-07-08:
Well it looks like a bug, it's true.
Perhaps a specific behavior due to your hp unit.
Unless it comes from your yaml:
you have an extra "(" in the lambda:

```yaml
 - lambda: 'id(hp).set_remote_temperature((x);'
```

Anyway, I think it is not a good practice to set a 10s heartbeat on the external temperature sensor. 
Setting if to 5 or 10 minutes could make sense. But sending the same temperature every 10 seconds overload the uart for nothing.
Could you try without the extra "("? 
Could you also confirm me that you can manage it to work with no external temperature sensor?

echavet, 2024-07-18:
hey @bunjg 
could you confirm you got it working?

bunjg, 2024-07-18:
Yes, I got it to work but I'm not sure how.  I added more info logger logs, uptime and wifi sensors, web server port number, and time sync with home assistant.  I suspect it was the time sync, but now that it's working I don't want to break it again.

Working YAML:
```
substitutions:
  name: esphome-web-768ac0
  friendly_name: Downstairs Heat Pump

esphome:
  name: ${name}
  friendly_name: ${friendly_name}
  name_add_mac_suffix: false
  project:
    name: esphome.web
    version: '1.0'

esp32:
  board: esp32dev
  framework:
    type: esp-idf

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: GPIO17
  rx_pin: GPIO16

# Enable logging
logger:
  level: INFO
  logs:
    EVT_SETS : INFO
    WIFI : INFO
    MQTT : INFO
    WRITE_SETTINGS : INFO
    SETTINGS : INFO
    STATUS : INFO
    CN105Climate: WARN
    CN105: INFO
    climate: WARN
    sensor: WARN
    chkSum : INFO
    WRITE : WARN
    READ : WARN
    Header: INFO
    Decoder : INFO
    CONTROL_WANTED_SETTINGS: INFO

# Enable Home Assistant API
api:

# Get temperature from Home Assistant
sensor:
  - platform: homeassistant
    id: current_temp
    name: "Temperature Sensor From Home Assistant"
    entity_id: sensor.downstairs_temperature
    internal: true
    on_value:
      then:
        - lambda: 'id(hp).set_remote_temperature((x - 32.0) * (5.0 / 9.0));'
    filters: 
      - heartbeat: 30s
  # Uptime sensor.
  - platform: uptime
    name: ${name} Uptime
  # WiFi Signal sensor.
  - platform: wifi_signal
    name: ${name} WiFi Signal
    update_interval: 60s

# Allow Over-The-Air updates
ota:
  - platform: esphome

# Allow provisioning Wi-Fi via serial
improv_serial:

wifi:
  # Set up a wifi access point
  ap: {}

# In combination with the `ap` this allows the user
# to provision wifi credentials to the device via WiFi AP.
captive_portal:

dashboard_import:
  package_import_url: github://esphome/example-configs/esphome-web/esp32.yaml@main
  import_full_config: true

# Sets up Bluetooth LE (Only on ESP32) to allow the user
# to provision wifi credentials to the device.
esp32_improv:
  authorizer: none

# To have a "next url" for improv serial
web_server:
  port: 80

# Sync time with Home Assistant.
time:
  - platform: homeassistant
    id: homeassistant_time

external_components:
  - source: github://echavet/MitsubishiCN105ESPHome

climate:
  - platform: cn105
    id: hp
    name: "Downstairs Heat Pump"
    icon: mdi:heat-pump
    update_interval: 4s
    visual:
      min_temperature: 15
      max_temperature: 31
      temperature_step:
        target_temperature: 1
        current_temperature: 0.1
```

closed 2024-07-18 by bunjg