echavet/MitsubishiCN105ESPHome #124: OnOff Command

To0wnn, 2024-06-26:
Hi,

Could you maybe add when you give a OnOff command it remembers the last mode and switches to that mode when switching On again

echavet, 2024-06-27:
Hi @To0wnn,
this is a feature request !!
Could you explain your use case with more details ?

To0wnn, 2024-06-27:
Hi @echavet ,

When you control the it via Google Assistant it goes automatically to HEAT  when switched on but the last mode was COOL.

[image]

With the MELCloud i did not had this issue it knew what the last mode was (i think).
Maybe it is a setting somewhere else but that I do not know then.

Protoncek, 2024-06-27:
Yes, i can confirm that using MelCloud state was being remembered and used next time. I know, because i've had to change all my automations when i went from MelCloud to esphome. Although i solved this with sending all commands each time it would be nice feature to have, if possible.

echavet, 2024-06-28:
Ok, I'll have a look...

echavet, 2024-06-28:
ok, I don't know how to implement this.
The only available modes are: 
```python
CLIMATE_MODES = {
    "OFF": ClimateMode.CLIMATE_MODE_OFF,
    "HEAT_COOL": ClimateMode.CLIMATE_MODE_HEAT_COOL,
    "COOL": ClimateMode.CLIMATE_MODE_COOL,
    "HEAT": ClimateMode.CLIMATE_MODE_HEAT,
    "DRY": ClimateMode.CLIMATE_MODE_DRY,
    "FAN_ONLY": ClimateMode.CLIMATE_MODE_FAN_ONLY,
    "AUTO": ClimateMode.CLIMATE_MODE_AUTO,
}
```
as you can see, there is no "ON". So there is no way for me (except a way I did not think to of course) to detect an "ON" command and to restore the previous mode. Unless I add a new service like the set_remote_temperature(), but this one won't be called when you ask you assistant to "set the heatpump ON".

I have checked by setting the control logger 
```yaml 
logger:        
  level: DEBUG
  logs:
    ...
    control: INFO
```

which added a way to check what mode is sent by service:

[image]

This means your google (and my alexa) are sending the mode with the command. 

So, unless anybody helps me to find a nice way to integrate this feature in the code, my advice is to add a routine for setting the heatpump in heat or cool mode, or to find a way to do it within home assistant automations.

darkozma, 2024-06-29:
Hello,
A very interesting thread...

The same thing happens to me, when you launch the 'turn_on' command (from HA or Alexa), the heat pump always turns on in heat mode, regardless of the last state.

What I have done to solve it is to capture the changes in mode (and temperature) with a Flow (permanently saved on disk) in Node-Red and every time I launch the 'turn_on' command, I retrieve the data from the Flow to send it to the heat pump.

I know it's not the best, but it's effective.

Thanks for your job in this awesome component @echavet

To0wnn, 2024-07-04:
Hi @darkozma 
could you maybe share your flow for node-red?

echavet, 2024-07-04:
You can also use a Home Assistant automation.

Just do this :

1. an select sensor from the Parameters -> Devices and Services -> Input menu. Call it "input_select.last_cvc_mode" for exemple, dans define the options modes you want to be able to record (HEAT and COOL).
2. an input switch called "input_boolean.cvc_switch"
3. Then define the automation to trigger the right actions:

```yaml
alias: CVC default mode
description: record the last CVC mode
trigger:
  - platform: device
    device_id: c6aa3a5ba55fab14421728cfeb1083cf
    domain: climate
    entity_id: a61fc827bed633e078425385668d6a78
    type: hvac_mode_changed
    id: HEAT
    to: heat
  - platform: device
    device_id: c6aa3a5ba55fab14421728cfeb1083cf
    domain: climate
    entity_id: a61fc827bed633e078425385668d6a78
    type: hvac_mode_changed
    id: COOL
    to: cool
  - platform: state
    entity_id:
      - input_boolean.cvc_switch
    to: "on"
    id: "ON"
  - platform: state
    entity_id:
      - input_boolean.cvc_switch
    to: "off"
    id: "OFF"
condition: []
action:
  - choose:
      - conditions:
          - condition: trigger
            id:
              - HEAT
        sequence:
          - service: input_select.select_option
            metadata: {}
            data:
              option: HEAT
            target:
              entity_id: input_select.last_cvc_mode
      - conditions:
          - condition: trigger
            id:
              - COOL
        sequence:
          - service: input_select.select_option
            metadata: {}
            data:
              option: COOL
            target:
              entity_id: input_select.last_cvc_mode
      - conditions:
          - condition: trigger
            id:
              - "ON"
        sequence:
          - choose:
              - conditions:
                  - condition: state
                    entity_id: input_select.last_cvc_mode
                    state: HEAT
                sequence:
                  - service: climate.set_hvac_mode
                    target:
                      device_id: c6aa3a5ba55fab14421728cfeb1083cf
                    data:
                      hvac_mode: heat
              - conditions:
                  - condition: state
                    entity_id: input_select.last_cvc_mode
                    state: COOL
                sequence:
                  - service: climate.set_hvac_mode
                    target:
                      device_id: c6aa3a5ba55fab14421728cfeb1083cf
                    data:
                      hvac_mode: cool
      - conditions:
          - condition: trigger
            id:
              - "OFF"
        sequence:
          - service: climate.set_hvac_mode
            target:
              device_id: c6aa3a5ba55fab14421728cfeb1083cf
            data:
              hvac_mode: "off"
mode: single
```

4. Expose the input_boolean.cvc_switch through your smart_home configuration

devalls, 2024-07-05:
> Hi @darkozma could you maybe share your flow for node-red?

Of course! @To0wnn
You must first enable Flow to disk writing by adding this code to settings.js (in addons_config NodeRed folder): 

`contextStorage: {
    default: "file",
    memoryOnly: { module: 'memory' },
    file: { module: 'localfilesystem' }
  },`

This is necessary so as not to lose data in case of HA reboot.

This is a commented sample of my flow to turn on the heat pump recovering the last state:
[Flow_turn_on_heat_pump.txt](https://github.com/user-attachments/files/16112780/Flow_turn_on_heat_pump.txt)

darkozma, 2024-07-05:
> Hi @darkozma could you maybe share your flow for node-red?

Of course! @To0wnn
You must first enable Flow to disk writing by adding this code to settings.js (in addons_config NodeRed folder): 

`contextStorage: {
    default: "file",
    memoryOnly: { module: 'memory' },
    file: { module: 'localfilesystem' }
  },`

This is necessary so as not to lose data in case of HA reboot.

This is a commented sample of my flow to turn on the heat pump recovering the last state:
[Flow_turn_on_heat_pump.txt](https://github.com/user-attachments/files/16112780/Flow_turn_on_heat_pump.txt)

PD: The @echavet solution should also work if you only want to set the last state.

closed 2024-07-08 by echavet