echavet/MitsubishiCN105ESPHome #126: ECONO Cool Mode

marcolino7, 2024-07-01:
HI,
on my Unit's remote, I have a button called ECONO COOL, that I use during night, it's something ligh a low energy mode.
It is possible to add this control?

[image]

Thanks

echavet, 2024-07-01:
Hello,

The nearest mode is "silence mode".
There is no support for specific modes.

marcolino7, 2024-07-01:
> Hello,
> 
> The nearest mode is "quiet mode". There is no support for specific modes.

And How can Activate Silence mode?

echavet, 2024-07-01:
Add a "supports:" section in the climate one with :

```yaml
fan_mode: [AUTO, QUIET, LOW, MEDIUM, HIGH] 
```
Then just chose this fan mode from home assistant climate UI component.