echavet/MitsubishiCN105ESPHome #130: Climate doesn't switch mode when clicked

Protoncek, 2024-07-17:
Hi!
My climate acts "funny": sometimes (not always) it doesn't turn on or off when asked. For climate control i use either "simple thermostat  card" (for manual control) or service calls via automations. 
So, when i (or automation) turn climate to COOL sometimes climate actually stays OFF (climate's LED stays off), while thermostat card shows that climate is ON. Then a second (sometimes third) click on COOL actually turns climate ON. It happens when switching from COOL to OFF, too: climate stays ON while HA card shows that it's OFF. And it's happening on (at least) two of my climates.

I turned on debug log mode on my esp module and i managed to catch such behaviour from OFF to COOL last night at 23:46. Since log is pretty bit (around 30M) i stored it on my server, so here's the link:

[https://gofile-364d594758.cz4.quickconnect.to/sharing/t4I9IAKGv](url)

On this log you can see climate normally turning to COOL with automation at 20:45, then i turned it OFF manually via HA card at 22:01, that was normal, too. At this point i even tried on-off a few times, but i didn't manage to catch this bug. Then i turned it back on again at 23:46 (also manually, via HA card) and then it happened.

Any clues? I see some odd changes of climate mode to 4, then back to 2 etc...

echavet, 2024-07-18:
Hey @Protoncek,
thank you for your message.
I'd like to diagnose the pb but I can't manage to see the log file. It looks like the link is dead.

Protoncek, 2024-07-18:
Hm.. that's odd... try this one:

http://gofile.me/6MYGX/t4I9IAKGv

Thanks!

echavet, 2024-07-18:
thanks, and now I need your full yaml please.

Protoncek, 2024-07-18:
Sure. It's a bit "complicated", since i've added some of my functions, but they shouldn't affect climate work, i'll just explain quickly:
 - first is esp reboot function if there's no info from climate in HA - that's leftover from geoff davis's version, because with that version it happened that after power loss sometimes climate wasn't connect to esp (i didn't get climate's info), so another reboot helped. this automation also checks regularly if climate sends info into HA.
 - second is temperature sensor selection - i have a switch in HA to select either internal or external temp.sensor (external is Xiaomi's temp/humi BLE sensor).
 - then i have sensor which tells me which AP is esp connected to (i have a router and 3 AP's)
 - a "light" sensor - this is LED which flashes according to connection (wifi, HA...)
 - "syslog" is add-on for sending log to synology
```
esphome:
  name: ${name}

substitutions:
  name: pavle_klima_spalnica
  friendly_name: Pavle klima spalnica
  connected_pin: GPIO2

esp8266:
  restore_from_flash: True
  board: esp01_1m

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  reboot_timeout: 5min 
  manual_ip:
    static_ip: 192.168.3.156
    gateway: 192.168.0.1
    subnet: 255.255.252.0
    dns1: 192.168.0.1
    dns2: 192.168.0.1

  ap:
    ssid: "${friendly_name} Hotspot"
    password: !secret ap_password

external_components:
#  - source: github://geoffdavis/esphome-mitsubishiheatpump
   - source: github://echavet/MitsubishiCN105ESPHome
   
captive_portal:

syslog:
  
uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3

# Enable logging
logger:
 level: DEBUG
 hardware_uart: UART1
 logs:
   EVT_SETS : DEBUG
   WIFI : INFO
   MQTT : INFO
   WRITE_SETTINGS : DEBUG
   SETTINGS : DEBUG
   STATUS : INFO
   CN105Climate: WARN
   CN105: DEBUG
   climate: WARN
   sensor: WARN
   chkSum : INFO
   WRITE : WARN
   READ : WARN
   Header: INFO
   Decoder : DEBUG
   CONTROL_WANTED_SETTINGS: DEBUG

[... 184 lines cut ...]

light: # ----------------------------------------------------------------------- LIGHT
  - platform: binary
    id: ${name}_led
    internal: true
    output: status_ledica_out

output:
  - id: status_ledica_out
    platform: gpio
    pin: ${connected_pin}

  # --------------------------------------------------------------- REBOOT
button:
  - platform: restart
    id: ${name}_reboot
    name: ${friendly_name} reboot

switch:   
  - platform: template
    name: "Xiaomi temp.senzor Pavle"     
    id: xiaomi_senzor_pavle
    optimistic: true
    restore_mode: RESTORE_DEFAULT_ON
```

Protoncek, 2024-07-21:
Today i played a bit with this and I remembered something, if it's in any way related:
In my automations i use "climate.turn_off" and "climate.turn_on" commands. I do remember that we talked about library supposely doesn't have "on" and "off", is that correct? 
I tried to turn on climate with direct command "climate.set_hvac_mode" --> "cool" and turn off with "climate.set_hvac_mode" --> "off" and it works. Should i use only these commands and drop "on" and "off" ?

echavet, 2024-07-23:
Hi @Protoncek,
I think it would be a good idea to disable all your automations and specific lambda to see if your issue comes from the firmware or not. Do it temporarily and tell us if you still have pb please.

Protoncek, 2024-07-23:
Ok, i'll do that. It can take a while, though, since, as i said, it happens only occasionally. But i guess that in a couple of weeks i should see... i'll stay in touch.
Thanks for your time!

echavet, 2024-07-23:
I'm looking at your yaml:

```yaml
- platform: homeassistant
    name: "Pavle Xiaomi temperatura"
    id: pavle_xiaomi_temperatura
    internal: false
    device_class: temperature
    unit_of_measurement: "°C"
    entity_id: sensor.pavle_kompenzirana_temperatura
    filters:
      - heartbeat: 5s
    on_value:
      then:
        - if:
            condition:
              - switch.is_on:
                  id: xiaomi_senzor_pavle
            then:
              - lambda: |-
                    if(id(pavle_xiaomi_temperatura).has_state()) {
                      id(pavle_klima).set_remote_temperature(x);
                      ESP_LOGW("temperature", "uporabljam XIAOMI senzor %.2f", x);
                      id(tip_senzorja_pavle).publish_state("xiaomi");
                    } else {
                      id(pavle_klima).set_remote_temperature(0);
                      ESP_LOGW("temperature", "uporabljam interni senzor");
                      id(tip_senzorja_pavle).publish_state("vgrajeni");
                    }
            else:
              - lambda: |-
                      id(pavle_klima).set_remote_temperature(0);
                      ESP_LOGW("temperature", "uporabljam interni senzor");      
                      id(tip_senzorja_pavle).publish_state("vgrajeni");   
```
I have a few observations:

1. I think setting the heartbeat to 5s is not a good thing. For two reasons. 
   - the temperature is not supposed to change so often. 
   - even if I've tried hard to set a mechanism to prevent concurrent calls to the firmware, calling the set_remote_temperature so often might increase the risk of concurrency issue. 

2. If the on_value event is called every heartbeat event (which is the case I think), in case the switch is off, you keep calling set_remote_temperature(0) every 5 s, which will interfere very often with the normal loop.

I won't be surprised that the strange behavior occurs because of a concurrent issue. 
I know it shouldn't but... you know, I try to do the best...

echavet, 2024-07-23:
oh I forgot, you can use the component `remote_temperature_timeout:` to manage external temperature sensor disconnection. It is reliable I think.

Protoncek, 2024-07-23:
Great, i'll change what you propose. In any way i think that 5s is "leftover" from testing when we were discussing 1 degree temperature difference not long ago. I see that i forgot to delete that filter.

echavet, 2024-08-11:
Hey @Protoncek 
Does this issue still needs to be resolved ?

Protoncek, 2024-08-11:
You won't believe...but i just remembered this topic half an hour ago... :-) and decided to "make an end to it"...
Since i didn't have any issues for 3 weeks now i decided to i re-insert "heartbeat" into one of my climates and try: problem immediately re-appeared, so i think that too frequent communication was indeed the culprit. 

Although now i don't use "climate on" and "climate off" commands anymore (i only change hvac mode to off now), i don't think this was the culprit, because i used that commands for a while before without any problem.

Many thanks for your help! I think we can close this issue now.

closed 2024-08-11 by Protoncek