echavet/MitsubishiCN105ESPHome #131: uptime_sensor.h has been removed in 2024.7.0

Matze5800, 2024-07-17:
Compilation fails with 2024.7.0 as uptime_sensor.h is missing.
It has been replaced with uptime_seconds_sensor.h and uptime_timestamp_sensor.h.
This is the breaking change: [https://github.com/esphome/esphome/pull/7029](https://github.com/esphome/esphome/pull/7029)

Thanks a lot for your hard work!

Schlichi83, 2024-07-17:
I have exactly the same issue.

echavet, 2024-07-17:
Hi guys,
Are you using the esp32 framework or the Arduino one?
I haven't committed any modification since a moment. 
So it might be a local issue.

Try to do a clean and rebuild:

```system
esphome clean your_yaml.yaml
```

Mense1978, 2024-07-17:
does not work either

Compiling .pioenvs/aa-mitsubishi-porchemesa/src/esphome/components/api/subscribe_state.cpp.o
Compiling .pioenvs/aa-mitsubishi-porchemesa/src/esphome/components/api/user_services.cpp.o
Compiling .pioenvs/aa-mitsubishi-porchemesa/src/esphome/components/binary_sensor/automation.cpp.o
Compiling .pioenvs/aa-mitsubishi-porchemesa/src/esphome/components/binary_sensor/binary_sensor.cpp.o
Compiling .pioenvs/aa-mitsubishi-porchemesa/src/esphome/components/binary_sensor/filter.cpp.o
Compiling .pioenvs/aa-mitsubishi-porchemesa/src/esphome/components/button/button.cpp.o
Compiling .pioenvs/aa-mitsubishi-porchemesa/src/esphome/components/captive_portal/captive_portal.cpp.o
Compiling .pioenvs/aa-mitsubishi-porchemesa/src/esphome/components/climate/climate.cpp.o
Compiling .pioenvs/aa-mitsubishi-porchemesa/src/esphome/components/climate/climate_mode.cpp.o
Compiling .pioenvs/aa-mitsubishi-porchemesa/src/esphome/components/climate/climate_traits.cpp.o
Compiling .pioenvs/aa-mitsubishi-porchemesa/src/esphome/components/cn105/climateControls.cpp.o
In file included from src/esphome.h:35,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/climateControls.cpp:1:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/uptime_sensor.h: No such file or directory
    2 | #include "esphome/components/uptime/uptime_sensor.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/aa-mitsubishi-porchemesa/src/esphome/components/cn105/climateControls.cpp.o] Error 1
========================== [FAILED] Took 7.49 seconds ==========================

c-mholm, 2024-07-17:
I get the same error. I have ESP-IDF framework. I have tried to clean build files and reinstall, but still get the same error.

revoka, 2024-07-17:
I can confirm this error shows up while compiling with latest ESP home 2024.7.0
It is caused by a breaking change in the uptime sensor of ESP Home latest release as mentioned in the opening post by Matze5800.
Check the link he has provided in the first post about the changes made in the latest ESP Home these changes break the code.
The code needs to be adapted to the new implementation of the uptime sensor.
Temporary solution for new users is to downgrade to last 2024.6 version of ESP Home for now.

phidauex, 2024-07-17:
This problem is a bit odd - the feature change is listed as "new feature/non-breaking change" so it shouldn't be expected to break current configurations. Either it was mislabeled/incorrectly tested on the ESPHome side, or there is a regression that needs to be addressed. https://github.com/esphome/esphome/pull/7029

Anyone more familiar with the build able to confirm if the required file changes were intentional? Or is this something that will get fixed in 2024.7.1?

ESPHome has been introducing a lot of breaking changes recently, it seems a bit chaotic for the integration developers.

echavet, 2024-07-17:
> esphome/esphome#7029 <https://github.com/esphome/esphome/pull/7029>

Ok thanks for the info.
I'll check that tomorrow.

ghisch, 2024-07-17:
Hey, the fix is simple enough that I could do it. Made a PR for it
https://github.com/echavet/MitsubishiCN105ESPHome/pull/133

uspino2, 2024-07-18:
Hello, I know this issue looks as fixed, and thank you for that, but how do I upload the fix to the device? (Sorry, not too used to ESPHome yet). These are the messages I get when trying to update wirelessly to 2024.7.0. Thanks.

```
Compiling .pioenvs/mitsubishi-esp32-office/src/esphome/components/cn105/cycle_management.cpp.o
In file included from src/esphome.h:33,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/climateControls.cpp:1:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/uptime_sensor.h: No such file or directory
 #include "esphome/components/uptime/uptime_sensor.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/mitsubishi-esp32-office/src/esphome/components/cn105/climateControls.cpp.o] Error 1
In file included from src/esphome.h:33,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/cycle_management.cpp:2:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/uptime_sensor.h: No such file or directory
 #include "esphome/components/uptime/uptime_sensor.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
In file included from src/esphome.h:33,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/cn105.cpp:2:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/uptime_sensor.h: No such file or directory
 #include "esphome/components/uptime/uptime_sensor.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/mitsubishi-esp32-office/src/esphome/components/cn105/cycle_management.cpp.o] Error 1
*** [.pioenvs/mitsubishi-esp32-office/src/esphome/components/cn105/cn105.cpp.o] Error 1
In file included from src/esphome.h:33,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/componentEntries.cpp:1:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/uptime_sensor.h: No such file or directory
 #include "esphome/components/uptime/uptime_sensor.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/mitsubishi-esp32-office/src/esphome/components/cn105/componentEntries.cpp.o] Error 1
```

revoka, 2024-07-18:
Hi uspino2,

Commits have been made by a pull request but the need to make it to the PR.
The software that is pulled from github during compilation is still the version that contains the old code.
That is the reason you still see the errors while trying to compile.

uspino2, 2024-07-18:
> Hi uspino2,
> 
> Commits have been made by a pull request but the need to make it to the PR. The software that is pulled from github during compilation is still the version that contains the old code. That is the reason you still see the errors while trying to compile.

I guess it's a matter of waiting now. Thank you so much!

PR #132 (2024-07-18): Quick attempt at fixing compatibility with ESPHome 2024.7.0
Based on a cursory look at the changes done, this should get this component compiling again.

Fixes #131
closed 2024-07-18 by echavet