echavet/MitsubishiCN105ESPHome #134: Compilation problem with ESPHome 2024.7.0 ?

Ben0it, 2024-07-18:
Since ESPHome 2024.7.0, compilation fails.

I did a clean-up of build files and tried full compilation again, without success :

[image]

Matze5800, 2024-07-18:
This is fixed now, please try again.
See [https://github.com/echavet/MitsubishiCN105ESPHome/issues/131](https://github.com/echavet/MitsubishiCN105ESPHome/issues/131)

Ben0it, 2024-07-18:
Thanks for pointing me to this, I made the mistake of searching in open issues only.

I still have the problem despite build file cleanups from web interface. It seems the previous cn105 code is used.

Is there a more effective way of cleaning or updating the external component cache ?

Web UI cleanup results into this : 
[image]

Matze5800, 2024-07-18:
You could try setting the refresh time temporarily to 0s, this should force an update of the local cache.

Like this:
```
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome
    refresh: 0s
```

Ben0it, 2024-07-18:
Thank you very much for the `refresh` attribute, that did the trick :)

closed 2024-07-18 by Ben0it