echavet/MitsubishiCN105ESPHome #135: Touche commande des Ailettes ( 2 Flow/ 1 Flow)

roumano, 2024-07-18:
In my air-conditioning ( MFZ-KT35VG ), i don't have horizontal pan.

But i can, with the remote, open or close horizontal air flow

In the documentation, it's the button : `2 Flow / 1 Flow`

Look like it's not (yet) possible to manage it with this code.

[image]

Do you think, it's option can be added ?

echavet, 2024-07-19:
Hi @Roumano,
thanks for you message. J'ai un peu envie de te répondre en français du coup!! 
I can't add this specific flow mode because I can't guess the digital value related to this flow mode.
You could try to find it by setting the `READ`, `Decoder` and LOGS level to debug and see what happens when you set the mode with your remote. Decoding such specific features is impossible without the specific unit as the mitsubitshi protocole is not public. We (actually that's what SwiCago did) have to practice reverse engineering.

roumano, 2024-07-19:
En français, ça sera plus simple du coup ;)

J'arrive pas à déployer le code esp si j'active le logger

* Message d'erreur lors de la compilation ( j'ai penser a faire un clean du build) : 
```
Compiling .pioenvs/stationmeteo/src/esphome/components/cn105/cn105.cpp.o
In file included from src/esphome.h:35,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/climateControls.cpp:1:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/uptime_seconds_sensor.h: No such file or directory
 #include "esphome/components/uptime/uptime_seconds_sensor.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/stationmeteo/src/esphome/components/cn105/climateControls.cpp.o] Error 1
In file included from src/esphome.h:35,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/cn105.cpp:2:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/uptime_seconds_sensor.h: No such file or directory
 #include "esphome/components/uptime/uptime_seconds_sensor.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/stationmeteo/src/esphome/components/cn105/cn105.cpp.o] Error 1
========================= [FAILED] Took 46.75 seconds =========================
```

Code rajouter : 
```
# Enable logging
logger:
  #level: DEBUG
  #level: verbose
  baud_rate: 0
  level: DEBUG
  logs:
    EVT_SETS : INFO
    WIFI : INFO
    MQTT : INFO
    WRITE_SETTINGS : INFO
    SETTINGS : INFO
    STATUS : INFO
    CN105Climate: WARN
    CN105: INFO
    climate: WARN
    sensor: WARN
    chkSum : INFO
    WRITE : WARN
    READ : DEBUG
    Header: INFO
    Decoder : DEBUG
    CONTROL_WANTED_SETTINGS: INFO
```

echavet, 2024-07-19:
Ce n'est pas ta faute, c'est depuis la mise à jour de ESPHome. Ca ne compilait plus (#134). 
On vient de faire un commit avec correction, mais ton intégration espHome utilise encore le cache.
rajoute l'option `refresh: 0s`pour forcer la récupération des dernières modifs.

```yaml
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome
    refresh: 0s
```

roumano, 2024-07-19:
> Ce n'est pas ta faute, c'est depuis la mise à jour de ESPHome. Ca ne compilait plus (#134). On vient de faire un commit avec correction, mais ton intégration espHome utilise encore le cache. rajoute [...]

C'est aller un peut plus loin (apres aussi un clean build) mais ca plante toujours : 
```
Compiling .pioenvs/stationmeteo/src/esphome/components/cn105/climateControls.cpp.o
In file included from src/esphome.h:35,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/climateControls.cpp:1:
src/esphome/components/cn105/uptime_connection_sensor.h:2:10: fatal error: esphome/components/uptime/uptime_seconds_sensor.h: No such file or directory
 #include "esphome/components/uptime/uptime_seconds_sensor.h"
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
*** [.pioenvs/stationmeteo/src/esphome/components/cn105/climateControls.cpp.o] Error 1
========================= [FAILED] Took 39.89 seconds =========================
```

revoka, 2024-07-19:
I think you are using an an older version of ESP Home then the latest 2024.7.0 version.
Versions prior to 2024.7.0 don´t have the new uptime_seconds_sensor defined and that is what your error is during compilation.
Upgrading to ESP Home latest version should solve your compilation problem.

roumano, 2024-07-19:
> I think you are using an older version of ESP Home than the latest 2024.7.0 version. Versions prior to 2024.7.0 don´t have the new uptime_seconds_sensor defined and that is what your error is during compilation. Upgrading to ESP Home latest version should solve your compilation problem.

Correct ( i was using the 2024.4.xxx , ...

Please found information i have found with the debug mode : 

When it's set to 1flow, i get : 
```
[D][READ:157]: FC 62 01 30 10 03 00 00 0A 00 B4 A8 A8 FE 42 00 00 14 7B 00 00 7D
[D][Decoder:239]: [Room °C: 20.000000]

[D][READ:157]: FC 62 01 30 10 03 00 00 0A 00 B4 A8 A8 FE 42 00 00 14 7B 00 00 7D
[D][Decoder:239]: [Room °C: 20.000000]

[D][READ:157]: FC 62 01 30 10 03 00 00 0A 00 B2 A8 A8 FE 42 00 00 14 7B 00 00 7F
[D][Decoder:239]: [Room °C: 20.000000]

[D][READ:157]: FC 62 01 30 10 03 00 00 0A 00 B4 A8 A8 FE 42 00 00 14 7B 00 00 7D
[D][Decoder:239]: [Room °C: 20.000000]

[D][READ:157]: FC 62 01 30 10 03 00 00 0A 00 B4 A8 A8 FE 42 00 00 14 7B 00 00 7D
[D][Decoder:239]: [Room °C: 20.000000]

[D][READ:157]: FC 62 01 30 10 03 00 00 0A 00 B4 A8 A8 FE 42 00 00 14 7C 00 00 7C
[D][Decoder:239]: [Room °C: 20.000000]
```

When i use 2 flow : 

```
[D][READ:157]: FC 62 01 30 10 03 00 00 0B 00 B2 AA AA FE 42 00 00 14 7A 00 00 7B
[D][Decoder:239]: [Room °C: 21.000000]

[D][READ:157]: FC 62 01 30 10 03 00 00 0A 00 B2 A9 A9 FE 42 00 00 14 7A 00 00 7E
[D][Decoder:239]: [Room °C: 20.500000]

[D][READ:157]: FC 62 01 30 10 03 00 00 0A 00 B2 A9 A9 FE 42 00 00 14 7A 00 00 7E
[D][Decoder:239]: [Room °C: 20.500000]

[D][READ:157]: FC 62 01 30 10 03 00 00 0A 00 B2 A9 A9 FE 42 00 00 14 7A 00 00 7E
[D][Decoder:239]: [Room °C: 20.500000]

[D][READ:157]: FC 62 01 30 10 03 00 00 0A 00 B2 A9 A8 FE 42 00 00 14 7A 00 00 7F
[D][Decoder:239]: [Room °C: 20.500000]

[D][READ:157]: FC 62 01 30 10 03 00 00 0A 00 B2 A9 A8 FE 42 00 00 14 7A 00 00 7F
[D][Decoder:239]: [Room °C: 20.500000]
```

I've try to get same temperature, but it's changing relatively quickly so it's not easy , ...

It's sufficient as this or not ?

roumano, 2024-07-23:
c'est bon comme info ou pas ? ( je part bientôt en vacances et je pourrais avoir d'autres données avant longtemps , ... )

echavet, 2024-07-23:
Salut,
Désolé j'ai zappé!
Non, sauf erreur (je vérifierai plus tard) je déduis de la trace que les trames reçues sont la température de la pièce.
J'ai du me tromper en te donnant le setup de logs.

La boucle globale enchaîne des requêtes auprès du climatiseur. La réponse du climatiseur est une trame qui représente des types de données particuliers. C'est cette réponse qu'il faut tracer. Je suis sur mon tel là, et je pars bientôt en vacances moi aussi. 
Pas d'urgence. Je regarde le code et te précise la conf dès que possible.

echavet, 2024-08-14:
salut @roumano,
je rentre de vacances et je jette un oeil aux issues ouvertes... 
La première chose (que je ne t'ai pas demandée) c'est ton yaml. Parce qu'en regardant de plus près je suis étonné que tu ne reçoives que des trames de température.
Peut-être as-tu toi aussi ajouté un filtre heartbeat avec un délai très court dans ton yaml ?
Autrement où en es-tu ? As-tu évolué ou es-tu toujours en attente ?

roumano, 2024-08-14:
Salut @echavet ,

Voici le code complet actuel : 

```
substitutions:

  name: stationmeteo
  tx_pin_uart0: GPIO17
  rx_pin_uart0: GPIO16
  modbus_adress: '1'
  modbus_adress2: '2'
  modbus_adress3: '3'
  baudrate: '9600'
  updates: 1s
  
  tx_pin_uart1: GPIO18
  rx_pin_uart1: GPIO19
  baudrate1: '9600'

  tx_pin_uart_clim: GPIO3
  rx_pin_uart_clim: GPIO1

esphome:
  name: ${name}
  friendly_name: Station Météo

  includes:
#    - include/veml6075_custom_sensor.h
#    - include/tx23_custom_sensor.h
  libraries:
#    - "https://github.com/adafruit/Adafruit_VEML6075"
#    - "https://github.com/adafruit/Adafruit_BusIO"
#    - "https://github.com/egzumer/Arduino-LaCrosse-TX23-Library"

esp32:
  board: esp32dev
  framework:
    type: arduino

# Enable logging
logger:
  #level: DEBUG
  #level: verbose
  baud_rate: 0
  level: INFO
  logs:
    EVT_SETS : INFO
    WIFI : INFO
    MQTT : INFO
    WRITE_SETTINGS : INFO
    SETTINGS : INFO
    STATUS : INFO
    CN105Climate: INFO
    CN105: INFO
    climate: INFO
    sensor: INFO
    chkSum : INFO
    WRITE : INFO
    READ : INFO
    Header: INFO
    Decoder : INFO
    CONTROL_WANTED_SETTINGS: INFO

# Enable Home Assistant API
api:
  encryption:
    key: "yyyyyyyyyyyyy"
  reboot_timeout: 0s

[... 182 lines cut ...]
      min_temperature: 16
      max_temperature: 31
      temperature_step:
        target_temperature: 1
        current_temperature: 0.1
    vertical_vane_select:
      name: Clim Palette Verticale
    isee_sensor:
      name: Clim ISEE Sensor
    stage_sensor:
      name: Stage Sensor
    sub_mode_sensor:
      name: Sub Mode Sensor
    auto_sub_mode_sensor:
      name: Auto Sub Mode Sensor
    remote_temperature_timeout: 30min
    debounce_delay : 500ms
    update_interval: 4s
```

J'ai bien sur retirer le niveau de log `level: DEBUG` car trop verbeux pour le laisser en permanance , ... 

Je suis toujours en attente, j'aimerais bien pouvoir activer les ailettes via HA , ...

Bonne journee

roumano, 2024-08-14:
Ce esp fait aussi station meteo (avant de gerer la clim) donc une bonne partie du code est inutile pour la clim, mais je l'ai poster

roumano, 2024-10-01:
Salut Eric, il te faut quoi comme information pour qu'on puisse rajouter cette fonctionnalité ? 
Cdt

echavet, 2024-10-01:
Salut,
Il faudrait repérer lors de la réception de paquets dans la méthode ou fonction (je sais plus) getSettingsFromResponsePacket() s'il y a des octets qui semblent correspondre à cette fonction.
Donc il faut activer les logs comme "Décoder" en mode debug, puis repérer les valeurs qui changent lorsque tu changes le réglage avec la télécommande.

Si tu identifies correctement les correspondances, on pourra envisager de l'implémenter.

TomasDJo, 2024-10-01:
Following. I have a floor mounted mitsubishi heatpump with 2flow/1flow setting on the remote.

I'll see if I can bring up my logs as well with 1flow/2flow function activated and deactivated.

roumano, 2024-10-01:
> Following. I have a floor mounted mitsubishi heatpump with 2flow/1flow setting on the remote.
> 
> I'll see if I can bring up my logs as well with 1flow/2flow function activated and deactivated.

Look like, it's need this part of log : (i'm not 100% sure) 
```
logger:
  level: DEBUG
  logs:
    EVT_SETS : DEBUG
    WIFI : INFO
    MQTT : INFO
    WRITE_SETTINGS : INFO
    SETTINGS : INFO
    STATUS : INFO
    CN105Climate: INFO
    CN105: INFO
    climate: INFO
    sensor: INFO
    chkSum : INFO
    WRITE : DEBUG
    READ : DEBUG
    Header: INFO
    Decoder : DEBUG
    CONTROL_WANTED_SETTINGS: INFO
```

roumano, 2024-11-14:
Salut @echavet ,

Voici des logs mais j'arrive pas a les interpreter et/ou trouver la bonne info dedans : 
```

[17:07:02][I][CYCLE:039]: 1: Cycle start
[17:07:02][D][CYCLE:311]: 2a: Sending settings request (0x02)
[17:07:02][D][WRITE:170]: FC 42 01 30 10 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7B 
[17:07:02][D][READ:170]: FC 62 01 30 10 02 00 00 01 07 07 03 03 00 00 83 B0 28 80 00 00 6B 
[17:07:02][D][CYCLE:317]: 2b: Receiving settings response
[17:07:02][D][Decoder:173]: [0x02 is settings]
[17:07:02][D][Decoder:181]: [Power : ON]
[17:07:02][D][Decoder:182]: [iSee  : 0]
[17:07:02][D][Decoder:183]: [Mode  : FAN]
[17:07:02][D][Decoder:195]: [Temp °C: 24.000000]
[17:07:02][D][Decoder:198]: [Fan: 2]
[17:07:02][D][Decoder:201]: [Vane: —]
[17:07:02][D][EVT_SETS:070]: [wanted]-> [power: -, target °C: -1.0, mode: -, fan: -, vane: -, wvane: -, hasChanged ? ->  NO, hasBeenSent ? ->  NO]
[17:07:02][D][CYCLE:320]: 3a: Sending room °C request (0x03)
[17:07:02][D][WRITE:170]: FC 42 01 30 10 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7A 
[17:07:02][W][component:237]: Component esphome.coroutine took a long time for an operation (58 ms).
[17:07:02][W][component:238]: Components should block for at most 30 ms.
[17:07:02][D][READ:170]: FC 62 01 30 10 03 00 00 08 00 A0 A4 A4 FE 42 00 00 8D CC 00 00 D1 
[17:07:02][D][CYCLE:326]: 3b: Receiving room °C response
[17:07:02][D][Decoder:255]: [Room °C: 18.000000]
[17:07:02][D][Decoder:256]: [OAT  °C: 16.000000]
[17:07:02][D][CYCLE:329]: 4a: Sending status request (0x06)
[17:07:02][D][WRITE:170]: FC 42 01 30 10 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 77 
[17:07:03][D][READ:170]: FC 62 01 30 10 06 00 00 00 00 00 00 06 72 00 00 42 00 00 00 00 9D 
[17:07:03][D][CYCLE:347]: 4b: Receiving status response
[17:07:03][D][Decoder:279]: [0x06 is status]
[17:07:03][D][CYCLE:351]: 5a: Sending power request (0x09)
[17:07:03][D][WRITE:170]: FC 42 01 30 10 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 74 
[17:07:03][D][READ:170]: FC 62 01 30 10 09 00 00 00 03 00 00 00 00 00 00 00 00 00 00 00 51 
[17:07:03][D][CYCLE:366]: 5b: Receiving Power/Standby response
[17:07:03][D][Decoder:148]: [0x09 is sub modes]
[17:07:03][D][Decoder:155]: [Stage : MEDIUM]
[17:07:03][D][Decoder:156]: [Sub Mode  : NORMAL]
[17:07:03][D][Decoder:157]: [Auto Mode Sub Mode  : AUTO_OFF]
[17:07:03][D][text_sensor:064]: 'Stage Sensor': Sending state 'MEDIUM'
[17:07:03][D][text_sensor:064]: 'Sub Mode Sensor': Sending state 'NORMAL'
[17:07:03][D][text_sensor:064]: 'Auto Sub Mode Sensor': Sending state 'AUTO_OFF'
[17:07:03][I][CYCLE:053]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)
[17:07:03][W][component:237]: Component esphome.coroutine took a long time for an operation (54 ms).
[17:07:03][W][component:238]: Components should block for at most 30 ms.
[17:07:04][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[17:07:06][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[17:07:07][I][CYCLE:039]: 1: Cycle start
[17:07:07][D][CYCLE:311]: 2a: Sending settings request (0x02)
[17:07:07][D][WRITE:170]: FC 42 01 30 10 02 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7B 
[17:07:07][D][READ:170]: FC 62 01 30 10 02 00 00 01 07 07 03 03 00 00 83 B0 28 80 00 00 6B 
[17:07:07][D][CYCLE:317]: 2b: Receiving settings response
[17:07:07][D][Decoder:173]: [0x02 is settings]
[17:07:07][D][Decoder:181]: [Power : ON]
[17:07:07][D][Decoder:182]: [iSee  : 0]
[17:07:07][D][Decoder:183]: [Mode  : FAN]
[17:07:07][D][Decoder:195]: [Temp °C: 24.000000]
[17:07:07][D][Decoder:198]: [Fan: 2]
[17:07:07][D][Decoder:201]: [Vane: —]
[17:07:07][D][EVT_SETS:070]: [wanted]-> [power: -, target °C: -1.0, mode: -, fan: -, vane: -, wvane: -, hasChanged ? ->  NO, hasBeenSent ? ->  NO]
[17:07:07][D][CYCLE:320]: 3a: Sending room °C request (0x03)
[17:07:07][D][WRITE:170]: FC 42 01 30 10 03 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 7A 
[17:07:07][W][component:237]: Component esphome.coroutine took a long time for an operation (59 ms).
[17:07:07][W][component:238]: Components should block for at most 30 ms.
[17:07:07][D][modbus_controller.sensor:025]: Sensor new state: 2.00
[17:07:07][D][modbus_controller.sensor:025]: Sensor new state: 90.00
[17:07:07][D][READ:170]: FC 62 01 30 10 03 00 00 08 00 A0 A4 A4 FE 42 00 00 8D CC 00 00 D1 
[17:07:07][D][CYCLE:326]: 3b: Receiving room °C response
[17:07:07][D][Decoder:255]: [Room °C: 18.000000]
[17:07:07][D][Decoder:256]: [OAT  °C: 16.000000]
[... 128 lines cut ...]
[17:07:23][D][WRITE:170]: FC 42 01 30 10 06 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 77 
[17:07:23][D][text_sensor:064]: 'Uptime': Sending state '6m 39s '
[17:07:23][D][READ:170]: FC 62 01 30 10 06 00 00 00 00 00 00 06 72 00 00 42 00 00 00 00 9D 
[17:07:23][D][CYCLE:347]: 4b: Receiving status response
[17:07:23][D][Decoder:279]: [0x06 is status]
[17:07:23][D][CYCLE:351]: 5a: Sending power request (0x09)
[17:07:23][D][WRITE:170]: FC 42 01 30 10 09 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 74 
[17:07:23][D][READ:170]: FC 62 01 30 10 09 00 00 00 03 00 00 00 00 00 00 00 00 00 00 00 51 
[17:07:23][D][CYCLE:366]: 5b: Receiving Power/Standby response
[17:07:23][D][Decoder:148]: [0x09 is sub modes]
[17:07:23][D][Decoder:155]: [Stage : MEDIUM]
[17:07:23][D][Decoder:156]: [Sub Mode  : NORMAL]
[17:07:23][D][Decoder:157]: [Auto Mode Sub Mode  : AUTO_OFF]
[17:07:23][D][text_sensor:064]: 'Stage Sensor': Sending state 'MEDIUM'
[17:07:23][D][text_sensor:064]: 'Sub Mode Sensor': Sending state 'NORMAL'
[... cut ...] 'AUTO_OFF'
[17:07:23][I][CYCLE:053]: 6: Cycle ended in 1.0 seconds (with timeout?:  NO)
[17:07:23][W][component:237]: Component esphome.coroutine took a long time for an operation (54 ms).
[17:07:23][W][component:238]: Components should block for at most 30 ms.
[17:07:24][D][modbus_controller.sensor:025]: Sensor new state: 0.00
[17:07:26][D][modbus_controller.sensor:025]: Sensor new state: 0.00
```

vers 17h07m01s, j'ai activer via la télécommande le mode 2 flow
et vers 17h07h15s, le mode 1 flow

roumano, 2024-11-28:
@echavet , Hi, my logs is sufficient or you need any other informations/logs
have a nice day

echavet, 2024-11-28:
sorry I don't have time for now

roumano, 2025-08-13:
@somlefant ,
i've done it with : 
```
logger:
  baud_rate: 0
  level: DEBUG
  logs:
    EVT_SETS : INFO
    WIFI : INFO
    MQTT : INFO
    WRITE_SETTINGS : INFO
    SETTINGS : INFO
    STATUS : INFO
    CN105Climate: WARN
    CN105: INFO
    climate: WARN
    sensor: WARN
    chkSum : INFO
    WRITE : WARN
    READ : DEBUG
    Header: INFO
    Decoder : DEBUG
    CONTROL_WANTED_SETTINGS: INFO
```

Please found all RED i've capture : 
```
[11:22:00][D][READ:172]: FC 62 01 30 10 02 00 00 01 03 09 06 07 00 00 83 AC 46 80 00 00 4C 
[11:22:00][D][READ:172]: FC 62 01 30 10 03 00 00 10 00 C4 B4 B4 FE 42 00 01 4F 33 00 00 5B 
[11:22:13][D][READ:172]: FC 62 01 30 10 02 00 00 01 03 09 06 04 00 00 83 AC 46 80 00 00 4F 
[11:22:13][D][READ:172]: FC 62 01 30 10 03 00 00 10 00 C4 B4 B4 FE 42 00 01 4F 33 00 00 5B 
[11:22:26][D][READ:172]: FC 62 01 30 10 02 00 00 01 03 09 06 04 00 00 83 AC 46 80 00 00 4F 
[11:22:26][D][READ:172]: FC 62 01 30 10 03 00 00 10 00 C4 B4 B4 FE 42 00 01 4F 33 00 00 5B 
[11:22:39][D][READ:172]: FC 62 01 30 10 02 00 00 01 03 09 06 04 00 00 83 AC 46 80 00 00 4F 
[11:22:39][D][READ:172]: FC 62 01 30 10 03 00 00 10 00 C2 B4 B4 FE 42 00 01 4F 33 00 00 5D 
[11:22:52][D][READ:172]: FC 62 01 30 10 02 00 00 01 03 09 06 07 00 00 83 AC 46 80 00 00 4C 
[11:22:52][D][READ:172]: FC 62 01 30 10 03 00 00 10 00 C2 B4 B4 FE 42 00 01 4F 34 00 00 5C 
[11:23:05][D][READ:172]: FC 62 01 30 10 02 00 00 01 03 09 06 07 00 00 83 AC 46 80 00 00 4C 
[11:23:05][D][READ:172]: FC 62 01 30 10 03 00 00 10 00 C2 B4 B4 FE 42 00 01 4F 34 00 00 5C 
```

Or if you prefer, all debug of the esphome (i've try to clean/remove unsed part of it as i'm also using it as a weather station, ... )

```
[11:21:56][I][app:137]: ESPHome version 2025.6.3 compiled on Aug 13 2025, 11:14:54
[11:21:56][C][wifi:613]: WiFi:
[11:21:56][C][logger:215]: Logger:
[11:21:56][C][logger:215]:   Max Level: DEBUG
[11:21:56][C][logger:215]:   Initial Level: DEBUG
[11:21:56][C][logger:220]:   Log Baud Rate: 0
[11:21:56][C][logger:220]:   Hardware UART: UART0
[11:21:56][C][logger:224]:   Task Log Buffer Size: 768
[11:21:56][C][logger:229]:   Level for 'CN105': INFO
[11:21:56][C][logger:229]:   Level for 'CN105Climate': WARN
[11:21:56][C][logger:229]:   Level for 'CONTROL_WANTED_SETTINGS': INFO
[11:21:56][C][logger:229]:   Level for 'Decoder': DEBUG
[11:21:56][C][logger:229]:   Level for 'EVT_SETS': INFO
[11:21:56][C][logger:229]:   Level for 'Header': INFO
[11:21:56][C][logger:229]:   Level for 'MQTT': INFO
[11:21:56][C][uart.arduino_esp32:151]: UART Bus 0:
[11:21:56][C][uart.arduino_esp32:152]:   TX Pin: GPIO17
[11:21:56][C][uart.arduino_esp32:153]:   RX Pin: GPIO16
[11:21:56][C][uart.arduino_esp32:155]:   RX Buffer Size: 256
[11:21:56][C][uart.arduino_esp32:162]:   Baud Rate: 9600 baud
[11:21:56][C][uart.arduino_esp32:162]:   Data Bits: 8
[11:21:56][C][uart.arduino_esp32:162]:   Parity: NONE
[11:21:56][C][uart.arduino_esp32:162]:   Stop bits: 1
[11:21:56][C][uart.arduino_esp32:151]: UART Bus 1:
[11:21:56][C][uart.arduino_esp32:152]:   TX Pin: GPIO18
[11:21:56][C][uart.arduino_esp32:153]:   RX Pin: GPIO19
[... 167 lines cut ...]
[11:23:03][D][modbus_controller.sensor:025]: Sensor new state: 809.00
[11:23:04][D][modbus_controller.sensor:025]: Sensor new state: 4.00
[11:23:04][D][modbus_controller.sensor:025]: Sensor new state: 180.00
[11:23:04][D][modbus_controller.sensor:025]: Sensor new state: 2.00
[11:23:05][I][CYCLE:039]: 1: Cycle start
[11:23:05][D][CYCLE:378]: 2a: Sending settings request (0x02)
[11:23:05][D][READ:172]: FC 62 01 30 10 02 00 00 01 03 09 06 07 00 00 83 AC 46 80 00 00 4C 
[11:23:05][D][CYCLE:416]: 2b: Receiving settings response
[11:23:05][D][Decoder:209]: [0x02 is settings]
[11:23:05][D][Decoder:216]: [Power : ON]
[11:23:05][D][Decoder:217]: [iSee  : 0]
[11:23:05][D][Decoder:218]: [Mode  : COOL]
[11:23:05][D][Decoder:232]: [Temp °C: 22.000000]
[11:23:05][D][Decoder:235]: [Fan: 4]
[11:23:05][D][Decoder:238]: [Vane: SWING]
[11:23:05][D][Decoder:244]: [wideVane: | (adj:1)]
[11:23:05][D][CYCLE:419]: 3a: Sending room °C request (0x03)
[11:23:05][D][READ:172]: FC 62 01 30 10 03 00 00 10 00 C2 B4 B4 FE 42 00 01 4F 34 00 00 5C 
[11:23:05][D][CYCLE:425]: 3b: Receiving room °C response
[11:23:05][D][Decoder:317]: [Room °C: 26.000000]
[11:23:05][D][Decoder:318]: [OAT  °C: 33.000000]
[11:23:05][I][STATUS:134]: [received]-> [room C°: 26.0, operating: NO , compressor freq: nan Hz]
[11:23:05][I][STATUS:134]: [current]-> [room C°: 26.0, operating: NO , compressor freq: nan Hz]
[11:23:05][D][CYCLE:431]: 3c: Sending HVAC options request (0x42)
```