echavet/MitsubishiCN105ESPHome #138: Cmake errors

tmchow, 2024-07-28:
I'm getting cmake errors just doing initial setup.. not sure where to go from here.  I'm using an Atom S3 lite ESP32 device:

```
Installing collected packages: pyparsing, urllib3, tqdm, six, pyyaml, pycparser, packaging, msgpack, kconfiglib, idna, future, filelock, contextlib2, colorama, click, charset-normalizer, certifi, schema, requests, cffi, requests-toolbelt, requests-file, cryptography, cachecontrol, idf-component-manager
Successfully installed cachecontrol-0.14.0 certifi-2024.7.4 cffi-1.16.0 charset-normalizer-3.3.2 click-8.1.7 colorama-0.4.6 contextlib2-21.6.0 cryptography-3.4.8 filelock-3.15.4 future-1.0.0 idf-component-manager-1.5.3 idna-3.7 kconfiglib-13.7.1 msgpack-1.0.8 packaging-24.1 pycparser-2.22 pyparsing-2.3.1 pyyaml-6.0.1 requests-2.32.3 requests-file-1.5.1 requests-toolbelt-1.0.0 schema-0.7.5 six-1.16.0 tqdm-4.66.4 urllib3-1.26.19
Reading CMake configuration...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- Configuring incomplete, errors occurred!
See also "/data/build/ruby-room-climate/.pioenvs/ruby-room-climate/CMakeFiles/CMakeOutput.log".
See also "/data/build/ruby-room-climate/.pioenvs/ruby-room-climate/CMakeFiles/CMakeError.log".

CMake Error at CMakeLists.txt:2 (include):
  include could not find load file:

    /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake

CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
```

echavet, 2024-07-29:
Hi @tmchow 
This is not an issue with the current project.
It looks like your esp32 environment has not been setup correctly.
This might not be the right place to find help.
I wish I could help, but I'm not an expert in setting up the environment. I think I have followed an esp32 installation guide...

closed 2024-07-29 by echavet

tmchow, 2024-07-29:
The issue ended up being I needed to use `type: Arduino` for ESPHome config instead of `type: esp-idf`