echavet/MitsubishiCN105ESPHome #139: No response to Connect Request - HRVU LGH-50RVX-E.

L4ky, 2024-07-29:
Hello, i'm playing a bit with the Mitsubishi installation in our business unit.
I successfully connected the D1 Mini using esphome and the esphome-mitsubishiheatpump library.

Now i'm trying to connect to our Heat Recovery Ventilation Unit without success. The model is LGH-50RVX-E.

Using MitsubishiCN105ESPHome i see that there is absolutely 0 response to the CONNECT REQUEST sent by the D1Mini ( [12:32:05][D][WRITE:157]: FC 5A 01 30 02 CA 01 A8 )

I tried many baud rates ( 2400 4900 9600 19200 and 115200 ). Always 0 response.

Any hint?

echavet, 2024-07-30:
Well, hard to tell!

1st. You could try the chip from the working first setup on the unit that doesn't work
2nd. You could try to add a voltage regulator 
3rd. You could try with a more reliable chip, like an esp32 S3
And at the end you could consider that maybe the unit is not supported

L4ky, 2024-07-30:
> And at the end you could consider that maybe the unit is not supported

I actually reached this conclusion, but i was hoping some hints on how to approach to support it.

Pre-installed control unit allows to change between 2 working modes + Auto mode, it also allow to regulate fan speed.

echavet, 2024-08-14:
hey @L4ky 
can I close this issue? 
I think nobody can help because this is a specific unit.

dwmw2, 2024-09-24:
What do you get if you send it these bytes at 2400 or 9600 baud:
```
02 FF FF 00 00 00 00 02
```

L4ky, 2024-09-24:
> What do you get if you send it these bytes at 2400 or 9600 baud:
> 
> ```
> 02 FF FF 00 00 00 00 02
> ```

Absolutely nothing.
I already tried with: 

>   - platform: uart
    name: "Connect Request 1"
    data: [0x02, 0xff, 0xff, 0x00, 0x00, 0x00, 0x00, 0x02]
  - platform: uart
    name: "Connect Request 2"
    data: [0x02, 0xff, 0xff, 0x01, 0x00, 0x00, 0x01, 0x00, 0x00]
  - platform: uart
    name: "Connect Request 3"
    data: [0xfc, 0x5a, 0x02, 0x7a, 0x02, 0xca, 0x01, 0x5d]
  - platform: uart
    name: "Connect Request 4"
    data: [0xfc, 0x5b, 0x02, 0x7a, 0x01, 0xc9, 0x5f]
  - platform: uart
    name: "Connect Request 5"
    data: [0xfc, 0x41, 0x02, 0x7a, 0x10, 0x34, 0x00, 0x01, 0, 0, 0, 0, 0, 0, 0, 0, 1, 0, 0, 0, 0, 0xfd]

dwmw2, 2024-09-24:
But you have a pre-installed control unit on this port which *does* work?

dwmw2, 2024-09-24:
Using an ESP32 box with two serial ports (e.g.https://www.tindie.com/products/thehognl/heishamon-communication-pcb/ ) I couldn't get [uart_mitm](https://github.com/ssieb/esphome_components/tree/master/components/uart_mitm) to work, but I *did* manage to get it to relay packets back and forth as pure waveforms using the `remote_receiver` and `remote_transmitter` components:
```
remote_receiver:
 - id: proxy_rx
   pin:
     number: GPIO9
     inverted: true
   dump: raw
   rmt_channel: 4
   clock_divider: 40
   filter: 5us
   idle: 100ms
   on_raw:
     then:
       - if:
          condition:
            mqtt.connected:
          then:
           - remote_transmitter.transmit_raw:
              transmitter_id: ecodan_tx
              code: !lambda return x;
 - id: ecodan_rx
   pin:
     number: GPIO18
     inverted: true
   dump: raw
   rmt_channel: 5
   clock_divider: 40
   filter: 5us
   idle: 100ms
   on_raw:
     then:
       - remote_transmitter.transmit_raw:
          transmitter_id: proxy_tx
          code: !lambda return x;

remote_transmitter:
 - id: proxy_tx
   pin:
    number: GPIO8
    inverted: true
   rmt_channel: 1
   carrier_duty_percent: 100
 - id: ecodan_tx
   pin:
     number: GPIO17
     inverted: true
   rmt_channel: 3
   carrier_duty_percent: 100
```
We can then look at the logged packets using https://dbuezas.github.io/esphome-remote_receiver-oscilloscope/

L4ky, 2024-09-24:
> But you have a pre-installed control unit on this port which _does_ work?

I'm not sure if it works.
I already tried multiple ESP without success.

Same setup on the HVAC works fine.

I was thinking about using the manual control ports which lets you set fan speed and mode by closing the circuit on different contacts.

phidauex, 2025-01-03:
Closing this issue as unsupported unit for now. Thanks for trying to make it work on the HRV, but sounds like the protocol is different despite the same connector being used. One longshot might be to look at the Ecodan component - it is a different Mitsubishi line using the same CN105 port, might be some similarities to the HRV? https://github.com/gekkekoe/esphome-ecodan-hp

closed 2025-01-03 by phidauex