echavet/MitsubishiCN105ESPHome #140: Reporting 0C/32F current temperature

tmchow, 2024-07-29:
without remote temperature sensor configuration, my unit is reporting a current temp of 0 degrees celcius for some reason.

Screenshot from the web server:
[image]

(In HA since it’s reporting in Fahrenheit it says it’s 32F). I expected this to work right out of the box so confused why it’s not.

phidauex, 2024-08-01:
Can you confirm which heatpump you have? The only difference I see between a clean setup and yours is that you are using the arduino framework rather than esp-idf. It is my understanding that the latest versions of this component need the esp-idf framework for ESP32 devices: https://github.com/echavet/MitsubishiCN105ESPHome#step-3-configure-the-board-and-uart-settings

That might require changes to your LED strip configuration, but even if you commented those sections out for the moment just to test if a build using esp-idf works, that might help.

echavet, 2024-08-01:
Wow I didn't know that

I'll check if the arduino framework is still supported when I am back from holiday. Near 9th August

phidauex, 2024-08-01:
I could be wrong, I thought that was something you had changed a few releases ago in order to implement a better UART?

echavet, 2024-08-01:
Yes but it is still supposed to work with arduino fw.

tmchow, 2024-08-02:
@phidauex 

> Can you confirm which heatpump you have? The only difference I see between a clean setup and yours is that you are using the arduino framework rather than esp-idf. It is my understanding that the [...]

I've tried to use `esp-idf` but each time I do, i get an error which i documented here:
https://github.com/echavet/MitsubishiCN105ESPHome/issues/138

I get the following CMAKE error. I'm very new to arduino and esp so is likely just some dumb error on my part.But as soon as I use `type: arduino` it works.

```
Installing collected packages: pyparsing, urllib3, tqdm, six, pyyaml, pycparser, packaging, msgpack, kconfiglib, idna, future, filelock, contextlib2, colorama, click, charset-normalizer, certifi, schema, requests, cffi, requests-toolbelt, requests-file, cryptography, cachecontrol, idf-component-manager
Successfully installed cachecontrol-0.14.0 certifi-2024.7.4 cffi-1.16.0 charset-normalizer-3.3.2 click-8.1.7 colorama-0.4.6 contextlib2-21.6.0 cryptography-3.4.8 filelock-3.15.4 future-1.0.0 idf-component-manager-1.5.3 idna-3.7 kconfiglib-13.7.1 msgpack-1.0.8 packaging-24.1 pycparser-2.22 pyparsing-2.3.1 pyyaml-6.0.1 requests-2.32.3 requests-file-1.5.1 requests-toolbelt-1.0.0 schema-0.7.5 six-1.16.0 tqdm-4.66.4 urllib3-1.26.19
Reading CMake configuration...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- Configuring incomplete, errors occurred!
See also "/data/build/ruby-room-climate/.pioenvs/ruby-room-climate/CMakeFiles/CMakeOutput.log".
See also "/data/build/ruby-room-climate/.pioenvs/ruby-room-climate/CMakeFiles/CMakeError.log".

CMake Error at CMakeLists.txt:2 (include):
  include could not find load file:

    /data/cache/platformio/packages/framework-espidf/tools/cmake/project.cmake

CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:3 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
```

tdvantine, 2024-10-09:
I solved this by creating a template sensor that takes my F reading and converts it to C and using that as the external sensor instead

```
{{ ((states('sensor.tbr_weather_temperature') | float - 32) / 1.8)|round(1) }}
```

phidauex, 2024-10-13:
Closing here, resolved in the discussion, PR pending on updating documentation. I'm still not sure why the original method was working for some but not others, but using the sensor and converting there is more elegant anyway.

closed 2024-10-13 by phidauex