echavet/MitsubishiCN105ESPHome #145: Remote Temperature by mqtt works but falls back to internal temp sensor after 1 min

c1328, 2024-08-05:
Hi all,
many thanks for that project - I never expected to get a 12 years old AC so smart :-)
All controls work great but one tiny issue can be probably solved easily.
Since I have no HomeAssistant I have to deal with ioBroker and therefor I update the external remote temperature with the help of MQTT:

mqtt:  
  broker: <ip>
  username: <user>
  password: <password>
  on_json_message:
    topic: ac-og/sensor/remote_temp/set
    then:
      - lambda: |-
          if (x.containsKey("temperature")){
            id(acog).set_remote_temperature(x["temperature"]);
          }

The AC reports back receiving the remote temperature but falls back to internal temperature after 1 minute (I read already that's normal). So I push the remote temperature every 55 seconds.

The statement
`remote_temperature_timeout: 30min`
does not seem to have any effect.

My question is: how can I adjust the remote temperature timeout in my scenario to have the update interval longer that 55 seconds.

Left side: update every 3min (fallback to internal sensor)
Right side: update every 55 seconds
[image]

echavet, 2024-08-05:
Thanks for the message.
I'm surprised because the timeout was supposed to work well
I will check that asap
I'm in holidays far from my computer !!

echavet, 2024-08-14:
hello @c1328 
could you provide your complete yaml please?

echavet, 2024-09-05:
hello @c1328,
thanks for your yaml. Sorry for late answer.
- There is an warning log that should appear in you logs each time the temperature sensor timeout occurs.
```cpp
ESP_LOGW(LOG_ACTION_EVT_TAG, "Remote temperature timeout occured, fall back to internal temperature!");
```
Can you confirm that you see this warning in logs?

- I've looked at the code, that has been tested : https://github.com/echavet/MitsubishiCN105ESPHome/issues/22#issuecomment-1927474420, https://github.com/echavet/MitsubishiCN105ESPHome/issues/8#issuecomment-1927888324)
And I see no reason why it could fail. 
Perhaps calling `set_remote_temperature(x["temperature"]);` from lambda fails because the x["temperature"] is not a float?

c1328, 2024-09-08:
hi @echavet, thanks for you effort!

I spend some time for analysis an will give a chronological event log:
- 12:30:00 an update was send via MQTT with the room temperature that should be used as current temperature (seen@MQTT explorer)
- 12:30:25 the AC reports that received temperature as current temperature (seen@MQTT explorer)
- 12:31:35 the AC falls back to it's own current temperature (seen@MQTT explorer)
- [12:34:00][W][EVT_SETS:064]: Remote temperature timeout occured, fall back to internal temperature! (the timeout was set to 4 minutes)

So the warning appears after the set timeout, but the fallback happens always around 60 seconds after the update. I recognized that AC behaves correct (slow down when given room temperature is reached), so I would assume, that the value I give is a correct float/number.

When I send this update every 55 seconds via MQTT I can observe that AC current temperature is always the given room temperature. Maybe it is because of my old AC that this remote temp timeout cannot be overwritten ;-)

c1328, 2024-09-08:
At the moment of fallback (after 60 seconds) I found this in my logs:
[19:46:12][I][STATUS:132]: [received]-> [room C°: 29.5, operating: NO , compressor freq:  3 Hz]
[19:46:12][I][STATUS:132]: [current]-> [room C°: 27.0, operating: NO , compressor freq:  3 Hz]

29.5 ist AC temperature and 27.0 is the remote room temperature that should be used.

echavet, 2024-09-08:
Well I'm wondering wether your unit could me the pb! Maybe it doesn't support remote température as it is an old one.
What is its reference ?

c1328, 2024-09-08:
The external unit is SUZ-KA71VA2 (year 2012).....

echavet, 2024-09-09:
And the internal one ?

c1328, 2024-09-09:
it is PEAD-RP71JAQ

djrm05, 2024-10-06:
have you tried to add a filter to the remote temp configuration? i.e.:.  30s or 1min

# receive remote temp
mqtt:  
  broker: 192.168.25.201
  username: xxx
  password: !secret mqtt_password
  on_json_message:
    topic: ac-og/sensor/remote_temp/set
    then:
      - lambda: |-
          if (x.containsKey("temperature")){
            id(acog).set_remote_temperature(x["temperature"]);
          }
    filters:
        - heartbeat: 30s

c1328, 2024-10-13:
Thanks for that approach but what should that filter achive? And I am sorry to say I get a syntax error....

phidauex, 2024-10-13:
When you get a syntax error it is customary to paste the error so people can help you. ;)

The heartbeat filter will send a value every 30s, even if an update hasn’t been received by the sensor. This would help with your timeout triggering too early, though it would also defeat the purpose of the timeout because even a stuck or broken sensor will keep repeating the same value every 30s indefinitely. I’d use it for troubleshooting, but not for the long term. 

Check the indentation, it looks like there may be too many spaces before the heartbeat line. It should be space space dash space heartbeat.

c1328, 2024-10-13:
Sorry, you're absolutely right - I was in a hurry. Now I tried again and get this
`[filters] is an invalid option for [on_json_message]. Please check the indentation.
    filters: 
      - heartbeat: 30s`

To what indentation should filters belong to? Thankful for a hint.

phidauex, 2024-10-14:
Ah, that is helpful, what it is telling you is that `filters` isn't a valid option for the `on_json_message` action, which makes sense. Move the indentation up two spaces, so that `filters` is under `mqtt` only. YAML is very indentation sensitive, as you can tell.

Unfortunately, it might complain about that as well, since looking at the docs, mqtt doesn't look like it supports filters either: https://esphome.io/components/mqtt.html#mqtt-on-json-message. If it doesn't work, then you may have to change tactics a bit, and create a new `sensor` in your configuration called `remote_temp_sensor`, and change your MQTT lambda to update that sensor, rather than update the heatpump directly. That way you can use all of the sensor features.

Try this - all written from memory here, so there are probably errors, but hopefully it gets you on your way:

```yaml
# Add to your sensors: section, this sensor receives the value from the MQTT client, processes if necessary,
# (unit conversion), checks for valid range, then sends out a the last reported value every 30s.
# NOTE: As written this bypasses the internal remote temp timeout, a stuck sensor will continue reporting forever!
sensor:
  - platform: template
    name: "Remote Temperature Sensor"
    id: remote_temp
    internal: false
    disabled_by_default: true
    device_class: temperature
    state_class: measurement
    unit_of_measurement: "°C"
    filters:
    # Uncomment this line to convert F to C on incoming temperature
    #  - lambda: return (x - 32) * (5.0/9.0);
      - clamp: # Limits values to range accepted by Mitsubishi units
          min_value: 1
          max_value: 40
          ignore_out_of_range: true
      - heartbeat: 30s
    on_value:
      then:
        - logger.log:
            level: INFO
            format: "Remote temperature received from HA: %.1f C"
            args: [ 'x' ]
        - lambda: 'id(acog).set_remote_temperature(x);'

# receive remote temp, get the temperature value from the JSON, and publish to sensor remote_temp
mqtt:  
  broker: 192.168.25.201
  username: xxx
  password: !secret mqtt_password
  on_json_message:
    topic: ac-og/sensor/remote_temp/set
    then:
      - lambda: |-
          if (x.containsKey("temperature")){
            id(remote_temp).publish_state(x["temperature"]);
          }
```

djrm05, 2024-10-14:
@phidauex you are right, this is not supported by mqtt ... my fault. @c1328 sorry but I have this configured as sensor as well

c1328, 2024-10-14:
@phidauex Thanks! I have it running for a few hours now and reduced the MQTT update cycle - and my AC keeps the temperature of the remote sensor. So I would say this trick (via sensor) works :-)

Thanks to all others that helped to find a solution here. Gracias!

closed 2024-10-14 by c1328