echavet/MitsubishiCN105ESPHome #146: Fan speeds in wrong/weird order

supermarkert, 2024-08-08:
I have a Mitsubishi MSY unit and everything works, but notice odd visual behavior of the Climate component that gets built in Home Assistant. The unit appears to support 6 fan modes:
- Auto
- Quiet
- 1 (low)
- 2 (medium)
- 3 (middle)
- 4 (high)

However, when I add a climate component to Home Assistant, the order is always:
- Auto
- 1 (low)
- 2 (medium)
- 4 (high)  **<----**
- 3 (middle)  **<----**
- Quiet

I don't mind the Quiet placement, but 3 & 4 are definitely out of place. I see the same behavior whether I use the official Thermostat card or use a 3rd part "Simple Thermostat" card.
[image]
[image]

Is there any way to re-order the fan speeds so they at least show 1-4 properly? Preferably, Quiet should be before 1 (low).

phidauex, 2024-08-08:
I've dug into this one a bit and ran into a dead end. The issue appears to be with how ESPHome enumerates the fan modes. In this component the default fan modes are defined here (I don't think the order matters here): https://github.com/echavet/MitsubishiCN105ESPHome/blob/89e75b4eb4e0232a90f374bbb6410d7990566427/components/cn105/climate.py#L45

You can also specify other fan modes in your climate component definition.

But when ESPhome sends that over to HomeAssistant, it uses its own order, which is defined here: https://esphome.io/api/namespaceesphome_1_1climate#adb5ca2756a8991cedb030edebb3028dc

[image]

You can see the order here matches what you are seeing in HA.

My workaround for my MSZ-GLxxNA units is to just drop the "middle" speed from my list. Honestly if I don't want auto fan speed, it is probably because I want minimum or maximum, so it doesn't add much value to me to select "middle" or "medium", so dropping one from the list is a small sacrifice.

```yaml
      fan:
        _name: Fan
        auto:
          name: null
          icon: mdi:fan-auto
        low:
          name: null
          icon: mdi:fan-speed-1
        medium:
          name: null
          icon: mdi:fan-speed-2
        high:
          name: null
          icon: mdi:fan-speed-3
        middle: false
        quiet:
          name: null
          icon: mdi:volume-mute
```

supermarkert, 2024-08-08:
Thanks for the info.  
I dug into it a little more after I posted, and noticed that the latest Thermostat card in HA actually allows you to customize the order that is displayed. It doesn't fix the climate entity, just what's shown on the dashboard. So I can fix it with the standard card and it works good enough for my use. This seems new and I don't know exactly when this feature was added.  
Oddly, I can't change the displayed order in Simple Thermostat card; I had started using it a while ago specifically because it gave me more custom options for my mini split.  But with all the updates to HA, I think I'll just switch back to the native one.

disruptivepatternmaterial, 2024-08-18:
yeah this is silly and frustrating - I have to this point assumed this is because it seems many of the people writing code for esphome/ha do not speak English as a first language. As someone who has lived in many place and tried to speak many languages with hysterical results at time (repeatedly asking for bread I my coffee because I confused milk for bread and when the person put bread in my coffee it had the coffee shop laughing or when I asked for stamps once and instead demanded a stampede, somehow which caused all the postal workers to laugh hysterically), I just *shrug* because middle makes no sense being at the end HAHA...but like @echavet mentioned, I too remove middle form my setup.

mminehanNZ, 2025-01-01:
I have a similar issue with fan speeds. We have two units (PFFY-P63VLRMM-E and PEFY-P50VMS1-E for the Australian and New Zealand markets). The supported fan speeds are:
- Auto
- 1 (low)
- 2 (medium)
- 3 (high)
Notice that HIGH is 3 for our units.  There is no such speed as middle.

I have to set as:
```
supports:
      swing_mode: []
      fan_mode: [AUTO, LOW, MEDIUM, MIDDLE]
```
So the HA card displays middle instead of high.

I'm relatively new to HA having previously been a long time HomeSeer user. I am using WeMos D1 boards that I am changing from mitsubishi2MQTT over to ESPHome. The update boards connect and work well except for the fan speed and the card only showing the action as 'Off' or 'Idle'. Our unit do not support compressor frequency.

Thanks for any advice.

phidauex, 2025-01-03:
I'm going to close this one for now, the issue is really between ESPHome and HA, and there is now an HA workaround. If anyone notices a change in ESPHome climate entities or something in the future, feel free to reopen this if there is anything we can change in this component to help.

closed 2025-01-03 by phidauex