echavet/MitsubishiCN105ESPHome #148: Compiler error format '%d' expects argument of type 'int', but argument 5 has type 'uint32_t'

markot74, 2024-08-20:
Hi,
I'm compiling with the latest esphome release for a esp32-c6. 

```
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c6-devkitc-1.html
PLATFORM: Espressif 32 (6.8.0) > Espressif ESP32-C6-DevKitC-1
HARDWARE: ESP32C6 160MHz, 512KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.50300.0 (5.3.0)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-esp32ulp @ 2.35.0-20220830
 - toolchain-riscv32-esp @ 13.2.0+20240530
Reading CMake configuration...
Generating assembly for certificate bundle...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
```
it compiles ok if i disable the debug, but as soon as i add the logger, i get these errors on different parts of the code:

_src/esphome/components/cn105/cn105.cpp:74:38: error: format '%d' expects argument of type 'int', but argument 5 has type 'uint32_t' {aka 'long unsigned int'} [-Werror=format=]
   74 |         ESP_LOGI(LOG_ACTION_EVT_TAG, "set_remote_temp_timeout is set to %d", timeout);_

I attach the full log of the compiler. 

is there an option that need to be set in the compiler to make it work?
thank you

[esp32-c6.log](https://github.com/user-attachments/files/16678502/esp32-c6.log)

echavet, 2024-08-21:
hi @markot74 ,
thank you for your message.
First: did you try to compile a fresh version with a `esphome clean` before you built?

I need your full yaml and your full terminal logs including the command.
look at mine, i have infos about esphome version, espidf version...
it looks like your esp-idf version is 5.3.0, which is weird because mine is only 4.4.7.
I don't know how esphome manages these fw integration, but I'd like to understand how you managed to compile with 5.3.0.
(chatGPT tells me esphome manages its esp-edf version through PlatformIO)

```terminal
echavet@MacBook-Pro-8 MitsubishiCN105ESPHome % esphome compile esp32-test.yaml
INFO ESPHome 2024.8.0
INFO Reading configuration esp32-test.yaml...
INFO Detected timezone 'Europe/Paris'
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing esp32-hp-test (board: esp32doit-devkit-v1; framework: espidf; platform: platformio/espressif32@5.4.0)
------------------------------------------------------------------------------------------------------------------------------------------------------------------
Library Manager: Installing bblanchon/ArduinoJson @ 6.18.5
INFO Installing bblanchon/ArduinoJson @ 6.18.5
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: ArduinoJson@6.18.5 has been installed!
INFO ArduinoJson@6.18.5 has been installed!
Library Manager: Installing esphome/noise-c @ 0.1.6
INFO Installing esphome/noise-c @ 0.1.6
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: noise-c@0.1.6 has been installed!
INFO noise-c@0.1.6 has been installed!
Library Manager: Resolving dependencies...
INFO Resolving dependencies...
Library Manager: Installing esphome/libsodium @ 1.10018.4
INFO Installing esphome/libsodium @ 1.10018.4
Downloading  [####################################]  100%
Unpacking  [####################################]  100%
Library Manager: libsodium@1.10018.4 has been installed!
INFO libsodium@1.10018.4 has been installed!
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
 - framework-espidf @ 3.40407.0 (4.4.7) 
 - tool-cmake @ 3.16.4 
 - tool-ninja @ 1.9.0 
 - toolchain-esp32ulp @ 2.35.0-20220830 
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch5
Reading CMake configuration...
Generating assembly for certificate bundle...
Dependency Graph
|-- ArduinoJson @ 6.18.5
|-- noise-c @ 0.1.6
```

markot74, 2024-08-21:
hi,
thank you for answering!
all the log coming from the command was included in the file i attached, but here it is (the command I run is "_esphome run file.yaml_"):

```
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32-c6-devkitc-1.html
PLATFORM: Espressif 32 (6.8.0) > Espressif ESP32-C6-DevKitC-1
HARDWARE: ESP32C6 160MHz, 512KB RAM, 8MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-builtin, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-espidf @ 3.50300.0 (5.3.0)
 - tool-cmake @ 3.16.4
 - tool-esptoolpy @ 1.40501.0 (4.5.1)
 - tool-idf @ 1.0.1
 - tool-mconf @ 1.4060000.20190628 (406.0.0)
 - tool-ninja @ 1.9.0
 - tool-riscv32-esp-elf-gdb @ 11.2.0+20220823
 - tool-xtensa-esp-elf-gdb @ 11.2.0+20230208
 - toolchain-esp32ulp @ 2.35.0-20220830
 - toolchain-riscv32-esp @ 13.2.0+20240530
Reading CMake configuration...
Generating assembly for certificate bundle...
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ off, Compatibility ~ soft
Found 0 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Compiling .pioenvs\clima_bimbi_5\src\esphome\components\atc_mithermometer\atc_mithermometer.cpp.o
Compiling .pioenvs\clima_bimbi_5\src\esphome\components\binary_sensor\automation.cpp.o
Compiling .pioenvs\clima_bimbi_5\src\esphome\components\binary_sensor\binary_sensor.cpp.o
Compiling .pioenvs\clima_bimbi_5\src\esphome\components\binary_sensor\filter.cpp.o
Compiling .pioenvs\clima_bimbi_5\src\esphome\components\climate\climate.cpp.o
Compiling .pioenvs\clima_bimbi_5\src\esphome\components\climate\climate_mode.cpp.o
Compiling .pioenvs\clima_bimbi_5\src\esphome\components\climate\climate_traits.cpp.o
Compiling .pioenvs\clima_bimbi_5\src\esphome\components\cn105\climateControls.cpp.o
Compiling .pioenvs\clima_bimbi_5\src\esphome\components\cn105\cn105.cpp.o
Compiling .pioenvs\clima_bimbi_5\src\esphome\components\cn105\componentEntries.cpp.o
Compiling .pioenvs\clima_bimbi_5\src\esphome\components\cn105\cycle_management.cpp.o
Compiling .pioenvs\clima_bimbi_5\src\esphome\components\cn105\extraComponents.cpp.o
Compiling .pioenvs\clima_bimbi_5\src\esphome\components\cn105\heatpumpFunctions.cpp.o
Compiling .pioenvs\clima_bimbi_5\src\esphome\components\cn105\hp_readings.cpp.o
In file included from src/esphome.h:28,
                 from src/esphome/components/cn105/Globals.h:2,
                 from src/esphome/components/cn105/cn105.h:2,
                 from src/esphome/components/cn105/climateControls.cpp:1:
src/esphome\components\esp32_ble\const_esp32c6.h:62:1: warning: missing initializer for member 'esp_bt_controller_config_t::cca_drop_mode' [-Wmissing-field-initializers]
   62 | };
      | ^
src/esphome\components\esp32_ble\const_esp32c6.h:62:1: warning: missing initializer for member 'esp_bt_controller_config_t::cca_low_tx_pwr' [-Wmissing-field-initializers]
src/esphome\components\esp32_ble\const_esp32c6.h:62:1: warning: missing initializer for member 'esp_bt_controller_config_t::csa2_select' [-Wmissing-field-initializers]
src/esphome/components/cn105/climateControls.cpp: In member function 'void CN105Climate::controlDelegate(const esphome::climate::ClimateCall&)':
src/esphome/components/cn105/climateControls.cpp:32:32: warning: implicitly-declared 'constexpr wantedHeatpumpSettings::wantedHeatpumpSettings(const wantedHeatpumpSettings&)' is deprecated [-Wdeprecated-copy]
   32 |     logCheckWantedSettingsMutex(this->wantedSettings);
      |     ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/cn105/Globals.h:195:29: note: because 'wantedHeatpumpSettings' has user-provided 'wantedHeatpumpSettings& wantedHeatpumpSettings::operator=(const wantedHeatpumpSettings&)'
  195 |     wantedHeatpumpSettings& operator=(const wantedHeatpumpSettings& other) {
      |                             ^~~~~~~~
src/esphome/components/cn105/climateControls.cpp:17:57: note:   initializing argument 1 of 'void logCheckWantedSettingsMutex(wantedHeatpumpSettings)'
   17 | void logCheckWantedSettingsMutex(wantedHeatpumpSettings settings) {
      |                                  ~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~
src/esphome/components/cn105/climateControls.cpp:83:36: warning: implicitly-declared 'constexpr wantedHeatpumpSettings::wantedHeatpumpSettings(const wantedHeatpumpSettings&)' is deprecated [-Wdeprecated-copy]
   83 |         logCheckWantedSettingsMutex(this->wantedSettings);
      |         ~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~
src/esphome/components/cn105/Globals.h:195:29: note: because 'wantedHeatpumpSettings' has user-provided 'wantedHeatpumpSettings& wantedHeatpumpSettings::operator=(const wantedHeatpumpSettings&)'
  195 |     wantedHeatpumpSettings& operator=(const wantedHeatpumpSettings& other) {
      |                             ^~~~~~~~
src/esphome/components/cn105/climateControls.cpp:17:57: note:   initializing argument 1 of 'void logCheckWantedSettingsMutex(wantedHeatpumpSettings)'
   17 | void logCheckWantedSettingsMutex(wantedHeatpumpSettings settings) {
[... 355 lines cut ...]
    battery_voltage:
      name: "ATC Battery-Voltage"
    signal_strength:
      name: "ATC Signal"
    
uart:
  id: CLIMA_UART
  baud_rate: 2400
  tx_pin: GPIO10
 rx_pin: GPIO11
  
# Climate entity configuration
climate:
  - platform: cn105
    name: "Clima Bimbi"
    update_interval: 4s        # update interval can be ajusted after a first run and logs monitoring* 
    debounce_delay : 500ms
    
logger:
  level: DEBUG
  hardware_uart: USB_SERIAL_JTAG
  
ota:
  - platform: esphome
```

markot74, 2024-08-21:
ps. I solved the issue easily substituting "%d" with "%lu" in the files where the errors occurred...

echavet, 2024-08-21:
yes I saw your attached files.
Thanks for the feedback and the tip to test with manual version. Did you notice the warning?
`WARNING The selected ESP-IDF framework version is not the recommended one. If there are connectivity or build issues please remove the manual version.`
Ok I'm am now able to reproduce your error. thanks again

echavet, 2024-08-21:
can hyou explain me the reason why you want to manually force the idf version? 
because I'm facing a regression when merging.
With the latest version I need to use %lu but with the default one, I'm facing the error:
`error: format '%lu' expects argument of type 'long unsigned int', but argument 5 has type 'uint32_t' {aka 'unsigned int'} [-Werror=format=]`
it looks like the uint32_t type has been changed in the recent version

markot74, 2024-08-21:
I have to force the version as the esp32-c6 platform needs the new version... (at least some of the functionality i want to use)
and it is not yet supported by esphome, so you need to declare and force the versions. 
Stange behaviour though, because I believe the problem relies in the compiler version (but looking at your log, the version appears to be the same), not in the framework version, as it's the compiler that complains...  

I copied and modified the files on my pc, thank you for trying!

echavet, 2024-08-22:
Actually, I did it with a test macro on GCC version

PR #149 (2024-08-22): Compiler error format '%d' expects argument of type 'int', but argument 5 has type 'uint32_t'
Fixes #148
commit 1c64a4f9 (2024-08-22): Compiler error format '%d' expects argument of type 'int', but argument 5 has type 'uint32_t'
Fixes #148
closed 2024-08-22 by echavet