echavet/MitsubishiCN105ESPHome #151: Controlling fan speed crashes ESP8266

sebi5361, 2024-08-23:
I am trying to control the fan speed of my MSZ-FH35VE2 HVAC but am having some trouble.
I implemented [this code](https://github.com/geoffdavis/esphome-mitsubishiheatpump/issues/138#issuecomment-1950899075) in my [yaml configuration file](https://gist.github.com/sebi5361/fd0f0d0fd4fbf6d1a30b5c05e30d9746), but the command `call.set_fan_mode(x.c_str());` crashes my ESP8266 (see [this post](https://github.com/geoffdavis/esphome-mitsubishiheatpump/issues/138#issuecomment-2305833711) for output log).
Any idea on how to solve this?

echavet, 2024-08-23:
Hello @sebi5361 
thanks for your message, but this is not an issue. You can use the "discussions" for this kind of topic.
As I understand you want to do this so as to change the fan mode from the web server page?
But the method signature is not the same as the Geoff Davis project.
in this component it is: `void setFanSpeed(const char* setting);`

But I think it won't work because the method is defined as protected. So if the esp lambda function is not executed in an inherited context, you might not be able to run it.
Just tell me if it fails, I will set the method public.

sebi5361, 2024-08-23:
Hi Eric,
Indeed this post would rather be a discussion. I just learned about discussion possibilities on your repository. Feel free to change this issue into a post as you already did.
I would like to control the fan intensity from a home assistant dashboard.
It seems the [working example configuration file](https://github.com/echavet/MitsubishiCN105ESPHome/discussions/129#discussioncomment-10016539) doesn't offer this possibility.
Thus, I implemented [this code](https://github.com/geoffdavis/esphome-mitsubishiheatpump/issues/138#issuecomment-1950899075) in [my configuration file](https://gist.github.com/sebi5361/fd0f0d0fd4fbf6d1a30b5c05e30d9746) as it was said to work with Geoff Davis code.
I couldn't make it work, so I tried doing the same with your code. But it crashed as well (see log below).
Should I use `setFanSpeed` instead of `call.set_fan_mode(x.c_str());`? Is it what you mean? Can you explain me what to modify in my configuration file?

---
```
[00:08:24][I][CN105:245]: sending wantedSettings..
[00:08:24]\xfcA0u[I][CN105:564]: fan setting changed
[00:08:25]Fatal exception 28(LoadProhibitedCause):
[00:08:25]epc1=0x4000bdc8, epc2=0x00000000, epc3=0x00000000, excvaddr=0x00000000, depc=0x00000000
[00:08:25]
[00:08:25]--------------- CUT HERE FOR EXCEPTION DECODER ---------------
[00:08:25]
[00:08:25]Exception (28):
WARNING Exception type: Access to invalid address: LOAD (wild pointer?)
[00:08:26]epc1=0x4000bdc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000
[00:08:26]
[00:08:26]>>>stack>>>
WARNING Found stack trace! Trying to decode it
[00:08:26]
[00:08:26]ctx: cont
[00:08:26]sp: 3ffffcd0 end: 3fffffd0 offset: 0150
[00:08:26]3ffffe20:  3ffffe96 3ffecdb8 00000006 3fff5534  
[00:08:26]3ffffe30:  3fff5518 3fff5334 3fff5334 4021cd3c  
WARNING Decoded 0x4021cd3c: CN105Climate::publishWantedSettingsStateToHA()
[00:08:27]3ffffe40:  3fff5518 3fff5334 00000000 4021d267  
WARNING Decoded 0x4021d267: CN105Climate::sendWantedSettingsDelegate()
[00:08:27]3ffffe50:  00000000 00000000 bf800000 3ffea75b  
[00:08:27]3ffffe60:  00000000 00000000 00000000 00000000  
[00:08:27]3ffffe70:  00000000 00000000 00000101 0000341b  
[00:08:28]3ffffe80:  300141fc 00080110 00000000 00000000  
[00:08:28]3ffffe90:  00000000 00007500 3fff3f24 4021fbd9  
WARNING Decoded 0x4021fbd9: esphome::logger::Logger::log_vprintf_(int, char const*, int, __FlashStringHelper const*, __va_list_tag)
[00:08:28]3ffffea0:  00000000 40216fd4 3fff62b0 ffffbb3c  
WARNING Decoded 0x40216fd4: std::_Function_handler<void (int, char const*, char const*), esphome::api::APIServer::setup()::{lambda(int, char const*, char const*)#1}>::_M_invoke(std::_Any_data const&, int&&, char const*&&, int&&) at api_server.cpp
[00:08:28]3ffffeb0:  3fff878c 3fff86dc 00000000 4021d8df  
WARNING Decoded 0x4021d8df: std::_Function_handler<esphome::RetryResult (unsigned char), CN105Climate::emulateMutex(char const*, std::function<void ()>&&)::{lambda(unsigned char)#1}>::_M_invoke(std::_Any_data const&, unsigned char&&) at utils.cpp
[00:08:28]3ffffec0:  3fff878c 3fff3d7c 3fff85ec 4022f5d0  
WARNING Decoded 0x4022f5d0: esphome::retry_handler(std::shared_ptr<esphome::RetryArgs> const&) at scheduler.cpp
[00:08:29]3ffffed0:  3fffff09 3ffffef0 00000010 40101410  
WARNING Decoded 0x40101410: get_unpoisoned_check_neighbors at umm_malloc.cpp
[00:08:29]3ffffee0:  00000000 0000e63a 69ba5e35 010a338a  
[00:08:29]3ffffef0:  3fff868c 00000020 3fff6304 ffffbb3c  
[00:08:30]3fffff00:  3fff878c 3fff53f4 3fffff30 ffffbb3c  
[00:08:30]3fffff10:  3fff878c 3fff2bf4 3fff85ec 4022fa41  
WARNING Decoded 0x4022fa41: esphome::Scheduler::call()
[00:08:30]3fffff20:  00000000 0000bb75 0000ce98 00000000  
[00:08:30]3fffff30:  0000e63a 3fff53f4 00000000 3fff868c  
[00:08:30]3fffff40:  3fff425f 40216fd4 00000010 00da81e1  
WARNING Decoded 0x40216fd4: std::_Function_handler<void (int, char const*, char const*), esphome::api::APIServer::setup()::{lambda(int, char const*, char const*)#1}>::_M_invoke(std::_Any_data const&, int&&, char const*&&, int&&) at api_server.cpp
[00:08:31]3fffff50:  0000ce98 00000000 3ffe8834 3fff53f4  
[00:08:31]3fffff60:  0000c033 3ffeba71 3fff46a4 3fff30c8  
[00:08:31]3fffff70:  3fffdad0 00000000 3fff2bf4 4022c645  
WARNING Decoded 0x4022c645: esphome::Application::loop()
[00:08:31]3fffff80:  00000000 3fff53f4 3fff680c 3fff30c8  
[00:08:32]3fffff90:  3fff3d00 3fff52a8 3fff2c28 3fff30c8  
[00:08:32]3fffffa0:  3fffdad0 00000000 3fff309c 3fff30c8  
[00:08:32]3fffffb0:  3fffdad0 00000000 3fff309c 402
\xe7\xfc/\xbc3
```

sebi5361, 2024-08-24:
I am really sorry Eric: I just found the possibility to control the fan speed within home assistant without adding those lines of code.
A first, I couldn't find the fan mode in the default lovelace climate card, which made me confused and I thought I needed to add some line of code.
But I found it when clicking on Integrations > ESPHome > Device > Controls > Climate.
I am closing this issue and will try to learn more about how to use this card.

closed 2024-08-24 by sebi5361