echavet/MitsubishiCN105ESPHome #152: Horizontal vane state doesn't show up after ESP reboots

sebi5361, 2024-08-27:
As can be seen on the below picture, the horizontal vane state doesn't show up after the ESP reboots:
[image]
...whereas the vertical vane state is shown properly.
I believe it is a (minor) bug as the configuration for both vanes is similar:
```yaml
    vertical_vane_select:
      name: ${friendly_node_name} Vertical Vane
    horizontal_vane_select:
      name: ${friendly_node_name} Horizontal Vane
```

echavet, 2024-08-27:
Ok thanks
I'll have a look
Could ou please provide your full yaml?

sebi5361, 2024-08-28:
Thanks Eric. Here is my full yaml:
[mitsubishi.yaml](https://github.com/user-attachments/files/16778179/mitsubishi.yaml.txt)

arnerek, 2024-11-08:
Im also having the same issue. It also does not respond to any command. I had this functionality before with mitsubishi2mqtt

echavet, 2024-11-08:
Hi, 
Sorry I had left this issue unresolved.
The issue only occurs when the ESP reboot? 
So you have to force sync by selecting a horizontal van position and then it keeps matching the unit state ?
What happens when you change the horizontal van setting with the IR remote control ?

echavet, 2024-11-08:
Weird question: are you absolutely sure your unit does have an horizontal vane ? Because I'm facing the same issue with my 2 units when activating the option, because both units don't support horizontal van orientation.

arnerek, 2024-11-08:
Yes I was able to control the horizontal vane on mitsubishi2mqtt set-up. I never see any signs of debug logs of horizontal vane when enabling DEBUG logging of Decoder neither.

arnerek, 2024-11-09:
I have been testing some more. 

My previous comment is partially incorrect. The set functionality does indeed work, but not the reading. When I restored this to SwiCago reading of the wideVane it worked perfectly. 

https://github.com/echavet/MitsubishiCN105ESPHome/commit/ee0ad7d0e45cf813a0dac6b0b731bc56ff5e4ac5

[Decoder:206] | [wideVane: →→ (adj:0)]

echavet, 2024-11-27:
thanks to @arnerek, this issue is fixed now.
Pull request merged

closed 2024-11-27 by echavet