echavet/MitsubishiCN105ESPHome #171: Input power differs between two units connected to the same multisplit

nununo, 2024-10-13:
Hello,

I just upgraded to the latest version which introduces the new energy sensors which are great. I actually have each unit connected to a Shelly Plug so that I can measure how much power each one draws. It seems I can get rid of those now. Thanks!!

And yet, I find something weird. The two units shown below are connected to the same multisplit. And yet, sensor `input_power` reports a distinct value for each one: one reports 1W and the other reports 16W. How can this be when they're both turned off? Shouldn't they both report the same power input?

[image]

Also, as you can see the Shelly Plug S reports 5.1W which is not 1W nor 16W but I guess that is expectable.

Here is the list of sensors for each one:
[image]
[image]

I also noticed that `energy_usage` is also very different between the two units but I guess that could make sense if each internal unit only adds to it when it's actually on. And indeed one of these units is used much more frequently than the other.

Please let me know if any other information is needed. I can try to get it.

Thanks in advance.

nununo, 2024-10-13:
Ok this is weirder:

I turned on the one reporting 1W (Suite) and this happened:
[image]

Slowly but steadily, both sensors Suite and Visitas increased their input power to respectively 20W and 635W. Visitas' 635W is very close to the 616W reported by the Shelly which suggests this one is reporting the right value. But this sensor actually belongs to the unit that remained off.

So I looked at the other multiplit unit I have and found that they're both reporting 20W when one of them is turned on:
[image]

This suggests that, of the 4 units (actually 2 multisplits), only one of them seems to be reporting the correct values.

So I thought: maybe doing a power cycle on the multisplit solves it. But it didn't. We're back to 1W and 16W:
[image]

It's weird because they're all sharing the same YAML, differing only in some `substitutions`:

Suite:
```yaml
substitutions:
  device_name: t4_suite_ac
  domain_name: t4-suite-ac
  api_key: !secret api_key
  ota_password: !secret ota_password
  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_password
  wifi_domain: .monte.nunogodinho.com
  remote_temperature_sensor: t4_suite_clima_temperature

packages:
  ac-mitsubishi: !include apps/ac-mitsubishi-ap20.yaml
```

Visitas:
```yaml
substitutions:
  device_name: t4_visitas_ac
  domain_name: t4-visitas-ac
  api_key: !secret api_key
  ota_password: !secret ota_password
  wifi_ssid: !secret wifi_ssid
  wifi_password: !secret wifi_password
  wifi_domain: .monte.nunogodinho.com
  remote_temperature_sensor: t4_visitas_clima_temperature

packages:
  ac-mitsubishi: !include apps/ac-mitsubishi-ap20.yaml
```

Then I decided to reinstall the firmwares (clearing all build files of course):
- After reinstalling the Suite firmware nothing changed: still reporting 1W.
- After reinstalling the Visitas firmware, nothing changed either: still reporting 16W.

I hope this helps because I don't know what else to do to better understand this weird behavior.

Please let me know if there's anything else I can do to help.

Regards.

enomam, 2024-10-13:
I've noticed that with a multi head unit - only one head actually reports power usage, and reports the usage of the outdoor unit.
So if there are 3 units, only one of them will report on the power draw of the outdoor unit (not at the individual level).

echavet, 2024-10-14:
Thank you very much @nununo for this detailed report. 
I have merged this great feature from [miipekk-ihme](https://github.com/miipekk-ihme)'s fork. I've sent him a invitation to join the project.

nununo, 2024-10-14:
> I've noticed that with a multi head unit - only one head actually reports power usage, and reports the usage of the outdoor unit. So if there are 3 units, only one of them will report on the power draw of the outdoor unit (not at the individual level).

Ok that makes sense. But it still doesn't explain two things:
- Why then does the other one still report some useless power input?
- Why is it that in my other multisplit both units are both reporting the wrong power input?

I hope we manage to figure this out because I would very much like to get rid of the Shellys that I'm currently connecting the units to.

f-vicente, 2024-12-30:
> Ok that makes sense. But it still doesn't explain two things:
>
> * Why then does the other one still report some useless power input?
> * Why is it that in my other multisplit both units are both [...]

One unit reports external compressor + unit fan power, the others only reports the unit fan power. Most of the power is consumed by the compressor, the unit fan even at full speed probably only uses 20-30w.

If the 1w and 16w "idle" are accurate is another problem, but sure not related with this project, it only shows what the unit is reporting. Keep in mind that most energy meters are not very accurate specially when measuring a few Watts.

I have exactly the same idle measures as you, 16w the first unit, 1w the second. When turned on, the energy report looks fine for me, but of course it has a 15% of difference compared with an external meter (but I also think that the meter is not very precise).

If you use from home assistant, you can sum the two values with a helper, to get the total usage. Although a few watts of difference from the second unit will not make a big difference at the end. So I will just simply use the energy report from the "main" unit that reports the most of the power usage.

nununo, 2024-12-30:
Yeah that makes sense. I'll do the same. There are definitely worse problems in the world than this one :) Closing. Thanks!

closed 2024-12-30 by nununo