echavet/MitsubishiCN105ESPHome #180: Unit's "Room Temperature" is never set to the external temperature sensor but continues to use internal temperature

ChadHydro, 2024-11-13:
When using this component my unit will only cool to the internal temperature and won't use the external temperature sensor provided. I've been successfully using this component for over a year: https://github.com/geoffdavis/esphome-mitsubishiheatpump. It's able to read the external temperature sensor and the unit will successfully cool to that temperature. The climate component will also display the current temperature as the external temperature sensor.

As you can see in the attached logs, the external temperature sensor has a value of 21.78 whereas the unit is still reporting the Room Temperature as 16.00. That is the internal value of the unit which should be overridden.

I have tried a number of different config values around the external sensor such as setting filters etc.

To be clear, I can control the unit, changing the modes etc. It just won't accept the external sensor values for some reason.

Mitsubishi Model: SVZKP18NA
ESP8266: Wemos D1 Mini

ESPHome Config:
```
esphome:
  name: esphome-web-149f52
  friendly_name: Attic Left Heat Pump

esp8266:
  board: d1_mini

# Enable logging
logger:
  baud_rate: 0
  level: DEBUG
  logs:
    EVT_SETS : DEBUG
    WIFI : INFO
    MQTT : INFO
    WRITE_SETTINGS : DEBUG
    SETTINGS : DEBUG
    STATUS : INFO
    CN105Climate: WARN
    CN105: DEBUG
    climate: WARN
    sensor: WARN
    chkSum : INFO
    WRITE : WARN
    READ : WARN
    Header: INFO
    Decoder : DEBUG
    CONTROL_WANTED_SETTINGS: DEBUG

uart:
 id: HP_UART
 baud_rate: 2400
 tx_pin: 1
 rx_pin: 3

external_components:
  - source: github://echavet/MitsubishiCN105ESPHome

climate:
- platform: cn105
  name: "Attic Left"
  # ESP32 only - change UART0 to UART1 or UART2 and remove the
  # logging:baud_rate above to allow the built-in UART0 to function for
  # logging.
  id: attic_left
  supports:
    mode: [COOL, HEAT, DRY, FAN_ONLY]
    fan_mode: [AUTO, LOW, MEDIUM, HIGH]
    swing_mode: []
  update_interval: 500ms
  visual:
    temperature_step: 1.0
    min_temperature: 7.2
    max_temperature: 33.3

# Enable Home Assistant API
api:
[... 350 lines cut ...]
[10:53:17][D][SETTINGS:087]: [climate]-> [mode: OFF, target °C: 23.0, fan: HIGH, swing: OFF]
[10:53:17][D][SETTINGS:521]: Settings changed, updating HA states
[10:53:17][I][CN105:649]: power or mode changed
[10:53:17][D][CN105:267]: Climate mode is: 0
[10:53:17][D][CN105:268]: Climate action is: 0
[10:53:17][I][CN105:619]: fan setting changed
[10:53:17][D][CN105:638]: Fan mode is: 5
[10:53:17][I][SETTINGS:528]: vane setting changed
[10:53:17][D][SETTINGS:550]: Swing mode is: 0
[10:53:17][D][CN105:024]: No value in hasChanged() function for wideVane
[10:53:17][D][CYCLE:320]: 3a: Sending room °C request (0x03)
[10:53:17][D][CN105:323]: creating Info packet
[10:53:17][D][CN105:084]: writing packet...
[10:53:17][D][CYCLE:326]: 3b: Receiving room °C response
[10:53:17][D][Decoder:255]: [Room °C: 16.000000]
[10:53:17][D][Decoder:256]: [OAT  °C: 7.000000]
[10:53:17][I][STATUS:134]: [received       ]-> [room C°: 16.0, operating: NO , compressor freq: nan Hz]
[10:53:17][I][STATUS:134]: [current        ]-> [room C°: 16.0, operating: NO , compressor freq: nan Hz]
[10:53:17][D][CN105:267]: Climate mode is: 0
[10:53:17][D][CN105:268]: Climate action is: 0
[10:53:17][D][CYCLE:329]: 4a: Sending status request (0x06)
[10:53:17][D][CN105:323]: creating Info packet
[10:53:17][D][CN105:084]: writing packet...
[10:53:19][W][CN105:009]: Cycle timeout, reseting cycle...
```

cortices, 2024-11-19:
It looks like you're getting `with timeout: YES` there with the debug printing enabled. I'm not certain but that might indicate that the cycles aren't able to complete.

Have you tried setting your `update_interval:` to longer, like `10s`, and seeing if it doesn't timeout?

ChadHydro, 2024-11-19:
@cortices appreciate you taking a look! Yes, I've tried a ton of different combinations of `update_interval`. Short, long, very long, I seem to get the timeout no matter what

ChadHydro, 2024-11-19:
SOLVED

For those who might run into a similar issue in the future, I'd had my esp8266 device daisy-chained with a Kumo cloud wifi adapter. Apparently it does not like that :)

closed 2024-11-19 by ChadHydro

cortices, 2024-11-20:
Oh yeah, a UART can only have one device at a time -- there's no collision detection/resolution. Buses like CAN and I2C are built differently to allow that.