echavet/MitsubishiCN105ESPHome #190: Widevane and Submode Cautions on SVZ Ducted Heat Pump

bass-windu, 2024-12-04:
Hello. I am overall very happy with this repo. I have implemented it with the ESP-01 and an airzone CN105 splitter alongside my MHK2 thermostat. It is integrated with homeassistant and is responding to commands.

My log is returning 2 issues I would like to fix:

> [16:13:55][W][lookup:204]: wideVane reading caution: value 0 not found, returning value at index 0
[16:13:55][W][lookup:204]: submode caution: value 1 not found, returning value at index 0

I do not believe my ducted unit has any vanes, just a fan. I have commented all of the lines referring to vanes. Here is my config yaml

My model numbers: SVZ-KP30NA (30k BTU ducted air handler) + MSZ-FS09NA (wall mounted mini split) on one MXZ-SM48NAMHZ2 48k BTU outdoor unit.

```
 esphome:
  name: ${name}
  friendly_name: ${friendly_name}

esp8266:
  board: esp01_1m

uart:
  id: HP_UART
  baud_rate: 2400
  tx_pin: 1
  rx_pin: 3

external_components:
  - source: github://echavet/MitsubishiCN105ESPHome
#    refresh: 0s

wifi:
  ssid: !secret ssid
  password: !secret password
  domain: !secret domain

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "${friendly_name} ESP"
    password: !secret fallback_password

captive_portal:

# Enable logging
logger:
  hardware_uart: UART1
  level: INFO
  logs:
    EVT_SETS : INFO
    WIFI : INFO
    MQTT : INFO
    WRITE_SETTINGS : INFO
    SETTINGS : INFO
    STATUS : INFO
    CN105Climate: WARN
    CN105: INFO
    climate: WARN
    sensor: WARN
    chkSum : INFO
    WRITE : WARN
    READ : WARN
    Header: INFO
    Decoder : INFO
    CONTROL_WANTED_SETTINGS: INFO

# Enable Home Assistant API
api:
  encryption:
    key: !secret api_key

ota:
  platform: esphome # Required for ESPhome 2024.6.0 and greater

# Enable Web server.
web_server:
  port: 80

time:
  - platform: homeassistant
    id: homeassistant_time

  - platform: version
    name: ESPHome Version
  - platform: wifi_info
    ip_address:
      name: IP
    ssid:
      name: SSID
    bssid:
      name: BSSID

  - platform: uptime
    name: Uptime

  - platform: wifi_signal
    name: WiFi Signal
    update_interval: 120s

button:
  - platform: restart
    name: "Restart ${friendly_name}"

climate:
  - platform: cn105
    id: hp
    name: "${friendly_name}"
    icon: mdi:heat-pump
    compressor_frequency_sensor:
      name: ${name} Compressor Frequency
#   vertical_vane_select:
#   name: ${name} Vertical Vane
#   horizontal_vane_select:
#   name: ${name} Horizontal Vane
#    isee_sensor:
#      name: ${name} ISEE Sensor
#   remote_temperature_timeout: 30min
#   debounce_delay : 500ms
    update_interval: 4s
    stage_sensor:
      name: Stage Sensor
    sub_mode_sensor:
      name: Sub Mode Sensor
    auto_sub_mode_sensor:
      name: Auto Sub Mode Sensor 
```

siljelb, 2024-12-31:
I also get the `[19:12:14][W][lookup:204]: wideVane reading caution: value 0 not found, returning value at index 0` on my MSZ-LN35VG2W, but only when I have the "Airflow control mode" set to anyhing but "off":
[image]

As OP points out, not sure how this would apply to a ducted system though.

phidauex, 2025-01-03:
It sounds like this unit is sending a wideVane value that isn't expected by this component. Currently the widevane value table is:

`WIDEVANE_MAP[7] = { "←←", "←", "|", "→", "→→", "←→", "SWING" }`

Read by the bytevaluemap func:
`receivedSettings.wideVane = lookupByteMapValue(WIDEVANE_MAP, WIDEVANE, 7, data[10] & 0x0F, "wideVane reading");`

It sounds like this unit is sending some other widevane value that isn't in this map. It may be possible to determine what that is by enabling the debug logs and cycling through the wideVane settings, if anyone has the inclination to do so.

joltman80, 2025-03-02:
@bass-windu Just to confirm, you are using your MHK2 thermostat alongside your ESP32 using the Airzone splitter?  What is the functionality of the MHK2 when you change the set point on the ESP32?  Is that temperature change reflected on the MHK2?  What about the other way around?  If you change the set point on the MHK2, does the ESP32 see that and report it to HA?  Thanks!

bass-windu, 2025-03-03:
@joltman80 yes I am using both an ESP32 and the MHK2. Communication and settings pass bidirectionally. If I set 20c on the MHK2 it is almost immediately reflected on the ESP32/HA and likewise HA/ESP32 temp and mode settings reflect in less than a minute on the MHK2. 

The purpose of the MHK2 is to allow for a physical interface for my ducted system for people in my house who may not be on HA and also to function as a remote temperature sensor. My ducted unit in the attic requires a remote sensor in the conditioned space off which it calculates how much heating/cooling to provide.

echavet, 2025-03-11:
hello,
like @phidauex wrote, you're facing unexpected values for both `wideVane`and 'submode`. It would be great if you could tell us about these two settings. 
When you get thoses messages, how is your unit configured? If you tell me the corresponding widevane mode (for code 0) and the corresponding submode fot code 1, I could complete the tab maps.

as you can see: WIDEVANE known codes are missing the 0 code and SUB_MODE[] is missing the 1 code.

````
static const uint8_t WIDEVANE[7] = { 0x01, 0x02, 0x03, 0x04, 0x05, 0x08, 0x0c };
static const char* WIDEVANE_MAP[7] = { "←←", "←", "|", "→", "→→", "←→", "SWING" };

static const uint8_t SUB_MODE[4] = { 0x00, 0x02, 0x04, 0x08 };
static const char* SUB_MODE_MAP[4] = { "NORMAL", "DEFROST", "PREHEAT", "STANDBY" };
```

phidauex, 2025-03-11:
It seems logical that 0x00 for WIDEVANE would mean something like, "NONE".

However, unclear what 0x01 for SUB_MODE would mean, there isn't an obvious intermediate between NORMAL and DEFROST, since we already have PREHEAT.

bass-windu, 2025-03-13:
@phidauex I'm not sure what 0x01 for SUBMODE means either. Would I be able to determine SUBMODE from my MHK2 thermostat? Currently all I see is "mode" which is set to heat. I do have the optional AUX heat strips. I do not have ERV or Humidifier control enabled.

djrm05, 2025-03-25:
Hi, I have the same messages ([W][lookup:206]: wideVane reading caution: value 0 not found, returning value at index 0) and i don't remember to see them few months back. Could it be related to any esphome update?
Anything I can do to help to see where the issue is?

Thanks!

djrm05, 2025-03-26:
@echavet @phidauex  This is what I see in debug. This is also a ducted one.
`[19:41:54][D][Decoder:206]: [Vane: ↓↓]
[19:41:54][W][lookup:206]: wideVane reading caution: value 0 not found, returning value at index 0
[19:41:54][D][Decoder:211]: [wideVane: ←← (adj:0)]
[19:41:54][D][SETTINGS:105]: [current]-> [power: OFF, target °C: 24.0, mode: AUTO, fan: 3, vane: ↓↓, wvane: ←←]
[19:41:54][D][SETTINGS:105]: [received]-> [power: OFF, target °C: 24.0, mode: AUTO, fan: 3, vane: ↓↓, wvane: ←←]
[19:41:54][D][EVT_SETS:062]: [wanted]-> [power: -, target °C: -1.0, mode: -, fan: -, vane: -, wvane: -, hasChanged ? ->  NO, hasBeenSent ? ->  NO]
[19:41:54][D][SETTINGS:089]: [climate]-> [mode: OFF, target °C: 24.0, fan: MIDDLE, swing: OFF]`

joergbattermann, 2025-04-08:
We've got a ducted PVA-A36AA7 & PUZ-A36NKA7 combo and I see the same warnings in the logs, so curious if/how this can be addressed/improved.

danolson1, 2025-04-26:
I've got a SLZ-KF09NA (indoor) SM48NAMHZ (outdoor) unit. I'm getting a similar message. See below.
```
[23:43:09][I][CYCLE:039]: 1: Cycle start
[23:43:09][D][CYCLE:311]: 2a: Sending settings request (0x02)
[23:43:09][D][CN105:323]: creating Info packet
[23:43:09][D][CN105:084]: writing packet...
[23:43:10][D][CYCLE:321]: 2b: Receiving settings response
[23:43:10][D][Decoder:178]: [0x02 is settings]
[23:43:10][D][Decoder:186]: [Power : OFF]
[23:43:10][D][Decoder:187]: [iSee  : 0]
[23:43:10][D][Decoder:188]: [Mode  : AUTO]
[23:43:10][D][Decoder:200]: [Temp °C: 19.000000]
[23:43:10][D][Decoder:203]: [Fan: 1]
[23:43:10][D][Decoder:206]: [Vane: SWING]
[23:43:10][W][lookup:206]: wideVane reading caution: value 0 not found, returning value at index 0
[23:43:10][D][Decoder:211]: [wideVane: ←← (adj:0)]
[23:43:10][D][SETTINGS:105]: [current]-> [power: OFF, target °C: 19.0, mode: AUTO, fan: 1, vane: SWING, wvane: ←←]
[23:43:10][D][SETTINGS:105]: [received]-> [power: OFF, target °C: 19.0, mode: AUTO, fan: 1, vane: SWING, wvane: ←←]
[23:43:10][D][EVT_SETS:062]: [wanted]-> [power: -, target °C: -1.0, mode: -, fan: -, vane: -, wvane: -, hasChanged ? ->  NO, hasBeenSent ? ->  NO]
[23:43:10][D][SETTINGS:089]: [climate]-> [mode: OFF, target °C: 19.0, fan: LOW, swing: VERTICAL]
[23:43:10][D][CYCLE:324]: 3a: Sending room °C request (0x03)
[23:43:10][D][CN105:323]: creating Info packet
[23:43:10][D][CN105:084]: writing packet...
[23:43:10][W][component:239]: Component esphome.coroutine took a long time for an operation (51 ms).
[23:43:10][W][component:240]: Components should block for at most 30 ms.
[23:43:10][D][CYCLE:330]: 3b: Receiving room °C response
[23:43:10][D][Decoder:258]: [Room °C: 25.000000]
[23:43:10][D][Decoder:259]: [OAT  °C: 15.000000]
[23:43:10][D][CYCLE:333]: 4a: Sending status request (0x06)
[23:43:10][D][CN105:323]: creating Info packet
[23:43:10][D][CN105:084]: writing packet...
[23:43:10][D][CYCLE:351]: 4b: Receiving status response
[23:43:10][D][Decoder:282]: [0x06 is status]
[23:43:10][D][CYCLE:355]: 5a: Sending power request (0x09)
[23:43:10][D][CN105:323]: creating Info packet
[23:43:10][D][CN105:084]: writing packet...
[23:43:10][D][CYCLE:370]: 5b: Receiving Power/Standby response
[23:43:10][D][Decoder:150]: [0x09 is sub modes]
[23:43:10][W][lookup:206]: submode caution: value 1 not found, returning value at index 0
[23:43:10][D][Decoder:157]: [Stage : IDLE]
[23:43:10][D][Decoder:158]: [Sub Mode  : NORMAL]
[23:43:10][D][Decoder:159]: [Auto Mode Sub Mode  : AUTO_LEADER]
[23:43:10][I][CYCLE:052]: 6: Cycle ended in 1.2 seconds (with timeout?:  NO)
[23:43:12][D][CN105:308]: sending a request for settings packet (0x02)
```

ams2990, 2025-05-13:
I've got an SVZ-KP12NA and an SVZ-KP18NA. I'm getting the same widevane error as everyone else, but no submode error:
```
[23:01:25][W][lookup:206]: wideVane reading caution: value 0 not found, returning value at index 0
```

I don't know what value 0 is supposed to mean, but the SVZ doesn't have vanes so this incompatibility isn't terribly surprising. Can we get a config option to disable vane support in the software (as well as disable the vane control option in the HA thermostat entity)?

djrm05, 2025-05-23:
> I've got an SVZ-KP12NA and an SVZ-KP18NA. I'm getting the same widevane error as everyone else, but no submode error:
> 
> ```
> [23:01:25][W][lookup:206]: wideVane reading caution: value 0 not found, returning value at index 0
> ```
> 
> I don't know what value 0 is supposed to mean, but the SVZ doesn't have vanes so this incompatibility isn't terribly surprising. Can we get a config option to disable vane support in the software? (~as well as disable the vane control option in the HA thermostat entity~ EDIT: I discovered `swing_mode: []` removes this)

@ams2990  Can you pls explain where did you added 'swing_mode[]' I have added in the climate component under supports: but the error still there.

umtauscher, 2025-06-03:
> It seems logical that 0x00 for WIDEVANE would mean something like, "NONE".

I would conclude, that WIDEVANE 0 means dynamic mode using the iSee Sensor. 
This mode doesn't have a fixed vane but changes by the sensor sensing people in the room.
The submodes would then represent the 4 different setting as [somlefant] pointed out..

disruptivepatternmaterial, 2025-07-22:
So my unit has NATURAL FLOW not EVEN, though EVEN shows up in the list. It is MSZ-FS09NA. Anyone else see this?

Not sure what to make of it yet...